espressif/arduino-esp32 #10343: Register a callback when the last byte is written on UART or when the RTS pin become low

hitecSmartHome, 2024-09-13:
### Related area

UART

### Hardware specification

UART

### Is your feature request related to a problem?

Currently ( with arduino v3 and idf v5 ) I can register callback functions for onReceive and onReceiveError.
When I receive a packet I immidiately want to write the next. It is not possible because the UART is still busy.
I had to implement a logic to wait for the UART to become available before I can write.

A cb function would be good if the UART is available again.

### Describe the solution you'd like

// Pass a cb just like in the case of onReceive
Serial1.onAvailable([](){
    // UART is avaiable again. Can write the next packet.
});

### Describe alternatives you've considered

Before I want to write I call this wrapper function

```cpp
while( isUartBusy() ){}
```

This is the implementation

```cpp
bool Modbus::isUartBusy(){
    return uart_wait_tx_done(UART_NUM_1, 0) == ESP_ERR_TIMEOUT;
}
```

This is not ideal because it freezes the task which wants to write.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2024-09-13:
Humm... so the feature request is about using UART in Half Duplex mode. 
While receiving, no writing.
While writing, no receiving.

You want a callback for  when all data has left UART FIFO?

hitecSmartHome, 2024-09-13:
Yes

hitecSmartHome, 2024-09-13:
Currently when I get a cb about the data receive, the app immediately wants to write the next packet. It cant write since the uart is still busy. I have to wait until the uart is available so I can write the next packet. This results in a busy while waiting loop on one of the task. If there would be a cb about the rts pin I could write the next packet in that cb because that would signal to the app that the uart is available

TD-er, 2024-09-13:
You can also look into hardware RS485 support of the ESP32.
This should then toggle some DE/RE pin and you could then register a callback on change of this pin (or rise/fall) to handle the next data.

hitecSmartHome, 2024-09-13:
Oh that is a good idea. Thank you very much will definitely try this

hitecSmartHome, 2024-09-13:
Current flow

- Write a packet
- Do other things while the slave processes the packet ( do not wait for anything in any task )
- Get a cb about a new response packet
- Process the response packet
- Wait for the busy uart
- Write a packet

This would be better

- Write a packet
- Do other things while the slave processes the packet ( do not wait for anything in any task )
- Get a cb about a new response packet
- Process the response packet
- Get a cb about the whole packet is transmitted ( rts pin is low from high ? )
- Write a new packet as soon as everything is transmitted ( in the cb ) ( no blocking while loop while the uart is busy in any task )

device111, 2024-09-18:
It looks like you want implement the Modbus Protocol. It gaves a lot of working librarys for this. 
i.e.: https://github.com/yaacov/ArduinoModbusSlave

Have you tried Serial.flush()? It does similar like "uart_wait_tx_done", without a timeout.

Why you don't use Serial.available() for the incoming Bytes? Serial.onReceive() is always a blocking Function.
And you can use the RTS Pin normaly before and after sending the packet. You don't must set the Serial to Half-Duplex.

TD-er, 2024-09-18:
@device111 If you need to implement RS485, then the ESP32-xx all do have hardware support for toggling the DE/RE pin and even have collision detection in hardware.
No need to make it complex.

device111, 2024-09-18:
Ok, you can use it for auto-toggle the DE Pin, but the rest of the implementation must affect to this. :wink:

hitecSmartHome, 2024-09-19:
Yes, I have implemented the modbus protocoll but not the traditional so I can't use any library for this.
How is the `onReceive` blocking? It loops in the `UART` task and calls my callback when the whole packet is received. It does not block anything. I have configured the hardware uart manager just like you said. Look at the `init` function

```cpp
void Modbus::init() {
    Serial1.begin(MBUS_BAUD, SERIAL_8N1, MBUS_RX, MBUS_TX);
    Serial1.setPins(-1, -1, -1, MBUS_RTS);
    Serial1.setMode(UART_MODE_RS485_HALF_DUPLEX);
    Serial1.setRxTimeout(MBUS_RX_TIMEOUT);

    Serial1.onReceive(
        std::bind(&Modbus::handlePacket, this),
        PACKET_TRIGGER_ONLY_ON_TIMEOUT
    );
    Serial1.onReceiveError(std::bind(&Modbus::handleReceiveError, this, std::placeholders::_1));
}
```

This uses `UART_MODE_RS485_HALF_DUPLEX` and I have also configured the pins 

```cpp
  Serial1.begin(MBUS_BAUD, SERIAL_8N1, MBUS_RX, MBUS_TX);
  Serial1.setPins(-1, -1, -1, MBUS_RTS);
```
so I don't really have to do any pin toggling.

In my application, when I got a packet, I immidiately send the next packet. ( it is a master on the bus )
When this happens, sometimes the bus is busy and the write won't happen. That is why I implemented this function

```cpp
bool Modbus::isUartBusy(){
    return uart_wait_tx_done(UART_NUM_1, 0) == ESP_ERR_TIMEOUT;
}
```

So it waits for the TX done. But I would prefer a callback style like `onReceive` about the transmit done event so I don't have to implement any busy loop.

hitecSmartHome, 2024-09-19:
This basically looks like this in a simple form

```cpp
void Modbus::init() {
    Serial1.begin(MBUS_BAUD, SERIAL_8N1, MBUS_RX, MBUS_TX);
    Serial1.setPins(-1, -1, -1, MBUS_RTS);
    Serial1.setMode(UART_MODE_RS485_HALF_DUPLEX);
    Serial1.setRxTimeout(MBUS_RX_TIMEOUT);

    Serial1.onReceive(
        std::bind(&Modbus::handlePacket, this),
        PACKET_TRIGGER_ONLY_ON_TIMEOUT
    );
    Serial1.onReceiveError(std::bind(&Modbus::handleReceiveError, this, std::placeholders::_1));
}

void Modbus::handlePacket() {
    // Check how many bytes are available
    int available = Serial1.available();
    // Check if it would overflow our buffer
    if (available >= MAX_MBUS_DATA_LENGTH) {
        ESP_LOGE(MBUS_DEBUG_TAG, "Packet is too big: %d bytes. Can't process it.", available);
        lastPacketError = BUFF_OVERFLOW;
        callErrorCb();
        return;
    }
    // Get all the data
    uint8_t rawPacket[available];
    int readBytes = Serial1.readBytes(rawPacket, available);
    // Check CRC and other error bytes.
    if (!isPacketValid(rawPacket, readBytes)) {
        ESP_LOGE(MBUS_DEBUG_TAG, "Invalid packet. Can't process it.");
        printRawPacket(rawPacket, readBytes);
        callErrorCb();
        return;
    }
    // Parse the packet if it was a scan packet.
    parseScanPacket(rawPacket, readBytes);
    // Call the packet callback if it wasn't a scan and we have a valid callback
    if (packetCallback && !isScanning) {
        packetCallback(rawPacket, readBytes);
    }
}
```

After these checks are done, the modbus calls the `packetCallback`.
This immidiately sends the next packet to the next slave but the write has to wait because there are cases when the bus is still busy. I need continous communication between the slaves and the master. It can not be interrupted or waited on. I have to write the next packet as soon as I can. The `onTxDone` or `onAvailable` callback would simplify this because I could send the next packet as soon as the `bus` is free.

hitecSmartHome, 2024-09-19:
Something like this would be really good

```cpp
void Modbus::init() {
    Serial1.begin(MBUS_BAUD, SERIAL_8N1, MBUS_RX, MBUS_TX);
    Serial1.setPins(-1, -1, -1, MBUS_RTS);
    Serial1.setMode(UART_MODE_RS485_HALF_DUPLEX);
    Serial1.setRxTimeout(MBUS_RX_TIMEOUT);

    Serial1.onReceive(
        std::bind(&Modbus::handlePacket, this),
        PACKET_TRIGGER_ONLY_ON_TIMEOUT
    );
    Serial1.onReceiveError(std::bind(&Modbus::handleReceiveError, this, std::placeholders::_1));
    Serial1.onTransmitDone(std::bind(&Modbus::handleTransmitDone, this, std::placeholders::_1));
}
```

I could send the next packet inside `handleTransmitDone`

whatdtech, 2024-09-27:
Hi @hitecSmartHome  did you find any solution for this?. I'm actually trying to adapt profibus stack for esp32 based on this project. [](https://github.com/tuenhidiy/Arduino_Profibus_DP). I need to adapt USART data register empty and uart receive interrupts for esp32.

hitecSmartHome, 2024-09-27:
Well, as I said, I'm doing it like this now:

```cpp
Serial1.onReceive([](){
 // Got a whole packet. That callback means the uart triggered a byte timeout.
 // Read the whole UART buffer into my own.
 int available = Serial1.available();
 uint8_t rawPacket[available];
 int readBytes = Serial1.readBytes(rawPacket, available);
 // Do some checks on the packet like CRC and things like that...
 if( isPacketValid() ){
     // If the packet was valid, process it.
     processValidPacket();
 }
 // Write the next packet but wait for uart busy in `writeNextPacket`
 writeNextPacket();
},PACKET_TRIGGER_ONLY_ON_TIMEOUT);
```

whatdtech, 2024-09-27:
@hitecSmartHome  uart receive interrupt is ok, but what about uart data register empty interrupt?

hitecSmartHome, 2024-09-30:
There is none.

Jason2866, 2025-03-13:
@SuGlider Is the feature request considered? If no let us close.

SuGlider, 2025-03-14:
No, we won't add a feature for that. But I can offer some ideas:

1- onReceive() will be called whenever UART RX sense a timeout (line is idle for a while) or whenever the RX FIFO is full (about 120 bytes for higher baud rates and just a single byte for lower baud rates). Therefore, the CB can be called many time before the package is completely received, depending on its length.
2- transmission (UART TX) can hold up to the last byte is sent using `HardwareSerial::flush()` => blocking until all data is sent out (TX FIFO is empty).

Suggestion would be:

- Create a FreeRTOS Queue with about double of the largest UART package.
- OnReceive() would just populate the Queue. It is an ISR and it should be short with a minimum neessary code for receive data. IT has a high priority, therefore, if it takes time, it will blokck all other Tasks, including `loop()`.
- Within a separated Task, it will read the Queue and process the Package checking its fields as defined in the protocol frame. Whenever a package is detected as completely reeived, it would flop some Semaphore or Spinlock. Such Semaphore would tell the system when it is time to Write back a response package.
- Within another separated Task, it would wait on the Semaphore (blocking the task) until it changes to tell the Task that Reading is Complete. Process the incoming package and formulate a Response. Write it back waiting (or not) until all bytes are sent.

SuGlider, 2025-03-14:
Regarding RTS pin detection:

Connect RTS to some free GPIO. Attach an ISR to that pin to detect whenever it goes low or high.
It is similar to detecting an Edge for a button.

hitecSmartHome, 2025-03-14:
These are good suggestions but a standard modbus protocol has no way to tell that we got a frame other than the byte timeout which is supposed to controll onReceive in half duplex mode. A modbus frame has no break or any signal character. It does not have a min length either and the packets are not consistent in length also. We only have the byte timeout which can not be determined from a task which waits on a queue. It looks like if we want to implement proper modbus protocol we must write our own driver and completly ditch the uart driver so we can go back to byte interrupts and measure the byte time manually.

TD-er, 2025-03-14:
@hitecSmartHome 
Not entirely sure if I do understand what you're writing.

You can enable collision detection and also trigger an interrupt when a collision happens if needed.
See: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#overview-of-rs485-specific-communication-0ptions

And for nice ASCII-art schematics (I use Circuit A in some of my designs), see: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#interface-connection-options
On this page there are some example applications linked.
I checked a few of them and most of those I checked use a while loop to continuously read from the UART. No interrupt or callback used on received data.

The NMEA GPS parser however is using quite an interesting approach with a pattern detection.
See: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/uart/uart_events/main/uart_events_example_main.c#L89
I don't know yet how this is working. NMEA sentences have an easy pattern to detect as the start and end of a message are always the same characters (and ending with a newline), so it might not be usable for your application.

I think the collision detection is probably the most elegant solution here.
This way DE/RE is dealt with in hardware and you can just focus on receiving data.
And maybe you can also install a callback directly on the RTS pin if you really need to be notified when the data has been sent.

Jason2866, 2025-03-14:
Closing since not planned. For further questions open a github discussion

closed 2025-03-14 by Jason2866

SuGlider, 2025-03-14:
> These are good suggestions but a standard modbus protocol has no way to tell that we got a frame other than the byte timeout which is supposed to controll onReceive in half duplex mode. A modbus [...]

If it needs to know when RX timeout happens, I'd suggest this:

`void onReceive(OnReceiveCb function, bool onlyOnTimeout = false)` has two parameters. Default (false) is calling whenever or it receives 1 (baudrate lower than 38.5K) or 120 bytes (higher baudrate) or when RX Timeout. FIFO number of bytes read that will trigger the callback (max UART RX FIFO is 127 bytes) canbe changed using `bool setRxFIFOFull(uint8_t fifoBytes)` for any baud rate.

About `onReceive()` parameters:
- The first is the callback `void function()`.
- The second parameter tells the UART data processing Task to only activate the callback function whenever the receiving data flow has finished within the RX Timeout defined with `setRxTimeout(uint8_t symbols_timeout)`. Note that the timeout is in number of symbols, considering all data frame bits (Example: 8N1 => Startbit + 8 data bits + Parity Bit + 1 Stop bit = 11 bits at the baud rate timing). Timeout in seconds shall be calculated based on baud rate and the UART Frame length.

This can be used with `setRxBufferSize(size_t new_size)` to make sure that all the data burst will fit in the UART buffer.
Note that it is necessary to set the RX Buffer Size before calling `Serial.begin()`

In that case, `Serial.available()` will return the length of the data stream that was received and it can be processed.

device111, 2025-03-14:
@hitecSmartHome 

Just a correction:

a.) Modbus does have a minimum frame length. It's 8 bytes.
b.) You also have the option of identifying a complete frame using the 2 CRC Bytes at the end of each frame.
c.) In addition, a gap time must be maintained between each frame to detect the beginning of a frame of each sender/receiver. You can measure it wih micros() 
(minimum is 1,75 ms...it's time enough, to meaure it.)

It is sufficient to measure the time between bytes and the time between two frames to identify a complete frame.
The frame is then checked using the two CRC bytes generated over the entire frame.

Please check the Modbus Protocol for more Details. None of the aivailable Librarys use interrupts or a callback for that "slow" Protocol.

All these works also for the RS485 half/full Duplex Mode.

br