espressif/arduino-esp32 #10346: Calling SD.begin() when no SD-card is inserted and CONFIG_DISABLE_HAL_LOCKS is not set causes LoadProhibited error

Gelemikke, 2024-09-13:
### Board

LilyGo T-SIM7000, LilyGo T-SIM7600

### Device Description

-

### Hardware Configuration

-

### Version

v3.0.4

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Calling `SD.begin()` when no SD card is inserted and `CONFIG_DISABLE_HAL_LOCKS is not set` in sdkconfig causes a _LoadProhibited_ error and crashes the code. The expectation is that `SD.begin()` would instead return `false` when no SD card is inserted.

Following the backtrace, the error is caused when `xSemaphoreGive(spi->lock);` is called in `void spiEndTransaction(spi_t *spi)`. I assume this fails because `spi->lock` does not point to a valid mutex. However, I can't figure out why that would be the case since `do {} while (xSemaphoreTake(spi->lock, portMAX_DELAY) != pdPASS)`  appears to be called a little bit earlier in `sdcard_uninit` and there it does not throw an error.

I unfortunately don't have a working debugger at the moment, so I can't properly step through the code in any more detail. Setting `CONFIG_DISABLE_HAL_LOCKS=y` in sdkconfig appears to be a working remedy but has its side effects as it disables all HAL mutexes. 

### Sketch

```cpp
#include <Arduino.h>
#include <FS.h>
#include <SPI.h>
#include <SD.h>

#define SD_MISO             2
#define SD_MOSI             15
#define SD_SCLK             14
#define SD_CS               13

extern "C" void app_main()
{
    initArduino();
    //Initialize SDCard
    SPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
    pinMode(SD_CS,OUTPUT);
    digitalWrite(SD_CS,HIGH);

    if (SD.begin(SD_CS)){ // This crashes if no SD-card is inserted and CONFIG_DISABLE_HAL_LOCKS is not set

    }
}
```

### Debug Message

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d8f3b  PS      : 0x00060730  A0      : 0x800d1ae1  A1      : 0x3ffbbb90
A2      : 0xcc491807  A3      : 0x00000000  A4      : 0x00000014  A5      : 0x00000063  
A6      : 0x00000063  A7      : 0x00000054  A8      : 0x80085d55  A9      : 0x3ffbbb70
A10     : 0x3ffb8000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0xcc49180b  LBEG    : 0x40089f89  LEND    : 0x40089f99  LCOUNT  : 0xfffffffc  

Backtrace: 0x400d8f38:0x3ffbbb90 0x400d1ade:0x3ffbbbb0 0x400d2c6e:0x3ffbbbd0 0x400d1ca3:0x3ffbbc00 0x400d1845:0x3ffbbc30 0x400fc45e:0x3ffbbc60 0x4008e6ba:0x3ffbbc80
  #0  0x400d8f38 in spiEndTransaction at C:/Users/mikfr01/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:1126
  #1  0x400d1ade in SPIClass::endTransaction() at C:/Users/mikfr01/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:193
  #2  0x400d2c6e in sdcard_uninit(unsigned char) at C:/Users/mikfr01/.platformio/packages/framework-arduinoespressif32/libraries/SD/src/sd_diskio.cpp:451
      (inlined by) sdcard_uninit(unsigned char) at C:/Users/mikfr01/.platformio/packages/framework-arduinoespressif32/libraries/SD/src/sd_diskio.cpp:685
  #3  0x400d1ca3 in fs::SDFS::begin(unsigned char, SPIClass&, unsigned long, char const*, unsigned char, bool) at C:/Users/mikfr01/.platformio/packages/framework-arduinoespressif32/libraries/SD/src/SD.cpp:39
  #4  0x400d1845 in app_main at src/main.cpp:20
  #5  0x400fc45e in main_task at C:\Users\mikfr01\.platformio\packages\framework-espidf\components\freertos/app_startup.c:208 (discriminator 13)
  #6  0x4008e6ba in vPortTaskWrapper at C:\Users\mikfr01\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/port.c:162
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-09-17:
Hi @Gelemikke, I am unable to reproduce the issue using Arduino IDE.
Looks you are using Arduino as a component in IDF. What IDF version are you using?

Gelemikke, 2024-09-17:
Yes, thats true! I'm running ESP-IDF version 5.1.4.240801

P-R-O-C-H-Y, 2024-09-17:
Thanks for quick response, Will try to reproduce there.

P-R-O-C-H-Y, 2024-09-17:
I can confirm the crash using Arduino as component. I will take look on what's wrong there. Thank you for reporting.

P-R-O-C-H-Y, 2024-09-17:
This issue was solved by https://github.com/espressif/arduino-esp32/pull/10122. The fix will be available in the 3.0.5 release.
I am closing this as solved.

closed 2024-09-17 by P-R-O-C-H-Y