espressif/arduino-esp32 #10347: CoAP GET returns ASCII 30 or char '0'

97Cweb, 2024-09-14:
### Board

ESP32-C6 to ESP32-C6

### Device Description

Waveshare ESP32-C6 dev board

### Hardware Configuration

no

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Cannot get a response from the Get command of coap. It returns 30 instead of a string

### Sketch

```cpp
2 Long sketches, need both parts, one is leader, the other is joiner

#include "OThreadCLI.h"
#include "OThreadCLI_Util.h"

#define OT_CHANNEL            "24"
#define OT_NETWORK_KEY        "00112233445566778899aabbccddeeff"
#define OT_MCAST_ADDR         "ff05::abcd"
#define OT_COAP_RESOURCE      "json"
#define USER_BUTTON           9

String joinerAddress = "";

const char *otSetupLeader[] = {
  // -- clear/disable all
  // stop CoAP
  "coap", "stop",
  // stop Thread
  "thread", "stop",
  // stop the interface
  "ifconfig", "down",
  // clear the dataset
  "dataset", "clear",
  // -- set dataset
  // create a new complete dataset with random data
  "dataset", "init new",
  // set the channel
[... 423 lines cut ...]
Joiner address set to: fdc7:388b:90bb:5a27:4611:3731:d16a:2515
Sending CoAP GET CMD: [coap get fdc7:388b:90bb:5a27:4611:3731:d16a:2515 json block-1024]
Try[5]::MSG[Done]
Command executed successfully.
Try[4]::MSG[coap response from fdc7:388b:90bb:5a27:4611:3731:d16a:2515 with payload: 30]
CoAP response received.
Sending CoAP GET CMD: [coap get fdc7:388b:90bb:5a27:4611:3731:d16a:2515 json block-1024]
Try[5]::MSG[Done]
Command executed successfully.
Try[4]::MSG[coap response from fdc7:388b:90bb:5a27:4611:3731:d16a:2515 with payload: 30]
CoAP response received.

[653707][I][beeton_joiner_v1.ino:114] otCoapPUT(): Send CLI CMD:[coap put ff05::abcd json con {"id":1}]
[653708][D][beeton_joiner_v1.ino:125] otCoapPUT(): Try[5]::MSG[Done]
[653726][D][beeton_joiner_v1.ino:125] otCoapPUT(): Try[4]::MSG[coap response from fdc7:388b:90bb:5a27:5085:7d23:e59c:18e8]
Responded with JSON: {"id":1}
```

### Other Steps to Reproduce

press button on joiner to send via put a string, leader prints address. Copy address and paste back into leader serial terminal to populate the destination. Press button on leader to try and activate the GET for the joiner

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

97Cweb, 2024-09-14:
found it finally, it is coap set to set the response that a get command gets. Could this please be added to documentation/instructions somewhere?

torntrousers, 2024-09-14:
would you show the fixed code?

97Cweb, 2024-09-14:
```
#include "OThreadCLI.h"
#include "OThreadCLI_Util.h"

#define OT_CHANNEL            "24"
#define OT_NETWORK_KEY        "00112233445566778899aabbccddeeff"
#define OT_MCAST_ADDR         "ff05::abcd"
#define OT_COAP_RESOURCE      "json"
#define USER_BUTTON           9

String joinerAddress = "";

const char *otSetupLeader[] = {
  // -- clear/disable all
  // stop CoAP
  "coap", "stop",
  // stop Thread
  "thread", "stop",
  // stop the interface
  "ifconfig", "down",
  // clear the dataset
  "dataset", "clear",
  // -- set dataset
  // create a new complete dataset with random data
  "dataset", "init new",
  // set the channel
  "dataset channel", OT_CHANNEL,
  // set the network key
  "dataset networkkey", OT_NETWORK_KEY,
  // commit the dataset
  "dataset", "commit active",
  // -- network start
  // start the interface
  "ifconfig", "up",
  // start the Thread network
  "thread", "start"
};

const char *otCoapHub[] = {
  // -- create a multicast IPv6 Address for this device
  "ipmaddr add", OT_MCAST_ADDR,
  // -- start and create a CoAP resource
  // start CoAP as server
  "coap", "start",
  // create a CoAP resource
  "coap resource", OT_COAP_RESOURCE,
  // set the CoAP resource initial value
  "coap set", "",
};

bool otDeviceSetup(const char **otSetupCmds, uint8_t nCmds1, const char **otCoapCmds, uint8_t nCmds2, ot_device_role_t expectedRole) {
  Serial.println("Starting OpenThread.");
  Serial.println("Running as HUB");
  uint8_t i;
  for (i = 0; i < nCmds1; i++) {
    if (!otExecCommand(otSetupCmds[i * 2], otSetupCmds[i * 2 + 1])) {
      break;
    }
  }
  if (i != nCmds1) {
    log_e("Sorry, OpenThread Network setup failed!");
    rgbLedWrite(RGB_BUILTIN, 255, 0, 0);  // RED ... failed!
    return false;
  }
  Serial.println("OpenThread started.\r\nWaiting for activating correct Device Role.");
  // wait for the expected Device Role to start
  uint8_t tries = 24;  // 24 x 2.5 sec = 1 min
  while (tries && otGetDeviceRole() != expectedRole) {
    Serial.print(".");
    delay(2500);
[... 331 lines cut ...]
}

void setup() {

  train["id"] = 1;
  serializeJson(train,serializedJson);
  Serial.begin(115200);
  // LED starts RED, indicating not connected to Thread network.
  rgbLedWrite(RGB_BUILTIN, 0, 255, 0);
  OThreadCLI.begin(false);     // No AutoStart is necessary
  OThreadCLI.setTimeout(250);  // waits 250ms for the OpenThread CLI response
  setupNode();
  // LED goes and keeps Blue when all is ready and Red when failed.
  pinMode(USER_BUTTON, INPUT_PULLUP);  // C6/H2 User Button

  
}

void loop() {
  checkUserButton();
  //otCOAPListenForGET();  // Continuously listen for GET requests
  delay(10);
}

```

SuGlider, 2024-10-08:
Great! As for the size of the CoAP message, which is limited to 16 bytes, please read https://github.com/espressif/arduino-esp32/issues/10350#issuecomment-2398522815

SuGlider, 2024-11-04:
Ok. I'll close this issue. Feel free to open it again, if necessary.

closed 2024-11-04 by SuGlider