espressif/arduino-esp32 #10349: Problem with Board Manager version 3.0.4 on ESP32.

ADEK007, 2024-09-14:
### Board

ESP32

### Device Description

DOIT ESP32 DEVKIT V1
[image]

### Hardware Configuration

No

### Version

v3.0.4

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Code is ok. But for the version 3.0.4 the ledc is not working properly. After installing  version 2.0.15 the problem does not occur. 
[image]

### Sketch

```cpp
const int freq = 90000;    // 90 kHz frequency
const int pwmChannel = 0;  // PWM channel
const int resolution = 8;  // 8-bit resolution
const int controlPin = 25; // Pin connected to TL494
int dutyCycle = 128;       // Initial duty cycle (50%)

void setup() {
  // Start the Serial Monitor at 115200 baud rate
  Serial.begin(115200);

  // Setup PWM channel with 90 kHz frequency and 8-bit resolution
  ledcSetup(pwmChannel, freq, resolution);

  // Attach controlPin to the PWM channel
  ledcAttachPin(controlPin, pwmChannel);

  // Set initial duty cycle
  bool ledcWrite(pwmChannel, dutyCycle);

  // Print initial configuration to Serial Monitor
  Serial.println("PWM Control by ESP32:");
  Serial.print("Frequency: ");
  Serial.print(freq);
  Serial.println(" Hz");

  Serial.print("Duty Cycle: ");
  Serial.print((dutyCycle * 100) / 255); // Convert to percentage
  Serial.println(" %");
}

void loop() {
  // Optionally adjust the duty cycle here if needed
  dutyCycle = random(0, 256);  // Random duty cycle for demonstration (0-255)

  // Write new duty cycle to PWM
  ledcWrite(pwmChannel, dutyCycle);

  // Print the updated duty cycle to the Serial Monitor
  Serial.print("Updated Duty Cycle: ");
  Serial.print((dutyCycle * 100) / 255); // Convert to percentage
  Serial.println(" %");

  // Add a short delay before the next update
  delay(1000);  // Update every second
}
```

### Debug Message

```plain
D:\Project Backups\oscilation_check\tl494_esp32\tl494_esp32.ino: In function 'void setup()':
D:\Project Backups\oscilation_check\tl494_esp32\tl494_esp32.ino:12:3: error: 'ledcSetup' was not declared in this scope
   12 |   ledcSetup(pwmChannel, freq, resolution);
      |   ^~~~~~~~~
D:\Project Backups\oscilation_check\tl494_esp32\tl494_esp32.ino:15:3: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
   15 |   ledcAttachPin(controlPin, pwmChannel);
      |   ^~~~~~~~~~~~~
      |   ledcAttach
D:\Project Backups\oscilation_check\tl494_esp32\tl494_esp32.ino:18:39: error: expression list treated as compound expression in initializer [-fpermissive]
   18 |   bool ledcWrite(pwmChannel, dutyCycle);
      |                                       ^

exit status 1

Compilation error: 'ledcSetup' was not declared in this scope
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-09-14:
v3 is a major version change, so the API has changed
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogOut/LEDCSoftwareFade/LEDCSoftwareFade.ino

lbernstone, 2024-09-15:
Please change title to problem with ledc (instead of board manager).

VojtechBartoska, 2024-09-16:
Hello @ADEK007, did you solve your problem with migration guide? can I consider this ticket as answered?

codemasters1, 2024-10-17:
Compile it with V<2.0.17  it's will work .

closed 2025-03-12 by Jason2866