espressif/arduino-esp32 #10350: CoAP set limited to 15 characters

97Cweb, 2024-09-15:
### Board

ESP32-C6 to ESP32-C6

### Device Description

ESP32-C6 Waveshare

### Hardware Configuration

no

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Trying to set "coap set 0123456789ABCDE" works but "coap set 0123456789ABCDEF" fails with Error 7: InvalidArgs

### Sketch

```cpp
/*
 * OpenThread.begin(false) will not automatically start a node in a Thread Network
 * The user will need to start it manually using the OpenThread CLI commands
 * Use the Serial Monitor to interact with the OpenThread CLI
 *
 * Type 'help' for a list of commands.
 * Documentation: https://openthread.io/reference/cli/commands
 *
 */

#include "OThreadCLI.h"

void setup() {
  Serial.begin(115200);
  OThreadCLI.begin(false);  // No AutoStart - fresh start
  Serial.println("OpenThread CLI started - type 'help' for a list of commands.");
  OThreadCLI.startConsole(Serial);
}

void loop() {}
```

### Debug Message

```plain
ot> coap set "00000000000000
Done

ot> coap set 0123456789ABCDE
Done

ot> coap set 0123456789ABCDEF
Error 7: InvalidArgs

ot>
```

### Other Steps to Reproduce

Follow the below cli commands to cause the error

coap stop
thread stop
ifconfig down
dataset clear
dataset channel 24
dataset networkkey 00112233445566778899aabbccddeeff
dataset commit active
ifconfig up
thread start
coap start
coap resource json
coap set 0123456789ABCDE
coap set 0123456789ABCDEF

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-09-15:
This seems very likely a limitation of the source.  You will be much better served by asking upstream at https://github.com/openthread/openthread/issues

97Cweb, 2024-09-15:
> This seems very likely a limitation of the source, seeing as how it is trapped and has a useful error. You will be much better served by asking upstream at https://github.com/openthread/openthread/issues

Thank you for the direction, I am crosslinking the post here so it can be tracked too.

https://github.com/openthread/openthread/issues/10712

Hopefully this gets resolved soon

VojtechBartoska, 2024-09-16:
Any ideas regarding this @SuGlider?

97Cweb, 2024-09-27:
I cannot find the point to overwrite the buffersize. Can you point me in the right direction?

lbernstone, 2024-09-27:
This code is compiled into the sdk libraries.  You would need to [compile your own libraries](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html) and modify the code in the esp-idf/openthread with that larger buffer.  In order words, this is not a simple one line change.

97Cweb, 2024-09-27:
Aw shoot. That is beyond me at least for now. Thank you for telling me this so I dont have to pull more hair out trying to find the line. Hopefully something is found soon to make it more flexible.

SuGlider, 2024-10-08:
OpenThread CLI sets maximum message size to 16 bytes and URI to 32 bytes.
It can't be changed, except by rebuilding the library with new values.
This is explained here: https://github.com/orgs/openthread/discussions/10713#discussioncomment-10654886

There are 2 possible solutions for sending a message bigger than 15 bytes (last byte is always a '\0')

1) spliting the original message into a few smaller messages.
Let say that the application needs to send 100 bytes.
It is possible to send 10 messages with a valid payload of 10 bytes each.
The message frame may have a format such as:
| msg# :: 1 byte | total number of messages for the whole payload :: 1 byte | len :: 1 byte | payload :: len bytes | CRC8 :: 1 byte | 
Total Frame Length = 1 + 1 + 1 + 11(max) + 1 = 15 bytes maximum, given that the payload should not exceed 11 bytes.
In the case of sending 100 bytes, it would be necessary to break it into 10 payloads of 10 bytes each.
Put the frame together and send each one, waiting for the ACK confirming that the other side has received it.
Example:  
  ``` 
  | 0x00 | 0x0A | 0x0A | payload_01 | CRC-8 calculation |
| 0x02 | 0x0A | 0x0A | payload_02 | CRC-8 calculation |
....
| 0x09 | 0x0A | 0x0A | payload_10 | CRC-8 calculation |
```

2) Another way to deal with  OpentThread/CoAP + Arduino would be by using OThread CoAP native API, instead o f using CLI.
This could be done using the CLI CoAP example from https://github.com/openthread/openthread/blob/d7810fa04d46f77eb5b78af81ad0bdb59bfcd436/src/cli/cli_coap.cpp

SuGlider, 2024-10-08:
@97Cweb - another solution is not using JSON. It is too expensive in term of bytes used versus real information.
The application could make it binary.
Example:
| Train# :: 1 byte | CMD ID :: 1 byte | Arg#1 : 1 byte | Arg#2 : 1 byte |  ==> total of 4 bytes.

97Cweb, 2024-10-08:
Thanks for the idea of the binary options. I'll look more into that, I do think I'll need more info, but we'll see. I do hope that one day it is not a baked in constraint though, as I can put much larger values, but cannot get them back

SuGlider, 2024-10-09:
It may be interesting considering other protocols, such as WiFi Mesh or Mesh BLE.

-------------------------------------
ESP-WIFI-MESH is a networking protocol built atop the Wi-Fi protocol. ESP-WIFI-MESH allows numerous devices (henceforth referred to as nodes) spread over a large physical area (both indoors and outdoors) to be interconnected under a single WLAN (Wireless Local-Area Network). ESP-WIFI-MESH is self-organizing and self-healing meaning the network can be built and maintained autonomously.

It needs a WiFi Router...
https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-guides/esp-wifi-mesh.html
-------------------------------------
Bluetooth® mesh networking enables many-to-many (m:m) device communications and is optimized for creating large-scale device networks.

Devices may relay data to other devices not in direct radio range of the originating device. In this way, mesh networks can span very large physical areas and contain large numbers of devices. It is ideally suited for building automation, sensor networks, and other IoT solutions where tens, hundreds, or thousands of devices need to reliably and securely communicate with one another.

Bluetooth mesh is not a wireless communications technology, but a networking technology. This technology is dependent upon Bluetooth Low Energy (BLE) - a wireless communications protocol stack.

No external router or coordinator is necessary
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-ble-mesh/ble-mesh-index.html
-------------------------------------

97Cweb, 2024-10-09:
Thank you for the idea of bluetooth mesh. I'll look into it more, as that
seems to be similar to what I want to achieve

SuGlider, 2024-11-04:
Ok. I'll close this issue. Feel free to open it again, if necessary.

closed 2024-11-04 by SuGlider