espressif/arduino-esp32 #10352: ESP32 C3 WIFIclientsecure  .connect() crashes

aeonSolutions, 2024-09-16:
### Board

ESP32 C3

### Device Description

custom board with a ESP32 C3 MCU

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.x

### Operating System

windows 11

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

WIFIclientsecure  .connect() crashes 

### Sketch

```cpp
#include <WiFiClientSecure.h>    // For secure HTTPS connection
WiFiClientSecure client;

String getFileSHA() {
  mySerial.println("test 0");

  if (WiFi.status() != WL_CONNECTED) {
    mySerial.println("WIFI not available. skipping upload.");
    return "";
  }
  
  mySerial.println("test 1");

  String sha = "";  
  String url = "/repos/" + String(githubUser) + "/" + String(githubRepo) + "/contents/" + String(githubFilePath) + "?ref=" + String(githubBranch);

  mySerial.println("test 2");

  // Make sure to trust the GitHub SSL certificate
  client.setInsecure();  // Bypasses SSL certificate verification
  
  mySerial.println("test 3");
  // Connect to GitHub
  bool check = client.connect("api.github.com", 443);
  mySerial.println("test 4");
  
  if ( check) {
    mySerial.println("Connected to GitHub");

    // Send the GET request
    client.println("GET " + url + " HTTP/1.1");
    client.println("Host: api.github.com");
    client.print("Authorization: token ");
    client.println( String(githubToken) );                 // Send the GitHub PAT for authorization
    client.println("User-Agent: ESP32");
    client.println("Connection: close");
    client.println();

    // Wait for the response
    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        break;  // End of headers
      }
    }

    // Read the response body
    String jsonResponse = client.readString();
    mySerial.println("Received JSON response:");

    // Print the JSON response for debugging
    mySerial.println(jsonResponse);

    // Parse the JSON response
    StaticJsonDocument<4000> doc;
    DeserializationError error = deserializeJson(doc, jsonResponse);
    if (error) {
      mySerial.print("Failed to parse JSON: ");
      mySerial.println(error.c_str());
      return "";
    }

    // Get the SHA of the file
    const char* sha = doc["sha"];
    if (sha) {
      mySerial.print("SHA: ");
      mySerial.println(sha);
      return sha;
    } else {
      mySerial.println("SHA not found in the response.");
      return "";
    }
  } else {
    mySerial.println("Connection to GitHub failed");
    BlinkLED(LED_GREEN, 5);
  }
  return "";
}
```

### Debug Message

```plain
get the file SHA Blob from GitHub if it exists ....
13:44:37.901 -> test 0
13:44:37.901 -> test 1
13:44:37.901 -> test 2
13:44:37.901 -> test 3
13:44:39.328 -> ESP-ROM:esp32c3-api1-20210207
13:44:39.328 -> Build:Feb  7 2021
13:44:39.328 -> rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
13:44:39.328 -> Saved PC:0x40048b82
13:44:39.328 -> SPIWP:0xee
13:44:39.328 -> mode:DIO, clock div:1
13:44:39.328 -> load:0x3fcd5820,len:0x1058
13:44:39.328 -> load:0x403cc710,len:0x8a8
13:44:39.328 -> load:0x403ce710,len:0x2fa0
13:44:39.328 -> entry 0x403cc710
```

### Other Steps to Reproduce

on the first boot of the firmware and on subsequent reboots 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-09-16:
please add minimal example that we can compile and run. We do not have reports for C3 having issues thus far.

lbernstone, 2024-09-16:
Cannot reproduce.  Perhaps you are out of memory.
https://wokwi.com/projects/409203958185858049

aeonSolutions, 2024-09-16:
memory available just before calling  _.connect()_  is 207,696 bytes

In the meantime i've uploaded the full firmware code here:
https://github.com/aeonSolutions/Aeonlabs-useArduino_Hardware_Test_Code/blob/main/github_api_test_upload/github_api_test_upload.ino

you can download and try it for yourself.

aeonSolutions, 2024-09-26:
In the meantime, I've decided to use another C++ library. 
thank you

closed 2024-09-26 by aeonSolutions