espressif/arduino-esp32 #10359: Unable to create static library if any #include "Arduino.h" is used. (IDFGH-13690)

umer-nazir02, 2024-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should make the static library i.e .a file. 

### What is the actual behavior?

Getting error

hello_world/main/main.cpp:1:10: fatal error: Arduino.h: No such file or directory
    1 | #include "Arduino.h"

### Steps to reproduce.

1. Create sample hello_world project

idf.py create-project-from-example "espressif/arduino-esp32^3.0.2:hello_world"

2. Update the CMakeLists.txt to include instruction to make STATIC library

add_library(hello_world STATIC main.cpp)

3. run idf.py build command

### Build or installation Logs.

```plain
Executing action: all (aliases: build)
Running ninja in directory /Users/abc/Documents/Projects/iot/hello_world/build
Executing "ninja all"...
[0/1] Re-running CMake...-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Manifest files have changed, solving dependencies.
NOTICE: Updating lock file at /Users/abc/Documents/Projects/iot/hello_world/dependencies.lock
Deleting 26 unused components
 espressif__esp-dsp
 espressif__json_parser
 espressif__libsodium
 espressif__esp-sr
 espressif__esp_modem
 chmorgan__esp-libhelix-mp3
 espressif__esp-zboss-lib
 espressif__esp_insights
 espressif__esp_secure_cert_mgr
 espressif__qrcode
 espressif__esp-modbus
 espressif__jsmn
 espressif__rmaker_common
[... 40 lines cut ...]
-- Adding linker script /Users/abc/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/abc/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/abc/esp/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script /Users/abc/esp/esp-idf/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /Users/abc/esp/esp-idf/components/bootloader /Users/abc/esp/esp-idf/components/bootloader_support /Users/abc/esp/esp-idf/components/efuse /Users/abc/esp/esp-idf/components/esp_app_format /Users/abc/esp/esp-idf/components/esp_common /Users/abc/esp/esp-idf/components/esp_hw_support /Users/abc/esp/esp-idf/components/esp_rom /Users/abc/esp/esp-idf/components/esp_system /Users/abc/esp/esp-idf/components/esptool_py /Users/abc/esp/esp-idf/components/freertos /Users/abc/esp/esp-idf/components/hal /Users/abc/esp/esp-idf/components/log /Users/abc/esp/esp-idf/components/bootloader/subproject/main /Users/abc/esp/esp-idf/components/bootloader/subproject/components/micro-ecc /Users/abc/esp/esp-idf/components/newlib /Users/abc/esp/esp-idf/components/partition_table /Users/abc/esp/esp-idf/components/soc /Users/abc/esp/esp-idf/components/spi_flash /Users/abc/esp/esp-idf/components/xtensa
-- Configuring done (4.0s)
-- Generating done (0.1s)
-- Build files have been written to: /Users/abc/Documents/Projects/iot/hello_world/build/bootloader
[1/2] cd /Users/abc/Documents/Projects/iot/hello_world/build/.../Documents/Projects/iot/hello_world/build/bootloader/bootloader.binBootloader binary size 0x51c0 bytes. 0x2e40 bytes (36%) free.
[898/918] Building CXX object esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.objFAILED: esp-idf/main/CMakeFiles/__idf_main.dir/main.cpp.obj
/Users/abc/required_idf_tools_path/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -DESP_PLATFORM -DIDF_VER=\"v5.1.4\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUNITY_INCLUDE_CONFIG_H -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/abc/Documents/Projects/iot/hello_world/build/config -I/Users/abc/Documents/Projects/iot/hello_world/main -I/Users/abc/esp/esp-idf/components/newlib/platform_include -I/Users/abc/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/abc/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/abc/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/abc/esp/esp-idf/components/freertos/esp_additions/include -I/Users/abc/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/Users/abc/esp/esp-idf/components/esp_hw_support/include -I/Users/abc/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/abc/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/Users/abc/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/Users/abc/esp/esp-idf/components/heap/include -I/Users/abc/esp/esp-idf/components/log/include -I/Users/abc/esp/esp-idf/components/soc/include -I/Users/abc/esp/esp-idf/components/soc/esp32s3 -I/Users/abc/esp/esp-idf/components/soc/esp32s3/include -I/Users/abc/esp/esp-idf/components/hal/esp32s3/include -I/Users/abc/esp/esp-idf/components/hal/include -I/Users/abc/esp/esp-idf/components/hal/platform_port/include -I/Users/abc/esp/esp-idf/components/esp_rom/include -I/Users/abc/esp/esp-idf/components/esp_rom/include/esp32s3 -I/Users/abc/esp/esp-idf/components/esp_rom/esp32s3 -I/Users/abc/esp/esp-idf/components/esp_common/include -I/Users/abc/esp/esp-idf/components/esp_system/include -I/Users/abc/esp/esp-idf/components/esp_system/port/soc -I/Users/abc/esp/esp-idf/components/esp_system/port/include/private -I/Users/abc/esp/esp-idf/components/xtensa/include -I/Users/abc/esp/esp-idf/components/xtensa/esp32s3/include -I/Users/abc/esp/esp-idf/components/lwip/include -I/Users/abc/esp/esp-idf/components/lwip/include/apps -I/Users/abc/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/abc/esp/esp-idf/components/lwip/lwip/src/include -I/Users/abc/esp/esp-idf/components/lwip/port/include -I/Users/abc/esp/esp-idf/components/lwip/port/freertos/include -I/Users/abc/esp/esp-idf/components/lwip/port/esp32xx/include -I/Users/abc/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/Users/abc/esp/esp-idf/components/esp_ringbuf/include -I/Users/abc/esp/esp-idf/components/efuse/include -I/Users/abc/esp/esp-idf/components/efuse/esp32s3/include -I/Users/abc/esp/esp-idf/components/driver/include -I/Users/abc/esp/esp-idf/components/driver/deprecated -I/Users/abc/esp/esp-idf/components/driver/analog_comparator/include -I/Users/abc/esp/esp-idf/components/driver/dac/include -I/Users/abc/esp/esp-idf/components/driver/gpio/include -I/Users/abc/esp/esp-idf/components/driver/gptimer/include -I/Users/abc/esp/esp-idf/components/driver/i2c/include -I/Users/abc/esp/esp-idf/components/driver/i2s/include -I/Users/abc/esp/esp-idf/components/driver/ledc/include -I/Users/abc/esp/esp-idf/components/driver/mcpwm/include -I/Users/abc/esp/esp-idf/components/driver/parlio/include -I/Users/abc/esp/esp-idf/components/driver/pcnt/include -I/Users/abc/esp/esp-idf/components/driver/rmt/include -I/Users/abc/esp/esp-idf/components/driver/sdio_slave/include -I/Users/abc/esp/esp-idf/components/driver/sdmmc/include -I/Users/abc/esp/esp-idf/components/driver/sigma_delta/include -I/Users/abc/esp/esp-idf/components/driver/spi/include -I/Users/abc/esp/esp-idf/components/driver/temperature_sensor/include -I/Users/abc/esp/esp-idf/components/driver/touch_sensor/include -I/Users/abc/esp/esp-idf/components/driver/twai/include -I/Users/abc/esp/esp-idf/components/driver/uart/include -I/Users/abc/esp/esp-idf/components/driver/usb_serial_jtag/include -I/Users/abc/esp/esp-idf/components/driver/touch_sensor/esp32s3/include -I/Users/abc/esp/esp-idf/components/esp_pm/include -I/Users/abc/esp/esp-idf/components/mbedtls/port/include -I/Users/abc/esp/esp-idf/components/mbedtls/mbedtls/include -I/Users/abc/esp/esp-idf/components/mbedtls/mbedtls/library -I/Users/abc/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/Users/abc/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -I/Users/abc/esp/esp-idf/com [... cut ...]
/Users/abc/Documents/Projects/iot/hello_world/main/main.cpp:1:10: fatal error: Arduino.h: No such file or directory
    1 | #include "Arduino.h"
      |          ^~~~~~~~~~~
compilation terminated.
[908/918] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/manager.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/abc/Documents/Projects/iot/hello_world/build/log/idf_py_stderr_output_54843 and /Users/abc/Documents/Projects/iot/hello_world/build/log/idf_py_stdout_output_54843
```

Jason2866, 2025-03-12:
Not an Arduino issue.

closed 2025-03-12 by Jason2866

umer-nazir02, 2025-03-17:
I resolved this issue by compiling Arduino as a static library as well. By compiling Arduino first and then creating the static library, everything works perfectly.