espressif/arduino-esp32 #10369: Allow custom mac address definition in ETH library

MicSG-dev, 2024-09-23:
### Related area

Mac Address Ethernet W5500 SPI

### Hardware specification

ESP32 DevKitC

### Is your feature request related to a problem?

Currently the MAC address configuration in the ETH library happens internally, so the user cannot decide which MAC address he wants to use.

### Describe the solution you'd like

I want it to be possible to change the MAC address through the begin function of the ETH.h library.

### Describe alternatives you've considered

The solution would be to create a new method for the ETH library. See below: 
</br></br>
[File ETH.h, line 125](https://github.com/espressif/arduino-esp32/blob/7018cd114d00249674567846c9d67fbb3a1240a3/libraries/Ethernet/src/ETH.h#L125-L127):

BEFORE: 
```
#if ETH_SPI_SUPPORTS_CUSTOM
  bool begin(eth_phy_type_t type, int32_t phy_addr, int cs, int irq, int rst, SPIClass &spi, uint8_t spi_freq_mhz = ETH_PHY_SPI_FREQ_MHZ);
#endif
```

AFTER: 
```
#if ETH_SPI_SUPPORTS_CUSTOM
  bool begin(eth_phy_type_t type, int32_t phy_addr, int cs, int irq, int rst, SPIClass &spi, uint8_t spi_freq_mhz = ETH_PHY_SPI_FREQ_MHZ);
  bool begin(eth_phy_type_t type, int32_t phy_addr, int cs, int irq, int rst, SPIClass &spi, uint8_t *mac_addr, uint8_t spi_freq_mhz = ETH_PHY_SPI_FREQ_MHZ);
#endif
```
</br></br>

[File ETH.cpp, line 810](https://github.com/espressif/arduino-esp32/blob/7018cd114d00249674567846c9d67fbb3a1240a3/libraries/Ethernet/src/ETH.cpp#L810-L815): 

BEFORE:
```
#if ETH_SPI_SUPPORTS_CUSTOM
bool ETHClass::begin(eth_phy_type_t type, int32_t phy_addr, int cs, int irq, int rst, SPIClass &spi, uint8_t spi_freq_mhz) {

  return beginSPI(type, phy_addr, nullptr, cs, irq, rst, &spi, -1, -1, -1, SPI2_HOST, spi_freq_mhz);
}
#endif
```

DEPOIS:
```
#if ETH_SPI_SUPPORTS_CUSTOM
bool ETHClass::begin(eth_phy_type_t type, int32_t phy_addr, int cs, int irq, int rst, SPIClass &spi, uint8_t spi_freq_mhz) {

  return beginSPI(type, phy_addr, nullptr, cs, irq, rst, &spi, -1, -1, -1, SPI2_HOST, spi_freq_mhz);
}

bool ETHClass::begin(eth_phy_type_t type, int32_t phy_addr, int cs, int irq, int rst, SPIClass &spi, uint8_t *mac_addr, uint8_t spi_freq_mhz) {

  return beginSPI(type, phy_addr, mac_addr, cs, irq, rst, &spi, -1, -1, -1, SPI2_HOST, spi_freq_mhz);
}
#endif
```

### Additional context

The usage of this new library method would be:
```
void setup() {
  Serial.begin(115200);
  Network.onEvent(onEvent);

  uint8_t mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

  SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI, mac);
}
```

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

JAndrassy, 2024-09-24:
you can use my EthernetESP32 library

as-iotex, 2024-10-21:
Hey @JAndrassy 

I am using the ArduinoOTA bundled library, and I would like to use a custom MAC address or static IP. I am using the W5500 ethernet module 
Does your custom EthernetESP32 library work with ArduinoOTA (which uses ETH.cpp) ?

If your library doesn't work with ArduinoOTA, could you point me to what you changed in order to set it. I could try and hack around ETH.cpp to port this particular change.

@me-no-dev tagging you as well in case you can help

MicSG-dev, 2024-10-21:
> Ei@JAndrassy
> 
> Estou usando a biblioteca empacotada ArduinoOTA e gostaria de usar um endereço MAC personalizado ou IP estático. Estou usando o módulo ethernet W5500. Sua biblioteca EthernetESP32 personalizada funciona com ArduinoOTA (que usa ETH.cpp)?
> 
> Se sua biblioteca não funcionar com ArduinoOTA, você poderia me indicar o que você alterou para configurá-la. Eu poderia tentar hackear o ETH.cpp para portar essa alteração em particular.
> 
> @me-no-devmarcando você também caso possa ajudar

Replace the ArduinoOTA library with the Update.h library, which is compatible with ETH.h (see a basic example [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/OTAWebUpdater/OTAWebUpdater.ino), but if you are going to use a webserver to perform the update, I suggest using it together with the [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer/) library).

To set a static IP address, use the config method (from the ETH library) with the following parameters:

- local_ip: the static IP address;
- gateway: the default gateway address;
- subnet: the subnet mask;
- dns1: the first DNS server;
- dns2: the second DNS server (optional);
- dns3: the third DNS server (optional).

Usage example:
```
#include <ETH.h>

void setup() {
  // Inicializa a interface Ethernet
  ETH.begin();

  // Defining the parameters for the static IP
  IPAddress local_ip(192, 168, 1, 100);    // static IP
  IPAddress gateway(192, 168, 1, 1);       // Gateway
  IPAddress subnet(255, 255, 255, 0);      // Subnet mask
  IPAddress dns1(8, 8, 8, 8);              // First DNS server (Google DNS)
  IPAddress dns2(8, 8, 4, 4);              // Second DNS server (Google DNS)
  IPAddress dns3(1, 1, 1, 1);              // Third DNS server (Cloudflare DNS, optional)

 // Setting the static IP
  ETH.config(local_ip, gateway, subnet, dns1, dns2, dns3);

 // Your code here
}

void loop() {
 // Main code
}

```

To set a custom MAC address, I'm using a library, which I forked from this main one, that allows you to set a custom MAC address: [github.com/MicSG-dev/ETH-Mac](https://github.com/MicSG-dev/ETH-Mac). Usage example:
```
uint8_t mac[6] = {0x00, 0x1A, 0x2B, 0x3C, 0x4D, 0x5D};
ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI, mac)
```

JAndrassy, 2024-10-22:
@as-iotex  https://github.com/Networking-for-Arduino/EthernetESP32/blob/master/examples/BasicOTA/BasicOTA.ino

as-iotex, 2024-10-22:
@JAndrassy I tried your library and I am seeing this conflicting declaration error:
```
In file included from /home/user/Arduino/libraries/EthernetESP32/src/EthernetESP32.h:4,
                 from /home/user/projects/smartpadel/smartpadelFW/src/ProdinoOTA.cpp:51:
/home/user/Arduino/libraries/EthernetESP32/src/utility/EMACDriver.h:32:3: error: 'ETH_PHY_LAN8720' conflicts with a previous declaration
   32 |   ETH_PHY_LAN8720,
      |   ^~~~~~~~~~~~~~~
In file included from /home/user/projects/smartpadel/smartpadelFW/src/ProdinoOTA.cpp:1:
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Ethernet/src/ETH.h:101:3: note: previous declaration 'eth_phy_type_t ETH_PHY_LAN8720'
  101 |   ETH_PHY_LAN8720,
      |   ^~~~~~~~~~~~~~~
/home/user/Arduino/libraries/EthernetESP32/src/utility/EMACDriver.h:33:3: error: 'ETH_PHY_TLK110' conflicts with a previous declaration
   33 |   ETH_PHY_TLK110,
      |   ^~~~~~~~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Ethernet/src/ETH.h:102:3: note: previous declaration 'eth_phy_type_t ETH_PHY_TLK110'
  102 |   ETH_PHY_TLK110,
      |   ^~~~~~~~~~~~~~
/home/user/Arduino/libraries/EthernetESP32/src/utility/EMACDriver.h:34:3: error: 'ETH_PHY_RTL8201' conflicts with a previous declaration
   34 |   ETH_PHY_RTL8201,
      |   ^~~~~~~~~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Ethernet/src/ETH.h:103:3: note: previous declaration 'eth_phy_type_t ETH_PHY_RTL8201'
  103 |   ETH_PHY_RTL8201,
      |   ^~~~~~~~~~~~~~~
/home/user/Arduino/libraries/EthernetESP32/src/utility/EMACDriver.h:35:3: error: 'ETH_PHY_DP83848' conflicts with a previous declaration
   35 |   ETH_PHY_DP83848,
      |   ^~~~~~~~~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Ethernet/src/ETH.h:104:3: note: previous declaration 'eth_phy_type_t ETH_PHY_DP83848'
  104 |   ETH_PHY_DP83848,
      |   ^~~~~~~~~~~~~~~
/home/user/Arduino/libraries/EthernetESP32/src/utility/EMACDriver.h:37:3: error: 'ETH_PHY_MAX' conflicts with a previous declaration
   37 |   ETH_PHY_MAX
      |   ^~~~~~~~~~~
/home/user/.arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/Ethernet/src/ETH.h:117:3: note: previous declaration 'eth_phy_type_t ETH_PHY_MAX'
  117 |   ETH_PHY_MAX
      |   ^~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

Am I doing something wrong or is this a bug in the library?

JAndrassy, 2024-10-22:
don't use both ETH.h and EthernetESP32.h in same sketch

as-iotex, 2024-10-23:
Thanks very much. I can confirm your library works like a charm

Parsaabasi, 2025-03-11:
Hello, 

This issue seems resolved. I close it for now. In case the issue persists, please feel free to reopen it. 

Thanks

JAndrassy, 2025-03-11:
@Parsaabasi OP used my EthernetESP32 library instead of the ETH library. The ETH library still doesn't support custom MAC address setting

Jason2866, 2025-03-13:
Not planned feature to spoof a random MAC address.

closed 2025-03-13 by Jason2866