espressif/arduino-esp32 #10384: riscv32-esp-elf-g++ extremely slow in Windows compare to VirtualBox VM Ubuntu 20 (Same hardware)

sywong2000, 2024-09-27:
### Board

ESP32C3

### Device Description

https://product.abrobot.club/ABrobot%E4%BA%A7%E5%93%81%E8%B5%84%E6%96%99%E4%B8%AD%E5%BF%83/ESP32-C3%E5%BC%80%E5%8F%91%E6%9D%BF%E8%B5%84%E6%96%99

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

When building project, C:\Users\XXXXX\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302/bin/riscv32-esp-elf-g++ is extremely slow and spawn cc1plus.exe which consumes a lot of CPU. 

Example compiling blink.ino, it takes may be up to 10 mins on Windows.
Comparatively, under the same hardware, using a VirtualBox Guest Ubuntu 20, it only takes 1-2 minutes.

### Sketch

```cpp
Blink.ino
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-09-27:
@SuGlider PTAL

Jason2866, 2024-09-28:
It is probably Windows defender. At all involved folders to exclude from virus scan.

TD-er, 2024-10-03:
I have not tried using Virtualbox, but I do now use WSL along with VS-code on Windows as it is a lot faster (3x - 5x faster) to build my project.
As long as you build your project on a Linux file system and not on a mounted Windows file system.

My theory is that Windows does trigger some check on each opened file which does serialize the entire build process.
Even if you have your virusscanner disabled and/or your build folders excluded from any scanning.

The gain in build time may differ from system to system, but there is for sure a huge performance gain possible on exactly the same hardware when building in a Linux environment on a Linux filesystem.

N.B. the difference when building ESP8266 builds is much smaller.