espressif/arduino-esp32 #10386: Core versions 3.0.0 and newer use 14% more flash memory after compiling for LOLIN ESP32-S2 Mini

jhsa, 2024-09-28:
### Board

LOLIN S2 MINI

### Device Description

Rotary Encoder, OLED display, External EEprom, Buttons..

### Hardware Configuration

i2c - SCL on GPIO 35, SDA on GPIO 33
SPI - MOSI 11, MISO 9, and SCK 7
Rotary Encoder is connected to GPIO 2 and 4.

### Version

v3.0.4

### IDE Name

Arduino IDE 2.3.3

### Operating System

Windows 10

### Flash frequency

240MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I tried to compile some code I wrote a few months ago and it compiled , but using 14% more flash memory. The code should  use 66% of flash and now it compiles at 80%. also, when I flash the code to my device, the Encoder only work in General Settings, once it starts running through the main loop, the rotary encoder stops working. If I downgrade the ESP core to version 2.0.17, everything works fine again and the code compiles at 66% flash memory usage.
The same happens when I tried some library example, like for example the WifiManager. With the newer versions of the ESP core, the example "OnDemandConfigPortal" uses 80% of flash memory, and it only uses 74% when compiled with version 2.0.17.

### Sketch

```cpp
here is the WiFi Manager exaample code:

/**
 * OnDemandConfigPortal.ino
 * example of running the configPortal AP manually, independantly from the captiveportal
 * trigger pin will start a configPortal AP for 120 seconds then turn it off.
 * 
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 0

int timeout = 120; // seconds to run for

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
}

void loop() {
  // is configuration portal requested?
  if ( digitalRead(TRIGGER_PIN) == LOW) {
    WiFiManager wm;    

    //reset settings - for testing
    //wm.resetSettings();
  
    // set configportal timeout
    wm.setConfigPortalTimeout(timeout);

    if (!wm.startConfigPortal("OnDemandAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.restart();
      delay(5000);
    }

    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");

  }

  // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
None.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-09-28:
Yes, frameworks rarely get smaller as they accumulate more cruft.
Libraries may need to be [updated to work in v3.x](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html) as there are breaking changes in the APIs.

jhsa, 2024-09-28:
> Yes, frameworks rarely get smaller as they accumulate more cruft. Libraries may need to be [updated to work in v3.x](https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html) as there are breaking changes in the APIs.

Thanks.. But the IDE says that all the libraries are updated.

me-no-dev, 2024-09-30:
majority of the size difference comes from the extra cyphers that need to be supported for newer SSL and WiFi authentication. We are looking into ways to make that optional, but do not have a concrete plan or way to provide that yet.

Rob58329, 2024-10-20:
This is a similar issue for the ESP32C3 with (for example) the standard "examples/wifi/SimpleWiFiServer" sketch which gives:

ESP_ARDUINO_VERSION v2.0.6
Sketch uses 694996 bytes (53%) of program storage space. Maximum is 1310720 bytes.	
Global variables use 39268 bytes (11%) of dynamic memory, leaving 288412 bytes for local variables. Maximum is 327680 bytes.	

ESP_ARDUINO_VERSION v3.0.5
Sketch uses 931690 bytes (71%) of program storage space. Maximum is 1310720 bytes.	
Global variables use 35460 bytes (10%) of dynamic memory, leaving 292220 bytes for local variables. Maximum is 327680 bytes.	

That's an increase of 236,694 bytes or about 34% of the original sketch size (and no SSL for this, just Wifi).

me-no-dev, 2024-10-21:
WiFi is enough to increase the size, because it now supports more security models. WPA3, etc. We looked into providing an options with less security supported, but that ended up being not possible for a few reasons.