espressif/arduino-esp32 #10391: Arduino WiFi coexistence with esp-matter

HamzaHajeir, 2024-09-30:
### Board

ESP32-S3-DevKitC-1-N8R2

### Device Description

.

### Hardware Configuration

..

### Version

v3.0.4

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

.

### Description

I'm building a custom project which coexists esp-matter and Arduino in one project. 

Such that the WiFi management is went through Arduino library (WiFi), and to inform CHIP's core that the application controls the WiFi. through:
```cpp
auto &connMgr = chip::DeviceLayer::ConnectivityMgrImpl();
connMgr.SetWiFiStationMode(chip::DeviceLayer::ConnectivityManager::WiFiStationMode::kWiFiStationMode_ApplicationControlled);
connMgr.SetWiFiAPMode(chip::DeviceLayer::ConnectivityManager::WiFiAPMode::kWiFiAPMode_ApplicationControlled);
```

All up to here is okay, but when engaging with `WiFi.enableAP(true)` API an error happens, which is a result of incompatibility with esp-matter's code regarding WiFi. 

Both stacks initializes the netif`'s ([Arduino](https://github.com/espressif/arduino-esp32/blob/e403f0b4815643ccd2d98f62b7accf5a8108f2cd/libraries/WiFi/src/WiFiGeneric.cpp#L261-L266))(esp-matter[[1](https://github.com/espressif/esp-matter/blob/9350d9d5f948d3b7c61c8659c4d6990d0ff00ea4/components/esp_matter/esp_matter_core.cpp#L996-L1001)][[2]([url](https://github.com/espressif/connectedhomeip/blob/17b1a38e909e7874593bcb87c31be03a5866f1d4/src/platform/ESP32/ESP32Utils.cpp#L323-L343))]). While there should be exactly one netif initialized, the program crashes in the second call of.

A suggestion of using the name keys, and the same pattern of code in esp-matter's `start()`, such that to ask for the netif by name, and if not existed to create it:

```cpp
      if ((esp_netifs[ESP_IF_WIFI_AP] = esp_netif_get_handle_from_ifkey("WIFI_AP_DEF")) == NULL) {
        esp_netifs[ESP_IF_WIFI_AP] = esp_netif_create_default_wifi_ap();
      }
      if ((esp_netifs[ESP_IF_WIFI_AP] = esp_netif_get_handle_from_ifkey("WIFI_STA_DEF")) == NULL) {
        esp_netifs[ESP_IF_WIFI_STA] = esp_netif_create_default_wifi_sta();
      }
```
Which worked well in my setup. 

### Sketch

```cpp
.
```

### Debug Message

```plain
E (4666) esp_netif_lwip: esp_netif_new_api: Failed to configure netif with config=0x3fccabf0 (config or if_key is NULL or duplicate key)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-09-30:
you are a bit ahead of the curve here. We do not guarantee that this will ever work, because matter manages WiFi on it's own. Meaning interfaces, events and so on. Let's wait for the work that will be done for v3.1 first

HamzaHajeir, 2024-09-30:
> you are a bit ahead of the curve here. We do not guarantee that this will ever work, because matter manages WiFi on it's own. Meaning interfaces, events and so on. Let's wait for the work that will be done for v3.1 first

I see that point, but its management of WiFi is overridable by the code I pasted. 

What's not so following is esp_matter library which does not check against the mode in its interference. Seems its interference stops at `esp_matter::start()`

Jason2866, 2024-10-01:
> We do not guarantee that this will ever work

No more to say currently

HamzaHajeir, 2024-10-01:
The good news that I've a primarily working code that coexists between esp-matter and arduino.

The provided fix/enhancement is the only primary issue I've met. 

Other issue appeared because of `CONFIG_LWIP_TCPIP_CORE_LOCKING` -and- `CONFIG_LWIP_CHECK_THREAD_SAFETY` checks exposed them. I'll provide a fix regarding that for `AsyncUDP.cpp` that calls raw LwIP APIs without performing any locking. The fix will help primarily arduino-as-a-idf-component users

Until https://github.com/espressif/esp32-arduino-lib-builder/pull/186 gets merged (and ofcourse the fix), direct Arduino users will be helped with.

HamzaHajeir, 2024-10-02:
The close was most probably a misclick.

SuGlider, 2025-02-13:
@HamzaHajeir - Please review this issue given the current Arduino Matter Library. #7432 
https://github.com/espressif/arduino-esp32/tree/master/libraries/Matter

At this time, Matter relies on Arduino WiFi Library for connectivity. 

There is a Matter example that uses WiFi AP or BLE for provissioning the WiFi Network using WiFi Prov Library.
https://github.com/espressif/arduino-esp32/blob/master/libraries/Matter/examples/WiFiProvWithinMatter/WiFiProvWithinMatter.ino

Let me know if we can consider this issue solved.

HamzaHajeir, 2025-03-05:
Hi @SuGlider, I've done a primary review of the issue, and I am not sure why it's not appearing at your example, as `CHIP_DEVICE_CONFIG_ENABLE_WIFI` is defined by default (As `CHIP_DEVICE_CONFIG_ENABLE_WIFI_STATION` is [defaulted to y](https://github.com/project-chip/connectedhomeip/blob/7dfd606a3f88125ef7fa4482b20a9e9be4c72318/config/esp32/components/chip/Kconfig#L555-L559) -> [[1](https://github.com/project-chip/connectedhomeip/blob/7dfd606a3f88125ef7fa4482b20a9e9be4c72318/src/platform/ESP32/CHIPPlatformConfig.h#L84-L88)] -> [[2](https://github.com/project-chip/connectedhomeip/blob/7dfd606a3f88125ef7fa4482b20a9e9be4c72318/src/platform/ESP32/CHIPDevicePlatformConfig.h#L78C76-L78C114)]).

Therefore both Arduino WiFi and CHIP's calls `esp_netif_create_default_wifi_ap()`/`esp_netif_create_default_wifi_sta()`, with a difference that Arduino does not check for an existent one, which I added in my post.

CHIP's one adds the [necessary guards](https://github.com/project-chip/connectedhomeip/blob/7dfd606a3f88125ef7fa4482b20a9e9be4c72318/src/platform/ESP32/ESP32Utils.cpp#L323-L344) to prevent duplicate call when it becomes second.

What I see that adding these guards is the wise decision - anyway -, what do you think?

BTW: I'm not sure why not adding `kWiFiStationMode_ApplicationControlled` would make your example/library stable, as the CHIP also reacts to the network events if not.