espressif/arduino-esp32 #10396: BLEServer::updateConnParams works intermittently

jefflongo, 2024-10-01:
### Board

ESP32-C3-DevKitM-1

### Device Description

Plain module

### Hardware Configuration

No

### Version

v3.0.4

### IDE Name

VSCode

### Operating System

Ubuntu 24.04

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I request the minimal connection parameters (7.5ms interval, 0 latency) upon connection. I have an Android app which connects and communicates frequently with the ESP32. In the main loop, I periodically print the connection parameters in the main loop. I expect to see the interval be set to 6, latency set to 0, timeout set to 100. Instead, I see interval set to 9, latency set to 0, and timeout set to 100 when successful. Around half the time, I instead see interval set to 40, latency set to 0, timeout set to 500 (which is the default before updating the connection parameters).

### Sketch

```cpp
Set the connection parameters on connect

class ServerCallbacks : public BLEServerCallbacks {
    void onConnect(BLEServer* server, esp_ble_gatts_cb_param_t* param) {
        // set the connection interval to the minimum of 7.5ms (6 * 1.25ms)
        server->updateConnParams(param->connect.remote_bda, 6, 6, 0, 100);
    };

    ...
};

Print the connection parameters periodically
```cpp
// in loop...
esp_gap_conn_params_t params;
esp_ble_get_current_conn_params(connected_addr, &params);
log_i("Conn params: %d, %d, %d", params.interval, params.latency, params.timeout);
```
```

### Debug Message

```plain
[ 12102][V][BLEUtils.cpp:1170] dumpGapEvent(): [status: 16, bd_addr: 4b:af:66:d4:44:e9, min_int: 6, max_int: 6, latency: 0, conn_int: 40, timeout: 500]
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-10-02:
We are currently working on refactoring the ESP32 Arduino BLE Library.
It will be based on nimBLE instead of Bludroid.
We expect to fix many issues, including a smaller firmware size.

This issue may be fixed after copleting the refactoring work.
Please follow it in https://github.com/espressif/arduino-esp32/issues/9878

jefflongo, 2024-10-02:
I ended up refactoring my project to use esp-idf with Arduino and NimBLE as components and I do believe it's solved the problem.

Parsaabasi, 2025-03-12:
Hello @jefflongo 

Seems like your issue is solved. I close this for now. In case the issue persists or you need any further assistance, please feel free to reopen, or create a new issue. 

Thanks.