espressif/arduino-esp32 #10405: Zigbee Implementation: Deep Sleep example

HiFiJ, 2024-10-02:
### Related area

Zigbee

### Hardware specification

ESP32-C6

### Is your feature request related to a problem?

Hi,
thanks for the tremendous and fast implementation and merge of the Zigbee library!

I am wondering if there is already an easy way at this point to set up a battery powered ESP32(-C6) endpoint utilising deep sleep?

Or are there higher obstacles to pass to realize deep sleep/short wake up times for e.g. sending sensor readings?

Thanks again very much, best regards,
Joost

### Describe the solution you'd like

Perhaps one of the examples could be enhanced slightly to include deep sleep with periodic wake up.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

P-R-O-C-H-Y, 2024-10-04:
Hi @HiFiJ, I will get to add more stuff to Zigbee soon as now I have different important tasks. So I will link this feature request to the original issue of Zigbee library support, and once I get to the Zigbee library again, I will investigate this :)

P-R-O-C-H-Y, 2024-10-24:
I was investigating this and we don't have Power Management  or ULP Core enabled in the sdkconfig. 
There is more work to be done before supporting that. I will keep this issue updated if there are som ongoing tasks :)

HiFiJ, 2024-10-24:
Thanks very much for taking a closer look!
From a hobbyist's perspective, I believe that there is great potential for this Zigbee stack to be the first to enable useful battery powered devices with wireless signal transmitting for DIYers/amateurs. 
At least I have been looking for easy ways to accomplish this for years (MySensors project etc...) without finding a viable alternative and also have read several times of other hobbyists struggles to achieve this. 
A lot of people are looking forward for this to happen :-)
Best regards, Joost

P-R-O-C-H-Y, 2024-10-24:
> Thanks very much for taking a closer look!
>
> From a hobbyist's perspective, I believe that there is great potential for this Zigbee stack to be the first to enable useful battery powered devices with [...]

I will be doing some testing in upcoming days to see what we can do with power consumption for Zigbee. The deep_sleep may be possible without a need of changing the core libs configuration. Will keep you updated :)

HiFiJ, 2024-11-03:
Hi,
thanks a lot for your tremendous work!
I am absolutely looking forward to try/use this, great choice of clusters as well.
I wil just have to wait until this ends up in an arduino-esp32, as I do not have the necessary knowledge to switch to your branch.

Thanks again! Best regards,

Joost

PR #10551 (2024-11-06): feat(Zigbee): Support for sleepy device + Power Source and battery level + Humidity sensor cluster
## Description of Change
This PR includes:

- Support for sleepy device (only deep-sleep is currently possible).
- Support for selecting power source for the device. If battery is selected, the remaining percentage can be updated.
- Support for Humidity cluster, which was added as optional for the Temperature sensor as there is no HA Device type for humidity sensor.
- New example `Zigbee_Temp_Hum_Sensor_Sleepy`, which demonstrates all the new functionalities
- Mirror fixes and improvements

## Te [... cut]
closed 2024-11-06 by me-no-dev