espressif/arduino-esp32 #10407: ledcWriteTone() always uses 10 bit resolution

vvb333007, 2024-10-02:
### Board

ESP32-WROOM-32D

### Device Description

devkit

### Hardware Configuration

tiny blue led connected to GPIO2

### Version

v3.0.4

### IDE Name

ArduinoIDE

### Operating System

windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

ledcWriteTone() configures ledc channel to requested frequency but assumes 10 bit resolution.
As result any attempt to use ledcWriteTone() for frequencies above 78 kHz result in error messages:

E (93107) ledc: requested frequency and duty resolution can not be achieved, try                                                                                                              reducing freq_hz or duty_resolution. div_param=0
[ 92088][E][esp32-hal-ledc.c:262] ledcWriteTone(): ledcWriteTone configuration f                                                                                                             ailed!

Interestingly, that setting resolution to 8 bit and doing ledcWrite(); ledcWriteTone() the same
message appears BUT PWM generation starts: i can see it by attached led. In particular the above
messages happened when i attempted to set 300kHz frequency and duty=1. I connected a scope and I see that PWM generation happens at exactly 300kHz but still I see these error messages.  This happens with the code attached, Arduino Core 3.04 and latest as well.

### Sketch

```cpp
static int tone_enable(unsigned int pin, unsigned int freq, float duty) {

  int resolution = 8;

  if (duty > 1.0f)       duty = 1.0f;

  pinMode(pin,OUTPUT);
  ledcWriteTone(pin, 0); //disable ledc at pin.

  if (freq) {
    if (ledcAttach(pin, freq, resolution) == 0)
      return -1;
    ledcWriteTone(pin, freq);
    ledcWrite(pin, (unsigned int)(duty * ((1 << resolution) - 1)));
  }

  return 0;
}
```

### Debug Message

```plain
E (93107) ledc: requested frequency and duty resolution can not be achieved, try                                                                                                              reducing freq_hz or duty_resolution. div_param=0
[ 92088][E][esp32-hal-ledc.c:262] ledcWriteTone(): ledcWriteTone configuration f                                                                                                             ailed!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

vvb333007, 2024-10-02:
Full debug message:

[ 82797][I][esp32-hal-ledc.c:166] ledcAttachChannel(): LEDC attached to pin 2 (c                                                                                                             hannel 0, resolution 8)
E (83826) ledc: requested frequency and duty resolution can not be achieved, try                                                                                                              reducing freq_hz or duty_resolution. div_param=0
[ 82807][E][esp32-hal-ledc.c:262] ledcWriteTone(): ledcWriteTone configuration f                                                                                                             ailed!

lbernstone, 2024-10-02:
What tone are you trying to generate at 78KHz?  Human hearing doesn't get anywhere near that. For general PWM, use ledc.

vvb333007, 2024-10-04:
I am using ledc. It is in the title.

lbernstone, 2024-10-04:
ledcWriteTone is a convenience function intended for producing audible signals.  This is why it is fixed at 10-bit, in order to generate frequencies b/w 3kHz-10kHz.
If you want to generate a PWM signal, use [ledcAttach and ledcWrite](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/ledc.html#ledcattach) to get the required frequency. Use a high bit resolution to get the widest range- 12-bit is what you would typically use to drive LEDs or motors at frequencies high enough (>20kHz) that you won't hear the hum.

closed 2024-10-04 by vvb333007