espressif/arduino-esp32 #10413: SERIAL_TX_BUFFER_SIZE/SERIAL_RX_BUFFER_SIZE are not defined

aaronw2, 2024-10-03:
### Board

custom esp32s3 board

### Device Description

Custom board with ESP32S3 and FUSB302 chip connected to USB-C port. The problem is unrelated to the board.

### Hardware Configuration

ESP32S3 with FUSB302 i2c chip connected. The problem is related to SERIAL_TX_BUFFER_SIZE not being defined for the FUSB302 library.

### Version

v3.0.4

### IDE Name

Arduino 2.3.3

### Operating System

Linux

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am attempting to use the FUSB302_UFP library, which fails to compile because it expects SERIAL_TX_BUFFER_SIZE to be defined. Performing a grep, I see it is also used by the MobaLedLib library and possibly several others. Using a hard-coded value may be problematic, so this may require more work with HardwareSerial.h to figure out what the current size is.

### Sketch

```cpp
#include <Wire.h>
#include <PD_UFP.h>

#define FUSB302_INT_PIN   12

PD_UFP_c PD_UFP;

void setup() {
  Wire.begin();
  PD_UFP.init(FUSB302_INT_PIN, PD_POWER_OPTION_MAX_20V);
  
  Serial.begin(9600);
}

void loop() {
  PD_UFP.run();
  if (PD_UFP.is_power_ready())
  {
    Serial.write("PD supply connected\n");
  }
  else
  {
    Serial.write("No PD supply available\n");
  }
}
```

### Debug Message

```plain
/home/aaronw/Arduino/libraries/FUSB302_PD_UFP_sink/src/PD_UFP_Log.cpp: In member function 'void PD_UFP_Log_c::print_status(HardwareSerial&)':
/home/aaronw/Arduino/libraries/FUSB302_PD_UFP_sink/src/PD_UFP_Log.cpp:234:49: error: 'SERIAL_TX_BUFFER_SIZE' was not declared in this scope
  234 |     if (serial && serial.availableForWrite() >= SERIAL_TX_BUFFER_SIZE - 1) {
      |                                                 ^~~~~~~~~~~~~~~~~~~~~
/home/aaronw/Arduino/libraries/FUSB302_PD_UFP_sink/src/PD_UFP_Log.cpp:236:33: error: 'buf' was not declared in this scope
  236 |         if (status_log_readline(buf, sizeof(buf) - 1)) {
      |                                 ^~~

exit status 1

Compilation error: exit status 1
```

### Other Steps to Reproduce

No other steps needed.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

BaiHengRui, 2024-10-06:
PD_UFP.run();
  if((millis() - LastTime) >= 1 * 100 )
  {
     LastTime = millis();
     PD_UFP.status_log_readline(buf, sizeof(buf) - 1);
  }
  PD_UFP.print_status(Serial);

Jason2866, 2025-03-12:
Closing since outdated. Feel free to ask to reopen if the issue still exists with actual Arduino core

closed 2025-03-12 by Jason2866