espressif/arduino-esp32 #10414: Support for ESP32C2

chen70854022, 2024-10-03:
### Related area

At present, it is known that ESP32C2 supports Arduino, but it needs to be compiled by oneself. When can C2 be added to the main program, just like using C3

### Hardware specification

ESP32C2 esp8684

### Is your feature request related to a problem?

At present, it is known that ESP32C2 supports Arduino, but it needs to be compiled by oneself. When can C2 be added to the main program, just like using C3

### Describe the solution you'd like

It can be used as conveniently as ESP32C3

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2024-10-03:
Hello @chen70854022,

we decided that ESP32-C2 is not so important SoC for Arduino project as this SoC will be mainly used in large volume commercial applications due to cost-reduction and for that can be used ESP-IDF framework.

You can read more about this in original ticket: https://github.com/espressif/arduino-esp32/issues/6701

ESP32-C2 support was added by Community.

VojtechBartoska, 2024-10-04:
I am closing this ticker as duplicate of #6701, thanks for understanding.

closed 2024-10-04 by VojtechBartoska