espressif/arduino-esp32 #10419: 2.0.18 release request

BlueAndi, 2024-10-03:
### Related area

2.0.18 release

### Hardware specification

*

### Is your feature request related to a problem?

Its related to the bugfixes which are in the release/2.0.x branch. I can't upgrade to v3.x and still rely on the v2.x

### Describe the solution you'd like

Would be cool to get a 2.0.18 release.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-10-04:
You can instead install it from git, which will allow you to keep up with the updates and still have the latest libs. We will not be releasing anymore 2.x versions

BlueAndi, 2024-10-04:
> You can instead install it from git, which will allow you to keep up with the updates and still have the latest libs. We will not be releasing anymore 2.x versions

Ok, thanks for clarification. I will see how I can manage it.

closed 2024-10-04 by BlueAndi