espressif/arduino-esp32 #10420: UART Communication Delays and Data Loss

mightChamp, 2024-10-04:
### Board

ESP32 Wroom 32E 4MB Flash

### Device Description

We are making an Uart Data logger, an external device send uart data, which stored in SD card.

### Hardware Configuration

GPIO13- Uart TX, and GPIO14- Uart RX
Uart setup:
Baud Rate: 115200
Flow Control: Disabled
Buffer Size: 4096

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

We are encountering an issue with UART communication on ESP32 where data reception is delayed, and sometimes data is lost. This issue occurs intermittently and was not present in version v1.0.6.

### Sketch

```cpp
void setup() {
    Serial.begin(115200);  // UART initialized at 115200 baud rate
}

void loop() {
    if (Serial.available()) {
        String data = Serial.readString();
        Serial.println(data);
    }
}
```

### Debug Message

```plain
No Debug Message
```

### Other Steps to Reproduce

Set up UART communication between ESP32 and another device.
Send continuous data streams over UART.
Observe occasional delays and missed data.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-10-04:
The HardwareSerial library has been modified to have more capabilities for handling intermittent and burst data flows.  Take a look at the [OnReceive](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/OnReceive_Demo/OnReceive_Demo.ino) and [RxTimeout](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/RxTimeout_Demo/RxTimeout_Demo.ino) examples to see how to handle high activity and sporadic flows.
@SuGlider This could definitely use some [doco](https://docs.espressif.com/projects/arduino-esp32/en/latest/libraries.html#apis)

SuGlider, 2024-10-04:
@mightChamp - The issue is related only to Core 3.1.0-RC1? Does it also happen with Core 3.0.5?

SuGlider, 2024-10-05:
This sounds related to #10397 
After testing the sketch from the issue, it seems clear that IDF 4.x and 5.x UART Driver used by ESP32 Arduino HardwareSerial is not efficient and causes a huge performance issue.

The evidence is that whenever `Serial0.begin()` is executed, there is a significative time slot reduction to the main Arduino Task, affecting the time that `loop()` has to process data.

In Arduino Core 1.0.6, the UART driver was local one, faster and way more efficient.
The solution for this issue is to refactor HardwareSerial and create its own Arduino UART Driver, the same way it is done for SPI.
@me-no-dev - FYI.

SuGlider, 2024-10-07:
@mightChamp - Could you please check if PR #10428 solves this issue too?
This PR improves the `loop()` execution efficiency and may give it more "CPU cycles" in order to solve the data loss issue.

mightChamp, 2024-10-07:
We have tested the fix from PR #10428, but unfortunately, we are still facing the same issue.

This issue is currently blocking our development progress, and we would greatly appreciate an estimate of how long it might take to investigate or resolve this. If there are any further tests or adjustments we could try, we are ready to assist.

SuGlider, 2024-10-07:
@mightChamp - based on what @lbernstone wrote, there are a few new methods to deal with UART that were not available in 1.0.6.

- It is possible to create a callback function that is executed as soon as data is available in the UART FIFO using `Serial.onReceive(calbackFunc)` - please check the example and its commentaries in 
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/OnReceive_Demo/OnReceive_Demo.ino
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.h#L229-L239

- Change RX FIFO Timeout and RX FIFO Full events. Those control when the driver will call `onReceive` callback.
Arduino Core 1.0.6 sets Timeout to 2 symbols (about 20 bits at the current baud rate ~ for 115200, its is 174 ms) and also sets FIFO Full to 1 character, which forces the backback function to be called as soon as the first byte arrives.
Example and documentation:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/RxFIFOFull_Demo/RxFIFOFull_Demo.ino
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/RxTimeout_Demo/RxTimeout_Demo.ino

- A example that reads UART data flow up to a certain timeout
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Serial/onReceiveExample/onReceiveExample.ino

SuGlider, 2024-10-07:
I'll  build a Python script + the sketch example that is here listed to find out more about the issue.
If I find out something else, I'll write about it here.

mightChamp, 2024-10-09:
> @mightChamp - based on what @lbernstone wrote, there are a few new methods to deal with UART that were not available in 1.0.6.
>
> * It is possible to create a callback function that is executed as [...]

Despite testing all examples, we encountered significant performance degradation, and also faces missing data bytes and delay issue.

SuGlider, 2024-10-09:
>Despite testing all examples, we encountered significant performance degradation, and also faces missing data bytes and delay issue.

Please elaborate more about "significant performance degradation", "missing data bytes" and "delay issue".
If you could post an example (sketch + PC side script + details) for each issue, it would accelerate the resolution and also help us to validate a possible fix.

You can count on me for fixing it ASAP.

Thanks!

mightChamp, 2024-10-10:
@SuGlider  

We are experiencing a delay issue that creates the illusion of missing bytes due to timeout. Also PR #10248, reduces this delay, but still face some delay, as compared to V1.0.6. 

Our project is quite large, using 1.6 MB of Flash and 35% of RAM, with two additional tasks running. The code works perfectly in v1.0.6, but we are looking to upgrade to the latest SDK. Due to the size and complexity of the code, providing a simplified version for testing is challenging.

EricHarbers, 2025-03-06:
I had the same issue.
I have solved it, by setting the following functions:
  Serial.setRxFIFOFull(60);
  Serial.setRxTimeout(60);

This works well on BSP version 3.0.3. The println function will then write the value of 60. And the connection will not loose data anymore.
If you set the value to more than 64, it will not work anymore. Then the println function will return random numbers, and maybe big numbers (much bigger than 60.
I had read on the internet, that the hardware FIFO will be 128 bytes big. If I take a look at the max value of 64, then I cann't believe that the FIFO is 128, It should be 64, or something wouldn't work properly inside the BSP.

For the latest BSP version now (V 3.1.3), the value of 60 wouldn't work anymore. Then the println function will also return random numbers, (and the connection will loose data).

Please can anyone repair this?

EricHarbers, 2025-03-06:
Update:
I've checked some more.
It seems that the value for Serial.setRxFIFOFull could be 120. That is valid.
But for Serial.setRxTimeout, the maximum value is 101. If you set the timeout to more than that, data will be lost.

EricHarbers, 2025-03-06:
The above works for esp32 V3.0.3.
For V3.1.3, it will not work at all!!

Jason2866, 2025-03-13:
@EricHarbers Could you test with core 3.2.0-rc2?

Jason2866, 2025-03-13:
@SuGlider Can you have a look, if the issue is still in 3.2.0-rc2 it would be good if it could be fixed before release of 3.2.0

EricHarbers, 2025-03-13:
@Jason2866 I've checked with core 3.2.0-RC2, but the the problem is still there. The version again does not read stable block sizes.
Let me know if I can check more for you.

Jason2866, 2025-03-13:
@EricHarbers Thx for the testing. SuGlider is the expert and will have for sure a look. I am only helping a bit to track the issues.

SuGlider, 2025-03-13:
I have tested it. It works fine with Arduino Core 3.1.3 or using master branch.
`readString()` is a `Stream` method, therefore, it has timeout of 1 second. It can be seen in the output time stamp
It will read all data stream and only return after 1 second after not reading anything.

There is no data loss.
I tested it sending data through the Serial Monitor, using 20 byte like this: `0123456789abcdefghij` with no nl/cr at the end.
Tested sending 20, 100, 500, 2000 and 5000 bytes at once. Check the output and timestamps.

100 bytes test  can use this input: `0123456789abcdefghij0123456789abcdefghij0123456789abcdefghij0123456789abcdefghij0123456789abcdefghij`

``` cpp
void setup() {
    Serial.begin(115200);  // UART initialized at 115200 baud rate
}

void loop() {
    if (Serial.available()) {
        Serial.println("got a data strem");
        String data = Serial.readString();
        Serial.printf("\r\nRead %d bytes.\r\n", data.length());
        uint32_t i = 0;
        uint32_t dataLen = data.length();
         while (i < dataLen) {
          int r = dataLen > 50 ? 50 : dataLen;
          for (int j = 0; j < r; j++) {
            Serial.print((char) *(data.c_str() + i + j));
          }
          i += r;
          Serial.println();
        }
    }
}
```
#### Output:
```
17:29:26.623 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:29:26.623 -> configsip: 0, SPIWP:0xee
17:29:26.623 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:29:26.623 -> mode:DIO, clock div:1
17:29:26.623 -> load:0x3fff0030,len:4888
17:29:26.623 -> load:0x40078000,len:16516
17:29:26.623 -> load:0x40080400,len:4
17:29:26.623 -> load:0x40080404,len:3476
17:29:26.623 -> entry 0x400805b4
17:29:33.863 -> got a data strem
17:29:34.848 -> 
17:29:34.848 -> Read 20 bytes.
17:29:34.848 -> 0123456789abcdefghij
17:29:45.627 -> got a data strem
17:29:46.644 -> 
17:29:46.644 -> Read 100 bytes.
17:29:46.644 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:29:46.644 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:29:57.890 -> got a data strem
17:29:58.941 -> 
17:29:58.941 -> Read 500 bytes.
17:29:58.941 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:29:58.941 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:29:58.941 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:29:58.941 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:29:58.974 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:29:58.974 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:29:58.974 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:29:58.974 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:29:58.974 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:29:58.974 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:04.532 -> got a data strem
17:30:05.599 -> 
17:30:05.599 -> Read 2000 bytes.
17:30:05.599 -> 0123456789abcdefghij0123456789abcdefghij0123456789
[... 118 lines cut ...]
17:30:14.306 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.341 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.341 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.341 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.341 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.341 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.341 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.341 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.341 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.375 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.375 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.375 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.375 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.375 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.375 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.375 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.375 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.408 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.408 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.408 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.408 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.408 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
17:30:14.408 -> 0123456789abcdefghij0123456789abcdefghij0123456789
17:30:14.408 -> abcdefghij0123456789abcdefghij0123456789abcdefghij
```

EricHarbers, 2025-03-13:
#define BAUD_RATE 230400

void receiveSerial();
void receiveErrorSerial(hardwareSerial_error_t e);

unsigned int uartBaudrate;

void setup() {
  // put your setup code here, to run once:
  uartBaudrate = BAUD_RATE;

  Serial.begin(uartBaudrate);
  Serial1.begin(uartBaudrate, SERIAL_8N1, 26, 27);

// the 2 lines below will work properly for esp32 BSP V3.0.3
// for the version V3.1.3, it will not work at all
  Serial.setRxFIFOFull(120);

  Serial.setRxTimeout(50);  // max value 101 on core 3.0.3

  Serial.onReceive(bluetooth_receiveSerial);  // bluetooth_receiveSerial
  Serial.onReceiveError(bluetooth_receiveErrorSerial);
}

void receiveSerial() {
  static int col = 0;
  size_t len;

  while ((len = Serial.read(ReceiveBuffer, sizeof(ReceiveBuffer))) > 0) {
    SerialBT.write(ReceiveBuffer, len);
    Serial1.printf("%5u", len);
    if (++col == 25) {
      Serial1.printf("\r\n");
      col = 0;
    }
  }
}

void receiveErrorSerial(hardwareSerial_error_t e) {
//  Serial1.printf("Error %X\r\n", e);
}

void loop() {
  // put your main code here, to run repeatedly:
}

////////////////////////
Serial.setRxTimeout(50); will set the timeout
For core 3.0.3 the above code works file if you send a data stream at 230400 baud to the serial channel
The output on Serial1 will be:
  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120
  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120
  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120
  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120  120

The 120 is the 120 bytes set in the setRxFIFOFull function

With core 3.1.3 and newer, the output will be random, eg:
   12    8   13   12    4  378   21    5    8, etc...
 
Most values will be very small releated to the setting of 120 set with the setRxFIFOFull function.
Sometimes a big byte count will be displayed, eg 378. When this is the case, also data will be lost.
It doesn't occur with a couple of hundreths of bytes, but with thousands of bytes.
Our application will receive the data, and this should be without any byte lost.

It seems that the Serial.setRxTimeout() function will not work properly.

SuGlider, 2025-03-13:
The issue is that `HardwareSerial::onReceive()` works like an UART ISR.
I see that it is used to write to BT Serial. This operation take time and it may lose incoming UART data.

Any ISR shall do the minimal necessary to just get the UART data.
All other operation shall take place in a separated task, such as `loop()`, or something else created for that need.

It is necessary to change the code.

SuGlider, 2025-03-13:
This issue is cross related to https://github.com/espressif/arduino-esp32/issues/10314#issuecomment-2722940609

EricHarbers, 2025-03-14:
@SuGlider How can you explain that the code with V3.0.3 and below will receive nice data chunks of 120 bytes, set by the setRxFIFOFull function, and not loose data at any time but for V3.1.3 it will read random length blocks, short and very long, and loose data.
Imho it is related to the RxTimeout, wich doesn't work properly for V3.1.3.

But, i've a workaround for this problem: I'll use V3.0.3 and everything works fine.
It's up to you what you do with the RxTimeout problem.

EricHarbers, 2025-03-14:
@SuGlider I've test it with a disabled BT write call, and it will output the same random lengthes, for V3.1.3

EricHarbers, 2025-03-14:
@suglider I've now called my receiveSerial() function, which calls the BT write(), from within the loop() function, and it works the same for V3.1.3. It also looses data regularly.

EricHarbers, 2025-03-17:
@SuGlider This works for me! There is no data loss now.
The main problem was, that I did call the Serial.setRxBufferSize() after the Serial.begin() call. That caused the data loss, because the default RxBuffer was too small for my application.

Jason2866, 2025-03-17:
@EricHarbers Can we consider as solved and close this issue?

EricHarbers, 2025-03-17:
@SuGlider I agree with that.

closed 2025-03-17 by Jason2866