espressif/arduino-esp32 #10422: Example CaptivePortal from DNSServer.h

RodrigoStefanini, 2024-10-05:
### Related area

Esp32

### Hardware specification

Esp32 wroom

### Is your feature request related to a problem?

The example of CaptivePortal do not work for me, and i modify the code for be usefull.

### Describe the solution you'd like

The proposal is to make the example functional, if this occurs for other users.

### Describe alternatives you've considered

Change this:
  "if (dnsServer.start()) {
    Serial.println("Started DNS server in captive portal-mode");
  } else {
    Serial.println("Err: Can't start DNS server!");
  }
"

### Additional context

The code only work for me wen i put:

"dnsServer.start(53, "*", WiFi.softAPIP());"

I dont know if this is about some version, i used the 3.0.5 esp32 lib for arduino.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-10-05:
Already fixed in https://github.com/espressif/arduino-esp32/pull/10366

lbernstone, 2025-02-24:
@N0edL Many android phones are hard coded to use 8.8.8.8 as their DNS server.  If you set the softAP to use that address, it can redirect those phones.
```
WiFi.softAPConfig(IPAddress(8,8,8,8), IPAddress(8,8,8,8), IPAddress(255,255,255,0));
```

N0edL, 2025-02-24:
@lbernstone Ahh okay what about apple ?

lbernstone, 2025-02-25:
Please don't hijack this issue, unless the exact same workaround works for you. Turn verbose logging on and open a new issue with the log.

closed 2025-03-12 by Jason2866