espressif/arduino-esp32 #10425: After first time flash i am not able to  flash it again

faizannazir, 2024-10-06:
### Board

ESP32-D0WD-V3

### Device Description

it is custom design module 

### Hardware Configuration

There is external flash memory AT25SF041B-SSHB-B and 1 sd card port is connected  using some shift register or some thing 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Window 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

When i get board it was giving invalid header which is normal to new board first time i am able to enter in download mode and i flashed empty sketch from arduino ide  but after flash success it keep resetting and i am not able to enter in download mode again 

**Serial log** 
`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:03:55.274 -> configsip: 0, SPIWP:0xee
11:03:55.274 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:03:55.274 -> mode:DIO, clock div:1
11:03:55.274 -> load:0x3fff0030,len:1184
11:03:55.274 -> load:0x40078000,len:13260
11:03:55.274 -> load:0x40080400,len:3028
11:03:55.274 -> entry 0x400805e4
11:03:55.451 -> E (119) spi_flash: Detected size(0k) smaller than the size in the binary image header(4096k). Probe failed.
11:03:55.451 -> 
11:03:55.451 -> assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)
11:03:55.451 -> 
E (141) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
11:03:55.274 -> Rebooting...
11:03:55.274 -> ets Jul 29 2019 12:21:46`

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
E (141) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
11:03:55.274 -> Rebooting...
11:03:55.274 -> ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:03:55.274 -> configsip: 0, SPIWP:0xee
11:03:55.274 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:03:55.274 -> mode:DIO, clock div:1
11:03:55.274 -> load:0x3fff0030,len:1184
11:03:55.274 -> load:0x40078000,len:13260
11:03:55.274 -> load:0x40080400,len:3028
11:03:55.274 -> entry 0x400805e4
11:03:55.451 -> E (119) spi_flash: Detected size(0k) smaller than the size in the binary image header(4096k). Probe failed.
11:03:55.451 -> 
11:03:55.451 -> assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)
11:03:55.451 ->
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-10-06:
It certainly looks like it is not able to identify the chip after the bootloader hands off control to the main app.  Does this work with a [basic esp-idf app](https://github.com/espressif/esp-idf/tree/release/v5.1/examples/storage/nvs_rw_value) or [a custom driver](https://github.com/espressif/esp-idf/tree/release/v5.1/examples/storage/custom_flash_driver)?

faizannazir, 2024-10-06:
I am able to flash again it was not entering auto download mode using rts and dtr so i enter manually  but still it printing spi flash memory 0k and keep rebooting

lbernstone, 2024-10-06:
What do you get if you run [`esptool chip_id`](https://github.com/espressif/esptool)?

faizannazir, 2024-10-06:
[image]

lbernstone, 2024-10-06:
Sure does sound like unsupported hardware.  As suggested, compile a firmware in [esp-idf](https://docs.espressif.com/projects/esp-idf/en/v5.1.4/esp32s3/get-started/windows-setup.html), and then submit the [issue upstream](https://github.com/espressif/esp-idf/issues).

faizannazir, 2024-10-06:
How you guess flash size?

faizannazir, 2024-10-06:
I had tried idf it also give same error instead 4MB It gives 2MB error

faizannazir, 2024-10-07:
Currentl
This flash chip support esp32 https://www.mouser.co.uk/ProductDetail/Winbond/W25Q80RVSNJQ?qs=mELouGlnn3eiLlSxpfAziw%3D%3D

Jason2866, 2024-10-07:
Look through IDF source code which flash chips are supported.

lbernstone, 2024-10-07:
W25Q80RVS will likely be recognized, but 2MB (16Mbit) is the smallest supported by the framework.

faizannazir, 2024-10-07:
> Look through IDF source code which flash chips are supported.

> W25Q80RVS will likely be recognized, but 2MB (16Mbit) is the smallest supported by the framework.

> Sure does sound like unsupported hardware. 4MBit is not enough to boot arduino-esp32. As suggested, compile a firmware in [esp-idf](https://docs.espressif.com/projects/esp-idf/en/v5.1.4/esp32s3/get-started/windows-setup.html), and then submit the [issue upstream](https://github.com/espressif/esp-idf/issues).

How you know its unsupported and size of flash 4Mbit?

faizannazir, 2024-10-07:
> Look through IDF source code which flash chips are supported.

> W25Q80RVS will likely be recognized, but 2MB (16Mbit) is the smallest supported by the framework.

> Sure does sound like unsupported hardware. 4MBit is not enough to boot arduino-esp32. As suggested, compile a firmware in [...]

How you know its unsupported and size of flash 4Mbit?

lbernstone, 2024-10-07:
I know it is unsupported since it is not recognized by your device, and there is no driver for Renesas chips.  I know it is 4Mbit b/c I read the datasheet.