espressif/arduino-esp32 #10440: ETH.cpp: Compilation error in release/3.1.x branch (CONFIG_ETH_RMII_CLK_IN_GPIO)

hurricanefrog, 2024-10-09:
### Board

ESP32 based custom board (doesn't matter)

### Device Description

Doesn't matter

### Hardware Configuration

Doesn't matter

### Version

latest development Release Candidate (RC-X)

### IDE Name

VScode

### Operating System

Windows 11, Linux

### Flash frequency

Doesn't matter

### PSRAM enabled

no

### Upload speed

Doesn't matter

### Description

I am using arduino as a component in an ESP-IDF project, included as a submodule. Both the RC tag and the currect branch `release/3.1.x` suffer from the following problem.

ESP-IDF is v5.3.1.

With this configuration, compilation fails with

```
C:/DEVELOP/firmware/ipuartbridge/components/arduino/libraries/Ethernet/src/ETH.cpp: In member function 'bool ETHClass::begin(eth_phy_type_t, int32_t, int, int, int, eth_clock_mode_t)':
C:/DEVELOP/firmware/ipuartbridge/components/arduino/libraries/Ethernet/src/ETH.cpp:175:56: error: 'CONFIG_ETH_RMII_CLK_IN_GPIO' was not declared in this scope; did you mean 'CONFIG_ETH_RMII_CLK_OUT_GPIO'?       
  175 | #define DEFAULT_RMII_CLK_GPIO (emac_rmii_clock_gpio_t)(CONFIG_ETH_RMII_CLK_IN_GPIO)
      |                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/VSCode/espressif-v5/idf/components/esp_eth/include/esp_eth_mac_esp.h:233:31: note: in expansion of macro 'DEFAULT_RMII_CLK_GPIO'
  233 |                 .clock_gpio = DEFAULT_RMII_CLK_GPIO   \
      |                               ^~~~~~~~~~~~~~~~~~~~~
C:/DEVELOP/firmware/ipuartbridge/components/arduino/libraries/Ethernet/src/ETH.cpp:178:40: note: in expansion of macro 'ETH_ESP32_EMAC_DEFAULT_CONFIG'
  178 |   eth_esp32_emac_config_t mac_config = ETH_ESP32_EMAC_DEFAULT_CONFIG();
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

This is because my `sdkconfig` looks like this:

```
CONFIG_ETH_USE_ESP32_EMAC=y
CONFIG_ETH_PHY_INTERFACE_RMII=y
# CONFIG_ETH_RMII_CLK_INPUT is not set
CONFIG_ETH_RMII_CLK_OUTPUT=y
# CONFIG_ETH_RMII_CLK_OUTPUT_GPIO0 is not set
CONFIG_ETH_RMII_CLK_OUT_GPIO=17
```

So the value for `CONFIG_ETH_RMII_CLK_IN_GPIO` is not there, and can not be due to the Kconfig.
Master branch compiles fine, because it doesn't undefine-then-redefine the value for `DEFAULT_RMII_CLK_GPIO`:

```c
// Eth.cpp...
#if CONFIG_IDF_TARGET_ESP32
#undef DEFAULT_RMII_CLK_GPIO
#define DEFAULT_RMII_CLK_GPIO (emac_rmii_clock_gpio_t)(CONFIG_ETH_RMII_CLK_IN_GPIO)
#endif
```

The value `DEFAULT_RMII_CLK_GPIO` would be the correct one for my project, because `esp-idf\components\esp_eth\include\esp_eth_mac_esp.h` defines it correctly.

The lines were introduced in e1673e4c0ca89153e26d31cd783f0f8260c6d0fb.

### Sketch

```cpp
Doesn't matter, compilation fails as soon as arduino is used as a component.
```

### Debug Message

```plain
See description
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

hurricanefrog, 2024-10-09:
addendum: But apprerently, without these lines, compilation fails with `invalid conversion from 'int' to 'emac_rmii_clock_gpio_t'`...

SuGlider, 2024-10-09:
@hurricanefrog - This is the default sdkconfig settings used in Aduino Core:
Try it with your project and let me know.

```
#
# Ethernet
#
CONFIG_ETH_ENABLED=y
CONFIG_ETH_USE_ESP32_EMAC=y
CONFIG_ETH_PHY_INTERFACE_RMII=y
CONFIG_ETH_RMII_CLK_INPUT=y
# CONFIG_ETH_RMII_CLK_OUTPUT is not set
CONFIG_ETH_RMII_CLK_IN_GPIO=0
CONFIG_ETH_DMA_BUFFER_SIZE=512
CONFIG_ETH_DMA_RX_BUFFER_NUM=10
CONFIG_ETH_DMA_TX_BUFFER_NUM=10
# CONFIG_ETH_IRAM_OPTIMIZATION is not set
CONFIG_ETH_USE_SPI_ETHERNET=y
CONFIG_ETH_SPI_ETHERNET_DM9051=y
CONFIG_ETH_SPI_ETHERNET_W5500=y
CONFIG_ETH_SPI_ETHERNET_KSZ8851SNL=y
# CONFIG_ETH_USE_OPENETH is not set
# CONFIG_ETH_TRANSMIT_MUTEX is not set
# end of Ethernet
```

hurricanefrog, 2024-10-10:
> @hurricanefrog - This is the default sdkconfig settings used in Aduino Core: Please try it with your project and let me know.
> 
> ```
> #
> # Ethernet
> #
> CONFIG_ETH_ENABLED=y
> CONFIG_ETH_USE_ESP32_EMAC=y
> CONFIG_ETH_PHY_INTERFACE_RMII=y
> CONFIG_ETH_RMII_CLK_INPUT=y
> # CONFIG_ETH_RMII_CLK_OUTPUT is not set
> CONFIG_ETH_RMII_CLK_IN_GPIO=0
> CONFIG_ETH_DMA_BUFFER_SIZE=512
> CONFIG_ETH_DMA_RX_BUFFER_NUM=10
> CONFIG_ETH_DMA_TX_BUFFER_NUM=10
> # CONFIG_ETH_IRAM_OPTIMIZATION is not set
> CONFIG_ETH_USE_SPI_ETHERNET=y
> CONFIG_ETH_SPI_ETHERNET_DM9051=y
> CONFIG_ETH_SPI_ETHERNET_W5500=y
> CONFIG_ETH_SPI_ETHERNET_KSZ8851SNL=y
> # CONFIG_ETH_USE_OPENETH is not set
> # CONFIG_ETH_TRANSMIT_MUTEX is not set
> # end of Ethernet
> ```

While this will probably solve compiling, I am not sure if this will help me in practice. My hardware is the way it is. As documented in https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_eth.html , option c is used.

hurricanefrog, 2024-10-11:
I am not sure, maybe something like this would solve the problem for all possible hardware configs?

```cpp
#undef DEFAULT_RMII_CLK_GPIO
#if defined(CONFIG_ETH_RMII_CLK_IN_GPIO)
#define DEFAULT_RMII_CLK_GPIO (emac_rmii_clock_gpio_t)CONFIG_ETH_RMII_CLK_IN_GPIO
#endif
#if defined(CONFIG_ETH_RMII_CLK_OUT_GPIO)
#define DEFAULT_RMII_CLK_GPIO (emac_rmii_clock_gpio_t)CONFIG_ETH_RMII_CLK_OUT_GPIO
#endif
```

me-no-dev, 2024-10-23:
@hurricanefrog `CONFIG_ETH_RMII_CLK_IN_GPIO` can safely be replaced by `0` here. It will get overwritten later, depending on your clock config anyway. Please try :)

hurricanefrog, 2024-10-24:
> @hurricanefrog `CONFIG_ETH_RMII_CLK_IN_GPIO` can safely be replaced by `0` here. It will get overwritten later, depending on your clock config anyway. Please try :)

Ah, I should probably elaborate my problem: I am using arduino-esp32 as git submodule. Since the firmware is built in a CI/CD server, I would like to avoid "local modifications" as much as possible, i.e. use the submodule/component as-it-is. For the time being, I have a patch file (with the code I posted above) which I apply with `patch` / `git apply` via `CMakeLists.txt`, but I consider this ugly. That's why I would like this to be fixed here.

I am not *using* the ETH class in my code anyway, but I need a clean compile.

me-no-dev, 2024-10-24:
It's already merged in git (release/v3/1/x branch) and will be part of next 3.1 release: https://github.com/espressif/arduino-esp32/commit/6dfd95898387bf10eb6348a22c2499073efb9f16

closed 2024-10-24 by me-no-dev