espressif/arduino-esp32 #10445: Flash corruption after many resets - brownout related

tyeth, 2024-10-10:
### Board

All with extra no low power supervisor

### Device Description

Adafruit QTPY using S3 N4R2. Devices on i2c bus. Power supplied through battery connections, usb D+/D-/GND connected to host computer (USB VBUS disconnected).

### Hardware Configuration

got a pressure sensor and lipo fuel guage over i2c, not being initialised but still powered when the i2c power is toggled to simulate a change in load without using wifi

### Version

latest master (checkout manually)

### IDE Name

Arduino-IDE

### Operating System

win11 + mac latest non-beta

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

Using a nordic PPK2 as a programmable power supply I've been trying to recreate the FATFS partition corruption we've seen sporadically. It appears anecdotally to happen to battery powered devices, but not 100% sure it's limited to battery powered.

https://github.com/tyeth/python_nordic_ppk2_wipper_secrets_brownout

I ran the voltage at preset levels and let the device boot 50times, checking to see if it connected to the wifi or instead reported missing secrets and therefore partition corruption. I eventually saw the corruption (win11 laptop did 3.00v x 50 then 2.99v x 1 before corrupting, whereas the mac did less than 50 around 3v). 
I was able to boot the devices without corruption multiple times at lower and higher voltages, although booting unreliably, so it's obviously on the cusp.

Since proving to myself the boards corrupt around the 3v mark, depending on i2c load / wifi etc using the adafruit Wippersnapper firmware/sketch, I've now tried to create a brownout aware simple reproduction where I can catch the event instead of resetting and flag the system as unstable / react accordingly like disable flash + wifi (+ i2c power if controllable per board) then wait and reattempt flash init. 

I can't seem to find the values for Threshold for brownout_hal_config_t, nor brownout_ll_set_threshold function.
My interrupt function has also never been called, probably I'm using this all wrong, but as there was not a clear example except using boot reason I've ended up here.

### Sketch

```cpp
#include <esp_system.h>
#include "hal/brownout_ll.h"
#include "hal/brownout_hal.h"
#include <driver/gpio.h>

#define I2C_POWER_PIN TFT_I2C_POWER  // Use the appropriate I2C power pin for your board - testing on adafruit feather s3 reverse tft
#define BROWNOUT_THRESHOLD 3  // ESP_BROWNOUT_THRESH_2V8 // Adjust based on the range that fits 2.8V for this chip

bool brownoutDetected = false;

// Brownout interrupt handler
void IRAM_ATTR brownoutHandler(void* arg) {
    brownoutDetected = true;
    brownout_ll_intr_clear();  // Clear the interrupt to avoid re-triggering
    Serial.println("BROWNOUT! BROWNOUT!");
    Serial.println("BROWNOUT! BROWNOUT!");
    Serial.println("BROWNOUT! BROWNOUT!");
}

void printBootReason() {
    esp_reset_reason_t reset_reason = esp_reset_reason();
    Serial.print("Boot Reason: ");
    switch (reset_reason) {
        case ESP_RST_POWERON:
            Serial.println("Power-on reset");
            break;
        case ESP_RST_EXT:
            Serial.println("External pin reset");
            break;
        case ESP_RST_SW:
            Serial.println("Software reset");
            break;
        case ESP_RST_PANIC:
            Serial.println("Exception/panic reset");
            break;
        case ESP_RST_INT_WDT:
            Serial.println("Interrupt watchdog reset");
            break;
        case ESP_RST_TASK_WDT:
            Serial.println("Task watchdog reset");
            break;
        case ESP_RST_WDT:
            Serial.println("Other watchdog reset");
            break;
        case ESP_RST_DEEPSLEEP:
            Serial.println("Woke from deep sleep");
            break;
        case ESP_RST_BROWNOUT:
            Serial.println("Brownout reset");
            break;
        case ESP_RST_SDIO:
            Serial.println("Reset over SDIO");
            break;
        default:
            Serial.println("Unknown reset reason");
            break;
    }
}

void setup() {
    Serial.begin(115200);
    delay(2000);

    // Initialize the I2C power pin for simulating flash/file system initialization
    pinMode(I2C_POWER_PIN, OUTPUT);
    digitalWrite(I2C_POWER_PIN, HIGH);
    // digitalWrite(I2C_POWER_PIN, LOW);

    printBootReason();

    // Configure brownout with HAL structure
    brownout_hal_config_t config = {
        .threshold = BROWNOUT_THRESHOLD, // Set to the highest level chip supports
        .enabled = true,
        .reset_enabled = false,  // Disable reset to control via interrupt
        .flash_power_down = true,  // Control flash power if needed
        .rf_power_down = true  // Control RF power if needed
    };
    brownout_hal_config(&config);

    // Enable brownout detection and interrupt handling
    brownout_ll_bod_enable(true);
    brownout_ll_intr_enable(true);

    // Simulate flash/file system initialization
    if (!brownoutDetected) {
        Serial.println("Initializing I2C power (simulated flash init)...");
        digitalWrite(I2C_POWER_PIN, HIGH);  // Turn on the I2C power
    } else {
        Serial.println("Brownout detected, skipping I2C power initialization.");
    }
}

bool currentI2cPowerOn=true;
void loop() {
    // Monitor brownout status and update I2C power pin accordingly
    if (brownoutDetected) {
        Serial.println("Brownout active. Keeping I2C power off...");
        digitalWrite(I2C_POWER_PIN, LOW);
    } else {
        if(currentI2cPowerOn){
          Serial.println("Normal operation. Switching I2C power off.");
          digitalWrite(I2C_POWER_PIN, LOW);
        } else {
          Serial.println("Normal operation. Switching I2C power on.");
          digitalWrite(I2C_POWER_PIN, HIGH);
        }
        currentI2cPowerOn = !currentI2cPowerOn;
    }
    delay(2000);
}
```

### Debug Message

```plain
I haven't seen a stacktrace for it yet
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tyeth, 2024-10-10:
I'm also considering something similar to the comment here: https://github.com/espressif/arduino-esp32/issues/863#issuecomment-788121117

tyeth, 2024-10-10:
Spotted my mistake in terms of the interrupt handler not being assigned / set.

VojtechBartoska, 2024-10-21:
@me-no-dev Can you please help here with triage? Thanks

tyeth, 2024-10-21:
I'm happy to close this (our scenario is resetting after an erase operation on the file system which caused a brownout), but it would be useful if a more complex scenario could be supported in Arduino than just reboot on brownout (and reading reset reason).

If you're happy to continue discussing that, andy awful attempt at it, then leave we can leave the issue open

VojtechBartoska, 2024-10-22:
@tyeth seems for now there are no action points for us so I will close the ticket. Thanks for understanding.

closed 2024-10-22 by VojtechBartoska