espressif/arduino-esp32 #10468: Add Waveshare ESP32-S3-Nano Support

SteveNotSet, 2024-10-16:
### Related area

Board Support

### Hardware specification

Support for Waveshare ESP32-S3-Nano

### Is your feature request related to a problem?

Waveshare ESP32-S3-Nano is a compatible board for Arduino Nano ESP32, but with 8MB PSRAM.
The original Arduino Nano ESP32 seems not to have PSRAM.
PSRAM cannot be used if selected Arduino Nano ESP32.

### Describe the solution you'd like

Add Waveshare ESP32-S3-Nano Support. Just an Arduino Nano ESP32 with 8MB extra PSRAM.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2024-10-16:
Original Arduino Nano also has 8MB PSRAM. Looking at the specs, they should be the same.

dquadros, 2024-10-16:
> Original Arduino Nano also has 8MB PSRAM. Looking at the specs, they should be the same.

Complementing...
The PSRAM is inside the Nora module and is not mentioned in the specs at the product page. You will find it mentioned in the PDF datasheet.

closed 2024-10-16 by SteveNotSet