espressif/arduino-esp32 #10474: esp32-S2 usb cdc works only after serial monitor is opened

Sonic803, 2024-10-16:
### Board

Wemos S2 mini

### Device Description

Just the board connected with usbc connected to pc

### Hardware Configuration

 Nothing

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino ide v2

### Operating System

Arch Linux

### Flash frequency

Lolin s2 default

### PSRAM enabled

yes

### Upload speed

Lolin s2 default

### Description

I'm using the Lolin s2 mini board in arduino ide, with usb cdc enabled .
After uploading this program while in dfu mode, If i try to re-upload it in cdc mode it fails with an error.

While if I first open the serial monitor and wait for it to receive some messages, and then try to upload it works.

### Sketch

```cpp
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("Hello");
  delay(50);
}
```

### Debug Message

```plain
FQBN: esp32:esp32:lolin_s2_mini:DebugLevel=verbose
Using board 'lolin_s2_mini' from platform in folder: /home/sonic/.arduino15/packages/esp32/hardware/esp32/3.1.0-RC1
Using core 'esp32' from platform in folder: /home/sonic/.arduino15/packages/esp32/hardware/esp32/3.1.0-RC1

/usr/bin/env bash -c "[ ! -f \"/home/sonic/Desktop/music/arduino/adc_test\"/partitions.csv ] || cp -f \"/home/sonic/Desktop/music/arduino/adc_test\"/partitions.csv \"/tmp/arduino/sketches/5047A9EEBC05BDA2B2491CADD4ACBBF5\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/tmp/arduino/sketches/5047A9EEBC05BDA2B2491CADD4ACBBF5\"/partitions.csv ] || [ ! -f \"/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.1.0-RC1/variants/lolin_s2_mini\"/partitions.csv ] || cp \"/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.1.0-RC1/variants/lolin_s2_mini\"/partitions.csv \"/tmp/arduino/sketches/5047A9EEBC05BDA2B2491CADD4ACBBF5\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/tmp/arduino/sketches/5047A9EEBC05BDA2B2491CADD4ACBBF5\"/partitions.csv ] || cp \"/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.1.0-RC1\"/tools/partitions/default.csv \"/tmp/arduino/sketches/5047A9EEBC05BDA2B2491CADD4ACBBF5\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/home/sonic/Desktop/music/arduino/adc_test\"/bootloader.bin ] && cp -f \"/home/sonic/Desktop/music/arduino/adc_test\"/bootloader.bin \"/tmp/arduino/sketches/5047A9EEBC05BDA2B2491CADD4ACBBF5\"/adc_test.ino.bootloader.bin || ( [ -f \"/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.1.0-RC1/variants/lolin_s2_mini\"/bootloader.bin ] && cp \"/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.1.0-RC1/variants/lolin_s2_mini\"/bootloader.bin \"/tmp/arduino/sketches/5047A9EEBC05BDA2B2491CADD4ACBBF5\"/adc_test.ino.bootloader.bin || python3 \"/home/sonic/.arduino15/packages/esp32/tools/esptool_py/4.6\"/esptool.py --chip esp32s2 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o \"/tmp/arduino/sketches/5047A9EEBC05BDA2B2491CADD4ACBBF5\"/adc_test.ino.bootloader.bin \"/home/sonic/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-466a392a/esp32s2\"/bin/bootloader_qio_80m.elf )"
esptool.py v4.6
Creating esp32s2 image...
Merged 2 ELF sections
[... 41 lines cut ...]

    esptool._main()
  File "/home/sonic/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/__init__.py", line 1064, in _main
    main()
  File "/home/sonic/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/__init__.py", line 678, in main
    esp = esp or get_default_connected_device(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sonic/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/__init__.py", line 953, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/home/sonic/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/loader.py", line 642, in connect
    last_error = self._connect_attempt(reset_strategy, mode)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sonic/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/loader.py", line 527, in _connect_attempt
    reset_strategy()  # Reset the chip to bootloader (download mode)
    ^^^^^^^^^^^^^^^^
  File "/home/sonic/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/reset.py", line 83, in __call__
    self._setDTRandRTS(False, False)
  File "/home/sonic/.arduino15/packages/esp32/tools/esptool_py/4.6/esptool/reset.py", line 58, in _setDTRandRTS
    fcntl.ioctl(self.port.fileno(), TIOCMSET, struct.pack("I", status))
OSError: [Errno 71] Protocol error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2024-10-21:
@SuGlider Please help with triage, thanks

SuGlider, 2024-10-30:
I'm not sure how to test the issue.

I can help by describing how the S2 USB is suppose to work:
- S2 USB port only work using the TinyUSB  stack. It is enabled in Arduino when "IDE Menu->Tools->USB CDC On Boot:" is "Enabled".
- DFU Mode can also be activated using the IDE Menu, but this is also part of TinyUSB.
- Once "USB CDC On Boot" is enabled and a new sketch is downloaded and executed, the port will show up in the OS. It can be used to download a new sketch that keeps "USB CDC On Boot" enbled.
- Lolin S2 Mini has only yhe USB port available. In that case when the firmware it already has in its flash has not "USB CDC On Boot: Enabled", it is necessary to put the S2 into doenload mode. This is achieved by holding the BOOT button pressed and pulsing the EN/RESET button once. When this is done, a USB port will be enumerated in the OS and it can be used to download a new firmware. 

Please describe the steps you have followed when you see the issue. I can replicate them here and check what is going on.
Thanks.

SuGlider, 2024-10-30:
Not sure if this is the issue, but the USB CDC port only shows up (is enumerated by the OS) when CDC is enabled and DFU is disabled.

SuGlider, 2024-10-30:
Yes, indeed something has changed from 3.0.7 to 3.1.0-RC1.
S2 could have both CDC and DFU enabled and CDC would work fine in 3.0.7.
With 3.1.0-RC1 and RC2 when CDC and DFU are enabled, CDC doesn't work and the host doesn't enumerate the USB port.

@me-no-dev -- would you know about any change that would result in blocking CDC when DFU is enabled?

Sonic803, 2024-10-30:
[image]
These are the setting that I am using, even the Lolin s2 mini settings don't work.

Using the 3.0.7 version doesn't solve the issue.

> I'm using the Lolin s2 mini board in arduino ide, with usb cdc enabled . After uploading this program while in dfu mode, If i try to re-upload it in cdc mode it fails with an error.

When I said this I meant that the first time I upload the firmware while in download mode (GPIO0 low while toggling reset).

What I actually do to replicate this issue:
1.  Set Board Settings as listed above
2. Connect the board in download mode
3. Upload the sketch (it works)
4. Reset
5. If now I try to upload the sketch another time it won't work
6. If I try to upload it with the serial monitor running and printing messages it works.

me-no-dev, 2024-10-31:
@SuGlider 3.0.7 and RC2 should have the exact same TinyUSB setup

BotSpace, 2024-12-02:
I have this this same problem.

Using core 2.0.17, with CDC on boot enabled, I put the ESP32 in boot mode, upload a simple blink sketch, and then power cycle the device. I can then upload as much as I want over USB in this case.

When I update to 3.0.7, and try to upload with the same process, after I power cycle the device upload fails every time.

Jason2866, 2025-03-13:
please try with actual Arduino core 3.1.3 and 3.2.0-rc2.

Sonic803, 2025-03-14:
I tried with both 3.1.3 and 3.2.0-rc2.
In both cases upload only works if the serial monitor is open.
This is the error i get on  3.2.0-rc2 when uploading without the serial monitor open. (The same error is present in 3.1.3)

```

FQBN: esp32:esp32:esp32s2:CDCOnBoot=cdc,UploadMode=cdc
Using board 'esp32s2' from platform in folder: /home/sonic/.arduino15/packages/esp32/hardware/esp32/3.2.0-RC2
Using core 'esp32' from platform in folder: /home/sonic/.arduino15/packages/esp32/hardware/esp32/3.2.0-RC2

/usr/bin/env bash -c "[ ! -f \"/home/sonic/Arduino/sketch_test_esp32s2\"/partitions.csv ] || cp -f \"/home/sonic/Arduino/sketch_test_esp32s2\"/partitions.csv \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/partitions.csv ] || [ ! -f \"/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.2.0-RC2/variants/esp32s2\"/partitions.csv ] || cp \"/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.2.0-RC2/variants/esp32s2\"/partitions.csv \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/partitions.csv ] || cp \"/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.2.0-RC2\"/tools/partitions/default.csv \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/home/sonic/Arduino/sketch_test_esp32s2\"/bootloader.bin ] && cp -f \"/home/sonic/Arduino/sketch_test_esp32s2\"/bootloader.bin \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/sketch_test_esp32s2.ino.bootloader.bin || ( [ -f \"/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.2.0-RC2/variants/esp32s2\"/bootloader.bin ] && cp \"/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.2.0-RC2/variants/esp32s2\"/bootloader.bin \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/sketch_test_esp32s2.ino.bootloader.bin || \"/home/sonic/.arduino15/packages/esp32/tools/esptool_py/4.9.dev3\"/esptool --chip esp32s2 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/sketch_test_esp32s2.ino.bootloader.bin \"/home/sonic/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-d4aa25a3-v1/esp32s2\"/bin/bootloader_qio_80m.elf )"
esptool.py v4.8.1
Creating esp32s2 image...
Merged 2 ELF sections
Successfully created esp32s2 image.
/usr/bin/env bash -c "[ ! -f \"/home/sonic/Arduino/sketch_test_esp32s2\"/build_opt.h ] || cp -f \"/home/sonic/Arduino/sketch_test_esp32s2\"/build_opt.h \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/build_opt.h"
/usr/bin/env bash -c "[ -f \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/build_opt.h ] || : > \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/build_opt.h"
/usr/bin/env bash -c ": > '/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048/file_opts'"
/usr/bin/env bash -c "cp -f \"/home/sonic/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-d4aa25a3-v1/esp32s2\"/sdkconfig \"/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048\"/sdkconfig"
Detecting libraries used...
/home/sonic/.arduino15/packages/esp32/tools/esp-x32/2411/bin/xtensa-esp32s2-elf-g++ -c @/home/sonic/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-d4aa25a3-v1/esp32s2/flags/cpp_flags -w -Os -Werror=return-type -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32S2_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32S2_DEV" -DARDUINO_VARIANT="esp32s2" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="linux" -DARDUINO_FQBN="esp32:esp32:esp32s2:UploadSpeed=921600,CDCOnBoot=cdc,MSCOnBoot=default,DFUOnBoot=default,UploadMode=cdc,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_USB_MODE=0 -DARDUINO_USB_CDC_ON_BOOT=1 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 @/home/sonic/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-d4aa25a3-v1/esp32s2/flags/defines -I/home/sonic/Arduino/sketch_test_esp32s2 -iprefix /home/sonic/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-d4aa25a3-v1/esp32s2/include/ @/home/sonic/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-d4aa25a3-v1/esp32s2/flags/includes -I/home/sonic/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-d4aa25a3-v1/esp32s2/qio_qspi/include -I/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.2.0-RC2/cores/esp32 -I/home/sonic/.arduino15/packages/esp32/hardware/esp32/3.2.0-RC2/variants/esp32s2 @/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048/build_opt.h @/home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048/file_opts /home/sonic/.cache/arduino/sketches/A2611C80EA7FB7D0C0CA08C377F61048/sketch/sketch_test_esp32s2.ino.cpp -o /dev/null
Generating function prototypes...
/home/sonic/.arduino15/packages/esp32/tools/esp-x32/2411/bin/xtensa-esp32s2-elf-g++ -c @/home/sonic/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-d4aa25a3-v1/esp32s2/flags/cpp_flags -w -Os -Werror=return-type -w -x c++ -E -CC -DF_CPU=240000000L  [... cut ...]
  File "esptool/__init__.py", line 809, in main
  File "esptool/__init__.py", line 1210, in get_default_connected_device
  File "esptool/loader.py", line 726, in connect
  File "esptool/loader.py", line 602, in _connect_attempt
  File "esptool/reset.py", line 47, in __call__
  File "esptool/reset.py", line 114, in reset
  File "esptool/reset.py", line 89, in _setDTRandRTS
BrokenPipeError: [Errno 32] Broken pipe
[233409] Failed to execute script 'esptool' due to unhandled exception!

Failed uploading: uploading error: exit status 1
```

SuGlider, 2025-03-14:
It may be a problem related to `esptool.py` rights to open the `/dev/ttyACM1` within Linux Arch.
Could you please execute `ls -l /dev/ttyACM1` and check if the file has o+rw?

SuGlider, 2025-03-14:
Is there a `/dev/ttyUSB0` port available?

Sonic803, 2025-03-14:
No, there is no /dev/ttyUSB0, only ACM.
The permissions should be right.
`crw-rw-rw- 1 root uucp 166, 1 Mar 14 21:38 /dev/ttyACM1`
I know that on Wemos S2 mini you can go in boot load mode, but it takes a lot of time to do it every time you upload.
Using esp-idf I don't have this issue, i can do `idf.py flash` and it works every time.

Running
`esptool.py --port /dev/ttyACM1 chip_id`
After flashing the board with esp-idf returns:
```
esptool.py v4.8.1
Serial port /dev/ttyACM1
Connecting........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Chip is ESP32-S2FNR2 (revision v1.0)
Features: WiFi, Embedded Flash 4MB, Embedded PSRAM 2MB, ADC and temperature sensor calibration in BLK2 of efuse V2
Crystal is 40MHz
MAC: cc:8d:a2:88:b5:f8
Uploading stub...
Running stub...
Stub running...
Warning: ESP32-S2 has no Chip ID. Reading MAC instead.
MAC: cc:8d:a2:88:b5:f8
Hard resetting via RTS pin...
```

While after uploading the simple sketch from arduino ide:
```
esptool.py v4.8.1
Serial port /dev/ttyACM1
Connecting...
Traceback (most recent call last):
  File "/home/sonic/.virtualenvs/espenv/bin/esptool.py", line 37, in <module>
    esptool._main()
  File "/home/sonic/.virtualenvs/espenv/lib/python3.12/site-packages/esptool/__init__.py", line 1314, in _main
    main()
  File "/home/sonic/.virtualenvs/espenv/lib/python3.12/site-packages/esptool/__init__.py", line 803, in main
    esp = esp or get_default_connected_device(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sonic/.virtualenvs/espenv/lib/python3.12/site-packages/esptool/__init__.py", line 1193, in get_default_connected_device
    _esp = detect_chip(
           ^^^^^^^^^^^^
  File "/home/sonic/.virtualenvs/espenv/lib/python3.12/site-packages/esptool/cmds.py", line 99, in detect_chip
    detect_port.connect(connect_mode, connect_attempts, detecting=True)
  File "/home/sonic/.virtualenvs/espenv/lib/python3.12/site-packages/esptool/loader.py", line 730, in connect
    last_error = self._connect_attempt(reset_strategy, mode)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sonic/.virtualenvs/espenv/lib/python3.12/site-packages/esptool/loader.py", line 606, in _connect_attempt
    reset_strategy()  # Reset the chip to bootloader (download mode)
    ^^^^^^^^^^^^^^^^
  File "/home/sonic/.virtualenvs/espenv/lib/python3.12/site-packages/esptool/reset.py", line 47, in __call__
    self.reset()
  File "/home/sonic/.virtualenvs/espenv/lib/python3.12/site-packages/esptool/reset.py", line 114, in reset
    self._setDTRandRTS(False, False)
  File "/home/sonic/.virtualenvs/espenv/lib/python3.12/site-packages/esptool/reset.py", line 89, in _setDTRandRTS
    fcntl.ioctl(self.port.fileno(), TIOCMSET, struct.pack("I", status))
BrokenPipeError: [Errno 32] Broken pipe
```
So I guess the issue is within the image flashed by arduino-esp

SuGlider, 2025-03-16:
Thanks for the detailed report. What is the `esptool.py` version that your Linux/IDF uses when it works fine?

Sonic803, 2025-03-16:
My virtualenv (espenv) uses Python 3.12.4 and esptool 4.8.1, arduino uses the same version of esptool

SuGlider, 2025-09-28:
@Sonic803 - This issue is solved with Arduino Core 3.3.1. Please update it and test it.
I have used Arduino IDE 2.3.6 and ESP32 Arduino Core 3.3.1 with an ESP32-S2 with CDC on Boot ENABLED.
In case there is any problem with the first upload, please hold BOOT button pressed, then  press and release RESET button in order to make the ESP32-S2 enter in Download Mode and be able to upload the sketch using USB Port.

This is the sketch used for testing;

``` cpp
void setup() {
  Serial.begin();
  while(!Serial) delay(100);
}

uint32_t c = 0;
void loop() {
  Serial.println(String(c++) + " ==> Hello");
  delay(1000);
}
```

It can be uploaded correctly with esptool v5.1.0 used in ESP32 Arduino Core 3.3.1 package.

@Parsaabasi - This issue shall be solved with the new ESPTOOL in Core 3.3.1.

Sonic803, 2025-10-01:
I tested it with Core 3.3.1 and it works, thanks!

closed 2025-10-01 by SuGlider