espressif/arduino-esp32 #10475: ABOUT ESP32 ETH_CLOCK_GPIO0_IN FOR LAN8720 Go in download mode

hzhh110, 2024-10-17:
### Board

ESP32

### Device Description

I use the external 50Mhz crystal calibrator to provide the GPIO0 and LAN8720 < ETH _ CLOCK _ GPIO0 _ in > with the crystal calibrator at the same time. The communication is good in the network cable mode, but it will enter the download mode randomly when it is powered on. How to deal with this?It is found that the power-on probability may fail to obtain the chip or the URL by using the ETH _ CLOCK _ GPIO0 _ OUT, the ETH _ CLOCK _ GPIO16 _ OUT, and the ETH _ CLOCK _ GPIO17 _ OUT

### Hardware Configuration

I use the external 50Mhz crystal calibrator to provide the GPIO0 and LAN8720 < ETH _ CLOCK _ GPIO0 _ in > with the crystal calibrator at the same time. The communication is good in the network cable mode, but it will enter the download mode randomly when it is powered on. How to deal with this?It is found that the power-on probability may fail to obtain the chip or the URL by using the ETH _ CLOCK _ GPIO0 _ OUT, the ETH _ CLOCK _ GPIO16 _ OUT, and the ETH _ CLOCK _ GPIO17 _ OUT

### Version

v3.0.4

### IDE Name

PlatformIO

### Operating System

MacOS V:12.1

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

LAN8720

### Sketch

```cpp
LAN8720
```

### Debug Message

```plain
LAN8720
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

hzhh110, 2024-10-17:
Want to know how to deal with the ETH _ CLOCK _ GPIO0 _ in to solve the problem of probability entering the download mode?

me-no-dev, 2024-10-17:
Since GPIO0 is the BOOT select pin, maybe your LAN is pulling it LOW sometimes and causing you to go into Download. Have you tried connecting the LAN RST to the ESP EN pin?

closed 2025-03-12 by Parsaabasi