espressif/arduino-esp32 #10478: Any documentation for NetworkManager and NetworkEvents?

hitecSmartHome, 2024-10-17:
### Board

ESP32

### Device Description

ETH_LAN8720

### Hardware Configuration

ETH_LAN8720

### Version

latest master (checkout manually)

### IDE Name

Pioarduino

### Operating System

Windows 10 / 11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I wonder if there is any example or documentation for these new classes

### Sketch

```cpp
#include <Network.h>
#include <NetworkManager.h>

NetworkEvents networkEvents;

void test(){
    networkEvents.onEvent([](arduino_event_id_t event, arduino_event_info_t info) {
        // profit?
    },ARDUINO_EVENT_MAX);
    Network.begin();
}
```

### Debug Message

```plain
no debug log
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-10-17:
See the ethernet examples. NetworkEvents is part of `Network`

hitecSmartHome, 2024-10-22:
@me-no-dev Do you mean here? https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino

I can't see any `Network` example in here.

me-no-dev, 2024-10-22:
https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino#L73

You can see the call to attach to the events. `Network` is `NetworkManager` + `NetworkEvents`

hitecSmartHome, 2024-10-22:
Oh I see. So I don't have to call `Network.begin();`

me-no-dev, 2024-10-22:
no. It's called by the interfaces

hitecSmartHome, 2024-10-24:
Thank you. It works.

closed 2024-10-24 by hitecSmartHome