espressif/arduino-esp32 #10481: Can't get hardwaretimer alarm to fire in 3.0.x

BlackysBoss, 2024-10-17:
### Board

ESP32-S2

### Device Description

Lolin S2-mini

### Hardware Configuration

GPIO 17 (DAC 1) attached to an amplifier.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

No clue, can't select for this board

### PSRAM enabled

yes

### Upload speed

No clue, can't select for this board

### Description

Tried to convert code for the timer from 2.0.17 to the new 3.0.x, but can't get it to work. Even the 'repeat' example in the timerdocs doesn't seem to work.

### Sketch

```cpp
fixedTimerTicks = 4000000 / sound_horn_start.sample_rate;
  
  //V2.0.17 Working
  //fixedTimer = timerBegin(1, 20, true);                        // timer 1, MWDT clock period = 12.5 ns * TIMGn_Tx_WDT_CLK_PRESCALE -> 12.5 ns * 20 -> 250 ns = 0.25 us, countUp
  //timerAttachInterrupt(fixedTimer, &fixedPlaybackTimer, true); // edge (not level) triggered
  //timerAlarmWrite(fixedTimer, fixedTimerTicks, true);          // autoreload true
  //timerAlarmEnable(fixedTimer);                                 // enable
  
  // V3.0.5 Not working, but why?
  fixedTimer = timerBegin(4000000);
  timerAttachInterrupt(fixedTimer, &fixedPlaybackTimer);
  timerAlarm(fixedTimer, fixedTimerTicks, true, 0);         // autoreload true
```

### Debug Message

```plain
No error messages, compiles just fine
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2024-10-21:
@P-R-O-C-H-Y Please help here with triage as well, thanks!

P-R-O-C-H-Y, 2024-10-23:
Hi @BlackysBoss, I have tested the RepeatTimer example on latest version and everything is working fine for me on ESP32-S2.