espressif/arduino-esp32 #10485: ZB_ESP_MACSPLIT_UART: mode is choosen but which host connection mode is not choosen

alimosalo, 2024-10-19:
### Board

ESP32H2

### Device Description

DEVKIT

### Hardware Configuration

GPIO 23(RX-PIN) and GPIO 24(TX-PIN) and GND of the esp32h2-devkit(esp_zigbee_rcp) are connected to the esp32-wroom(esp_zigbee_gateway).

### Version

latest master (checkout manually)

### IDE Name

ESP IDF

### Operating System

windows11

### Flash frequency

96MHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

hi
I'm having problem with esp_zigbee_rcp example at ESP IDF.
it say that the problem is for the host configuration and I changed .host_mode_configuration at ESP_ZB_DEFAULT_HOST_CONFIG()(.h file)   to HOST_CONNECTION_MODE_NONE, HOST_CONNECTION_MODE_CLI_UART, HOST_CONNECTION_MODE_RCP_UART but the problem didn't solve and also i change .rx_pin and .tx_pin to 23 and 24 respectively and i get the same problem too.
 I comment these two command     esp_zb_rcp_init(); esp_zb_rcp_main_loop_iteration(); at esp_zb_task function in  rcp.c and get the same result too

### Sketch

```cpp
// at esp_zigbee_rcp.c
static void esp_zb_task(void *pvParameters)
{
    esp_zb_rcp_init();
    esp_zb_rcp_main_loop_iteration();
} 
// at esp_zigbee_rcp.h
#define ESP_ZB_DEFAULT_HOST_CONFIG()                            \
    {                                                           \
        .host_connection_mode = HOST_CONNECTION_MODE_CLI_UART,  \
        .host_uart_config = {                                   \
            .port = 0,                                          \
            .uart_config =                                      \
                {                                               \
                    .baud_rate = 115200,                        \
                    .data_bits = UART_DATA_8_BITS,              \
                    .parity = UART_PARITY_DISABLE,              \
                    .stop_bits = UART_STOP_BITS_1,              \
                    .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,      \
                    .rx_flow_ctrl_thresh = 0,                   \
                    .source_clk = UART_SCLK_DEFAULT,            \
                },                                              \
            .rx_pin = 23,                       \
            .tx_pin = 24,                       \
        },
```

### Debug Message

```plain
ZB_ESP_MACSPLIT_UART: mode is choosen but which host connection mode is not choosen
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-10-19:
This is Arduino repo. Your zigbee example is not Arduino code.

P-R-O-C-H-Y, 2024-11-14:
Closing as its not related to our repository.

closed 2024-11-14 by P-R-O-C-H-Y