espressif/arduino-esp32 #10486: latest master and release 3.0.5 lib i2c slave compile fail with function implicit error

cqlhupt, 2024-10-20:
### Board

ESP32c3

### Device Description

ESP32c3

### Hardware Configuration

NA

### Version

latest master (checkout manually)

### IDE Name

vscode esp-ide arduino-as-component

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I had check the functions mention by the compile log, they are not been define in the file. I don't know what should I expect.

### Sketch

```cpp
// Copyright 2015-2021 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#include "soc/soc_caps.h"

#if SOC_I2C_SUPPORT_SLAVE
#include <stdint.h>
#include <stdbool.h>
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include <inttypes.h>
#include <string.h>
#include <math.h>

#include "sdkconfig.h"
[... 853 lines cut ...]
-IE:/vstudio/esp32_container/esp-idf/components/esp_bootloader_format/include -IE:/vstudio/esp32_container/esp-idf/components/spiffs/include -IE:/vstudio/esp32_container/esp-idf/components/esp_hid/include -IE:/vstudio/esp32_container/esp-idf/components/esp_psram/include -IE:/vstudio/esp32_container/esp-idf/components/esp_https_ota/include -IE:/vstudio/esp32_container/esp-idf/components/esp_http_client/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/dotprod/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/support/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/support/mem/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/windows/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/windows/hann/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/windows/blackman/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/windows/blackman_harris/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/windows/blackman_nuttall/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/windows/nuttall/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/windows/flat_top/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/iir/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/fir/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/math/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/math/add/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/math/sub/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/math/mul/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/math/addc/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/math/mulc/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/math/sqrt/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/matrix/mul/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/matrix/add/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/matrix/addc/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/matrix/mulc/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/matrix/sub/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/matrix/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/fft/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/dct/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/conv/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/common/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/matrix/mul/test/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/kalman/ekf/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-dsp/modules/kalman/ekf_imu13states/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-zboss-lib/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp-zigbee-lib/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp_insights/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp_diagnostics/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp_modem/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__esp_rainmaker/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__rmaker_common/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/espressif__qrcode/include -IE:/vstudio/esp32_example_project/arduino-as-component_bk/managed_components/joltwallet__littlefs/include -march=rv32imc_zicsr_zifencei  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Og -fno-shrink-wrap -fmacro-prefix-map=E:/vstudio/esp32_example_project/arduino-as-component_bk=. -fmacro-prefix-map=E:/vstudio/esp32_container/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -DARDUINO=10812 -DARDUINO_ESP32C3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"ESP32C3_DEV\" -DARDUINO_VARIANT=\"esp32c3\" -DESP32=ESP32 -MD -MT esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-i2c-slave.c.obj -MF esp-idf\arduino\CMakeFiles\__idf_arduino.dir\cores\esp32\esp32-hal-i2c-slave.c.obj.d -o esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-i2c-slave.c.obj -c E:/vstudio/esp32_example_project/arduino-as-component_bk/components/arduino/cores/esp32/esp32-hal-i2c-slave.c
E:/vstudio/esp32_example_project/arduino-as-component_bk/components/arduino/cores/esp32/esp32-hal-i2c-slave.c: In function 'i2cSlaveInit':      
E:/vstudio/esp32_example_project/arduino-as-component_bk/components/arduino/cores/esp32/esp32-hal-i2c-slave.c:319:3: error: implicit declaration of function 'i2c_ll_set_fifo_mode'; did you mean 'i2c_ll_set_data_mode'? [-Werror=implicit-function-declaration]
  319 |   i2c_ll_set_fifo_mode(i2c->dev, true);
      |   ^~~~~~~~~~~~~~~~~~~~
      |   i2c_ll_set_data_mode
E:/vstudio/esp32_example_project/arduino-as-component_bk/components/arduino/cores/esp32/esp32-hal-i2c-slave.c: In function 'i2c_slave_set_frequency':
[... cut ...]
E:/vstudio/esp32_example_project/arduino-as-component_bk/components/arduino/cores/esp32/esp32-hal-i2c-slave.c:527:3: error: implicit declaration of function 'i2c_ll_set_filter'; did you mean 'i2c_ll_set_tout'? [-Werror=implicit-function-declaration]
  527 |   i2c_ll_set_filter(i2c->dev, 3);
      |   ^~~~~~~~~~~~~~~~~
      |   i2c_ll_set_tout
cc1.exe: some warnings being treated as errors
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-10-21:
You are not using Arduino IDE, but IDF and Arduino as component. It is not clear which version of IDF you are using and I will bet is not the correct one. Current Arduino master branch works with IDF 5.1.x (latest)

Jason2866, 2024-10-21:
Using wrong IDF version. This is shown in log `-DIDF_VER=\"v5.3.1\` So using IDF 5.3.1 with Arduino branch master. This is wrong.

closed 2024-10-21 by Jason2866