espressif/arduino-esp32 #10487: Digital read on UART RX pin

dzungpv, 2024-10-20:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Module

### Hardware Configuration

GPIO 26 as TX PIN
GPIO 27 as RX PIN

### Version

v3.0.4

### IDE Name

Arduino IDE, ESP IDF

### Operating System

Mac OSX 14

### Flash frequency

40mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I use digtalRead(RX_PIN) to detect data collision for UART.
It working fine with Arduino 2.x, but from Arduino 3.x it not working, show error:
```
[ 20956][E][esp32-hal-gpio.c:190] __digitalRead(): IO 27 is not set as GPIO.
```
When I try to add pinMode(INPUT, RX_PIN) after UART init, it is not show error but UART stop working.

### Sketch

```cpp
#define TX_PIN 26
#define RX_PIN 27

setup() {
    Serial1.begin(115200, SERIAL_8N1, RX_PIN, TX_PIN);
    // pinMode(INPUT, RX_PIN); // Add this not show any error but UART not working
}

loop() {
    bool check_can_send = false;
    uint32_t millis_now = millis();
    while (true) {
        if (Serial1.available() > 0 || !digitalRead(RX_PIN)) {
            ESP_LOGE("TAG", "Line busy, not sending yet");
            check_can_send = false;
            break;
        }
        if (millis() - millis_now > 500) {
            check_can_send = true;
            break;
        }
        delay(5);
    }
    if (check_can_send) {
        // Send data
    }
}
```

### Debug Message

```plain
[ 20956][E][esp32-hal-gpio.c:190] __digitalRead(): IO 27 is not set as GPIO.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-10-21:
Yes, this is no longer possible in Arduino for safety reasons. You can use IDF's GPIO API directly if you want to read the level of the pin anyway. `gpio_get_level((gpio_num_t)RX_PIN);`

SuGlider, 2024-11-12:
Closing the issue given that @me-no-dev has answered and there are no more comments.

closed 2024-11-12 by SuGlider