espressif/arduino-esp32 #10489: [3.0.6] Zigbee stack assertion failed common/zb_error_indication.c:81

lboue, 2024-10-21:
### Board

ESP32-C6-WROOM-1

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v3.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

The application crashes and restarts in a loop when I run the Zigbee_On_Off_Light example provided.

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/tree/master/libraries/Zigbee/examples/Zigbee_On_Off_Light
```

### Debug Message

```plain
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xcf8
load:0x4086e610,len:0x2e30
load:0x40875728,len:0x113c
entry 0x4086c410
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C6
  Package           : 0
  Revision          : 0.00
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000052
  Embedded Flash    : No
[... 107 lines cut ...]
40820410: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4200777c
40820430: 0x00000000 0x00000000 0x00000000 0x42007792 0x00000000 0x00000000 0x00000000 0x4080a3ee
40820450: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40820470: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x0005c181 0x40815484
40820490: 0x40815484 0x00000010 0x40812e00 0x00000050 0x00000010 0x00000010 0x4080ea8c 0x4080374e
408204b0: 0x00000000 0x00000010 0x4080ea8c 0x40801b70 0x00000010 0x00000010 0x00000000 0x40800f9e
408204d0: 0x00000000 0x00000010 0x00000000 0x408015aa 0x00000003 0x001e0040 0x0000000b 0x4080ea1c
408204f0: 0x0000000b 0x00000000 0x00000000 0x00000000 0x00000001 0x00000014 0x00000000 0x00000007
40820510: 0x4080ea3c 0x40820544 0x60003000 0x4080b9d0 0x4080ea3c 0x40820554 0xffffffff 0x40820700
40820530: 0x0007a120 0x4082057c 0x4080ea3c 0x4080d28a 0x00000000 0x00010000 0x00000000 0x00000000
40820550: 0x40820540 0x00000000 0x00000005 0x00000000 0x40820550 0x0007a121 0x4080ea1c 0x4080d898
40820570: 0x00000100 0x003eb070 0x408206d0 0x4080ea1c 0x00000000 0x00000030 0x60003000 0x4080ba44
40820590: 0x4080a52c 0x4080966c 0x40820610 0x4080e870 0x4080dc70 0x40022494 0xffffffe0 0x00000007
408205b0: 0x00000019 0x00000000 0x0000002f 0x00000000 0x00000000 0x4202da14 0x4081eccc 0x408083a4
408205d0: 0x4080a52c 0x40809622 0x40820650 0xffffffff 0x4080dc70 0x40022494 0x4081ec04 0x408083a4
408205f0: 0x00000004 0x40815000 0x4081f650 0x00000000 0x4080a598 0x4080a57c 0x40820680 0x4080e870

ELF file SHA256: b22517d3285b6b1a

Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2024-10-21:
@P-R-O-C-H-Y Please help with triage, thanks

P-R-O-C-H-Y, 2024-10-21:
@lboue I will retest and reply :)

lboue, 2024-10-21:
Same issue with an ESP32-H2 

```
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1388.0 KB, type: DATA, subtype: SPIFFS
         zb_storage : addr: 0x003EB000, size:    16.0 KB, type: DATA, subtype: FAT
             zb_fct : addr: 0x003EF000, size:     4.0 KB, type: DATA, subtype: FAT
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Oct 21 2024 12:55:30
  Compile Host OS   : windows
  ESP-IDF Version   : v5.1.4-972-g632e0c2a9f-dirty
  Arduino Version   : 3.0.6
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32H2_DEV
  Arduino Variant   : esp32h2
  Arduino FQBN      : esp32:esp32:esp32h2:UploadSpeed=921600,CDCOnBoot=default,FlashFreq=64,FlashMode=qio,FlashSize=4M,PartitionScheme=zigbee,DebugLevel=debug,EraseFlash=all,JTAGAdapter=default,ZigbeeMode=ed
============ Before Setup End ============
[   651][D][Zigbee_Color_Dimmable_Light.ino:80] setup(): Adding ZigbeeLight endpoint to Zigbee Core
[   660][D][ZigbeeCore.cpp:63] addEndpoint(): Endpoint: 10, Device ID: 0x0102
[   667][D][Zigbee_Color_Dimmable_Light.ino:84] setup(): Calling Zigbee.begin()
[   674][D][ZigbeeCore.cpp:94] zigbeeInit(): Initialize Zigbee stack
[   693][D][ZigbeeCore.cpp:101] zigbeeInit(): Register all Zigbee EPs in list
[   702][I][ZigbeeCore.cpp:109] zigbeeInit(): List of registered Zigbee EPs:
[   709][I][ZigbeeCore.cpp:111] zigbeeInit(): Device type: Color Dimmable Light Device, Endpoint: 10, Device ID: 0x0102
=========== After Setup Start ============
[   724][I][ZigbeeCore.cpp:179] esp_zb_app_signal_handler(): Zigbee stack initialized
[   735][I][ZigbeeCore.cpp:185] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
[   744][I][ZigbeeCore.cpp:192] esp_zb_app_signal_handler(): Start network steering
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   246528 B ( 240.8 KB)
  Free Bytes        :   189776 B ( 185.3 KB)
  Allocated Bytes   :    50496 B (  49.3 KB)
  Minimum Free Bytes:   189768 B ( 185.3 KB)
  Largest Free Block:   184308 B ( 180.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     8 : RMT_TX
     9 : GPIO
    23 : UART_RX[0]
    24 : UART_TX[0]
============ After Setup End =============
Zigbee stack assertion failed common/zb_error_indication.c:81

abort() was called at PC 0x42012dd9 on core 0
Core  0 register dump:
MEPC    : 0x40801e46  RA      : 0x408063d4  SP      : 0x4081f360  GP      : 0x4080d220  
TP      : 0x4080bd98  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x4081f38c  S1      : 0x4081f38c  A0      : 0x4081f398  A1      : 0x4081f37a  
A2      : 0x00000000  A3      : 0x4081f3c5  A4      : 0x00000001  A5      : 0x40814000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x4081082c  
S4      : 0x00000001  S5      : 0x420232a8  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000011  S9      : 0x00000000  S10     : 0x0000000f  S11     : 0x00000007  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  
[... 11 lines cut ...]
4081f480: 0x00000005 0x0000000f 0x4081bbce 0x0000ffff 0x00000005 0x0000000f 0x4081bbce 0x4202336a
4081f4a0: 0xfe3bd100 0xe0798dff 0xc5119599 0xd4ad61f1 0xe39a88ad 0x783937b1 0x00000008 0x00000000
4081f4c0: 0x0000000f 0x00000001 0x4080ec18 0x000000ff 0x0000000f 0x00000000 0x4080ec18 0x42017126
4081f4e0: 0x00000000 0x00000000 0x00000000 0x0000001a 0x00000000 0x00000000 0x408172ec 0x00000000
4081f500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4081f520: 0x00000000 0x00000000 0x00000000 0x42007f60 0x00000000 0x00000000 0x00000000 0x42007f76
4081f540: 0x00000000 0x00000000 0x00000000 0x40808f48 0x00000000 0x00000000 0x00000000 0x00000000
4081f560: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4081f580: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x0002dde9 0x40813bf4 0x40813bf4 0xdbac70cc
4081f5a0: 0x7a814a02 0x0cfd8ba7 0x04e3eacb 0x4a9b4450 0x40c628f0 0x56506bd3 0x3171e1c2 0xbd538940
4081f5c0: 0x0016804a 0x28fdc06c 0xf7bc855e 0xfd42063c 0xd4225f21 0x1e51e15e 0x0ecce3ba 0x7ffe2d5b
4081f5e0: 0x3d6b9a0c 0x4d46a085 0xaef6dd6c 0x1cf9b7e6 0x01496230 0xba1a224e 0x8851c988 0xa3554fdb
4081f600: 0x200a602c 0x5c02f2c1 0x638e95b9 0xedddafef 0x09746042 0x6944bac4 0x01ebdde4 0x06d03a91
4081f620: 0x0a3b6516 0x16b6f718 0x79962c80 0x876d8c45 0x827f2666 0x1b27a057 0x113e5ddc 0x355aac25
4081f640: 0x477125d3 0xa8f3cd16 0xaaf63f3c 0x69bf9ae9 0x95449754 0x103070a1 0x6dfffeae 0x2e9ff678
[... cut ...]
4081f6a0: 0xba8fc2b4 0xe8157fcf 0xbf268c07 0x3fdbd86c 0x1c9896a0 0x6bd85845 0x59a77b3e 0x19517593
4081f6c0: 0x6ac1d0bb 0xda0c9d42 0xbae634b2 0x76c12b1b 0x1221eaeb 0x3338252b 0xc3f08b0e 0x53066f7c
4081f6e0: 0x56890dbb 0x88804a90 0x47dc7e19 0x699dfed1 0x405dc04f 0xc4439b8e 0xc3c6f366 0x6ddd52b1
4081f700: 0x29e4a4c3 0x46607980 0x60959315 0xf8c3bdaf 0x929f1033 0x781dd859 0x5e18f4da 0x75ef656a
4081f720: 0x829f7469 0xf5fc3c92 0x79d5ba3e 0x787cb1d8 0x9c5b047d 0x1f095a72 0x9ed201e2 0xa3b06abb
4081f740: 0x7782a68c 0x79097d06 0xc8f81bdb 0xcdaf9f69 0xaa44372c 0x94543411 0xc9b733e0 0xe4d05746

```

Jason2866, 2024-10-21:
Can not reproduce with H2. Works so far
```
Chip Info:
------------------------------------------
  Model             : ESP32-H2
  Package           : 0
  Revision          : 0.01
  Cores             : 1
  CPU Frequency     : 96 MHz
  XTAL Frequency    : 32 MHz
  Features Bitfield : 0x00000050
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : No
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : Yes
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   246640 B ( 240.9 KB)
  Free Bytes        :   213632 B ( 208.6 KB)
  Allocated Bytes   :    27808 B (  27.2 KB)
  Minimum Free Bytes:   213632 B ( 208.6 KB)
  Largest Free Block:   204788 B ( 200.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 64 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1388.0 KB, type: DATA, subtype: SPIFFS
         zb_storage : addr: 0x003EB000, size:    16.0 KB, type: DATA, subtype: FAT
             zb_fct : addr: 0x003EF000, size:     4.0 KB, type: DATA, subtype: FAT
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Oct 21 2024 13:26:58
  ESP-IDF Version   : v5.1.4-972-g632e0c2a9f-dirty
  Arduino Version   : 3.0.6
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : Espressif ESP32-H2-DevKit
  Arduino Variant   : esp32h2
  Core Debug Level  : 5
  Arduino Runs Core : 0
  Arduino Events on : 0
  CDC On Boot       : 0
============ Before Setup End ============
[   648][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x420026b4
[   659][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 8 successfully set to type GPIO (1) with bus 0x9
[   669][V][esp32-hal-rmt.c:454] rmtInit(): GPIO 8 - TX MODE - MemSize[48] - Freq=10000000Hz
[   678][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type RMT_TX (10) successfully set to 0x42003c6c
[   689][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type RMT_RX (11) successfully set to 0x42003c6c
[   700][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 8 successfully set to type INIT (0) with bus 0x0
[   711][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 8 successfully set to type RMT_TX (10) with bus 0x4084e19c
[   722][V][esp32-hal-rmt.c:305] _rmtWrite(): GPIO: 8 - Request: 24 RMT Symbols - Blocking - Timeout: -1
[   731][V][esp32-hal-rmt.c:306] _rmtWrite(): GPIO: 8 - Currently in Loop Mode: [NO] | Asked to Loop: NO, LoopCancel: NO
[   742][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type GPIO (1) successfully set to 0x420026b4
[   754][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 9 successfully set to type GPIO (1) with bus 0xa
[   763][D][Zigbee_On_Off_Light.ino:63] setup(): Adding ZigbeeLight endpoint to Zigbee Core
[   772][D][ZigbeeCore.cpp:63] addEndpoint(): Endpoint: 10, Device ID: 0x0100
[   779][D][Zigbee_On_Off_Light.ino:67] setup(): Calling Zigbee.begin()
[   785][D][ZigbeeCore.cpp:94] zigbeeInit(): Initialize Zigbee stack
[   807][D][ZigbeeCore.cpp:101] zigbeeInit(): Register all Zigbee EPs in list
[   816][I][ZigbeeCore.cpp:109] zigbeeInit(): List of registered Zigbee EPs:
[   823][I][ZigbeeCore.cpp:111] zigbeeInit(): Device type: On/Off Light Device, Endpoint: 10, Device ID: 0x0100
=========== After Setup Start ============
[   838][V][ZigbeeCore.cpp:287] esp_zb_app_signal_handler(): ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
[   849][I][ZigbeeCore.cpp:179] esp_zb_app_signal_handler(): Zigbee stack initialized
[   860][I][ZigbeeCore.cpp:185] esp_zb_app_signal_handler(): Device started up in  factory-reset mode
[   869][I][ZigbeeCore.cpp:192] esp_zb_app_signal_handler(): Start network steering
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   246640 B ( 240.9 KB)
  Free Bytes        :   190480 B ( 186.0 KB)
  Allocated Bytes   :    50096 B (  48.9 KB)
  Minimum Free Bytes:   190472 B ( 186.0 KB)
  Largest Free Block:   188404 B ( 184.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     8 : RMT_TX
     9 : GPIO
    23 : UART_RX[0]
    24 : UART_TX[0]
============ After Setup End =============
[  3105][I][ZigbeeCore.cpp:243] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
[  6344][I][ZigbeeCore.cpp:243] esp_zb_app_signal_handler(): Network steering was not successful (status: ESP_FAIL)
```

Built using pioarduino with core 3.0.6

lboue, 2024-10-21:
Can you try it with Arduino IDE?

Jason2866, 2024-10-21:
Nope, don't use ArduinoIDE

P-R-O-C-H-Y, 2024-10-21:
@lboue I did test with Arduino IDE on master(3.0.6) and also on the v3.1 branch. Both works fine, no crashes.
Can you try to erase the flash? Maybe that would help.

offbrandmark, 2024-11-04:
Hello there! I appreciate all the work you've been doing with the ESP32-C6. I recently got one of these and tried using the same example OP used, i'm not sure what exactly is breaking but the situation felt similar. I am using ArduinoIDE as well.

```
14:28:30.093 -> Rebooting...
14:28:30.093 -> ESP-ROM:esp32c6-20220919
14:28:30.093 -> Build:Sep 19 2022
14:28:30.136 -> rst:0xc (SW_CPU),boot:0x1c (SPI_FAST_FLASH_BOOT)
14:28:30.136 -> Saved PC:0x4001975a
14:28:30.136 -> SPIWP:0xee
14:28:30.136 -> mode:DIO, clock div:2
14:28:30.136 -> load:0x4086c410,len:0xcf8
14:28:30.136 -> load:0x4086e610,len:0x2e30
14:28:30.136 -> load:0x40875728,len:0x113c
14:28:30.136 -> entry 0x4086c410
14:28:30.771 -> Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
14:28:30.771 -> 
14:28:30.771 -> Core  0 register dump:
14:28:30.771 -> MEPC    : 0x40031c8a  RA      : 0x4080c36e  SP      : 0x4087ee40  GP      : 0x4080dac0  
14:28:30.771 -> TP      : 0x4086ea88  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
14:28:30.771 -> S0/FP   : 0x00000001  S1      : 0x00000011  A0      : 0x00000000  A1      : 0xffffffff  
14:28:30.771 -> A2      : 0x00000001  A3      : 0x7f7f7f7f  A4      : 0x00000000  A5      : 0x00000004  
14:28:30.771 -> A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x000000b6  S3      : 0x4087ee70  
14:28:30.771 -> S4      : 0x4087ee88  S5      : 0x42054ec4  S6      : 0x00000000  S7      : 0x00000000  
14:28:30.771 -> S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
14:28:30.771 -> T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
14:28:30.771 -> MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000000  
14:28:30.771 -> MHARTID : 0x00000000  
14:28:30.771 -> Stack memory:
14:28:30.771 -> 4087ee40: 0x00000004 0x00000000 0x4081614c 0x00000000 0x00000000 0x00000000 0x4081614c 0x40800030
14:28:30.771 -> 4087ee60: 0x00000000 0x4080ea54 0x4080ea50 0x4080eda4 0x00000000 0x4080ea64 0x4087ee5c 0x4080ea68
14:28:30.771 -> 4087ee80: 0x00000000 0x4080e5ac 0x00000000 0x00000000 0x00000000 0x65737361 0x66207472 0x656c6961
14:28:30.771 -> 4087eea0: 0x08203a64 0x00000020 0x40815fa0 0x40802a9e 0x40815510 0x00000140 0x40815fa0 0x4081614c
14:28:30.771 -> 4087eec0: 0x42066a52 0x00000000 0x40815fa0 0x42079e12 0x00000000 0x00000000 0x4087cd70 0x40807a88
14:28:30.771 -> 4087eee0: 0x00000004 0x40815000 0x00000003 0xffffffff 0x40820e50 0x40880000 0x40815510 0x40022574
14:28:30.771 -> 4087ef00: 0x40815510 0x00000054 0x4087cd70 0x40806f9c 0x00000000 0x00000000 0x4087cd70 0x408074d2
14:28:30.771 -> 4087ef20: 0x00000000 0x00000000 0x00000000 0x000000ff 0x40880000 0x40820e50 0x40815510 0x400225ca
14:28:30.771 -> 4087ef40: 0x40815000 0x40820e58 0x40815510 0x4080b0dc 0x40815000 0x00000000 0x40820e58 0x0f99d310
14:28:30.771 -> 4087ef60: 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4080fa30
14:28:30.771 -> 4087ef80: 0x42090000 0x40811a30 0x40810000 0x42054ec4 0x4081ae50 0x40880000 0x40810000 0x42011696
14:28:30.771 -> 4087efa0: 0x00000000 0x00000001 0x00000002 0x4080f000 0x4087f050 0x00000000 0x4087f050 0x4200579e
14:28:30.771 -> 4087efc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40800258 0x4080f2a0 0x4200054c
14:28:30.771 -> 4087efe0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:28:30.771 -> 4087f000: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:28:30.771 -> 4087f020: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:28:30.771 -> 4087f040: 0x4080d2d0 0x4080f000 0x4080f2a0 0x420005d8 0x00000000 0x00000000 0x0000000a 0x00000000
14:28:30.771 -> 4087f060: 0x4080d2d0 0x4080f000 0x4080f000 0x420001ec 0x00000000 0x4080d3d0 0x00000001 0x420026f4
14:28:30.771 -> 4087f080: 0x00000000 0x00000000 0x420030f2 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:28:30.771 -> 4087f0a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x42003118
14:28:30.771 -> 4087f0c0: 0x00000000 0x00000000 0x00000000 0x4080962c 0x00000000 0x00000000 0x00000000 0x00000000
14:28:30.771 -> 4087f0e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
14:28:30.771 -> 4087f100: 0xa5a5a5a5 0xbaad5678 0x00000160 0xabba1234 0x00000154 0x4087eec0 0x00000000 0x40813058
14:28:30.771 -> 4087f120: 0x40813058 0x4087f114 0x40813050 0x00000018 0x00000000 0x00000000 0x4087f114 0x00000000
14:28:30.771 -> 4087f140: 0x00000001 0x4087d104 0x706f6f6c 0x6b736154 0x00000000 0x00000000 0x00000000 0x4087f100
14:28:30.771 -> 4087f160: 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x40815f38 0x40815fa0 0x40816008
14:28:30.771 -> 4087f180: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x42078df8 0x00000000
14:28:30.771 -> 4087f1a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:28:30.771 -> 4087f1c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:28:30.771 -> 4087f1e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:28:30.771 -> 4087f200: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:28:30.771 -> 4087f220: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:28:30.835 -> 

```

P-R-O-C-H-Y, 2024-11-06:
> Hello there! I appreciate all the work you've been doing with the ESP32-C6. I recently got one of these and tried using the same example OP used, i'm not sure what exactly is breaking but the [...]

Hi @offbrandmark Can you set the debug level to "verbose" and post the complete log? Thanks

offbrandmark, 2024-11-08:
Hello @P-R-O-C-H-Y thanks for the reply! I set my debug level to verbose but the serial monitor log is still the same. I'm a bit of a noob when it comes to debugging in arduino, is there a place the logs are saved?  Oh and for the sake of time i've already tested the other Zigbee examples and get the same result. Any help is appreciated, ive noticed some changes had been made to the examples but I haven't tested those yet.

Heres an updated log from the serial monitor. It just prints this over and over until unplugged
`
10:23:00.398 -> load:0x40875728,len:0x1048
10:23:00.398 -> entry 0x4086c410
10:23:01.545 -> Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
10:23:01.545 -> 
10:23:01.545 -> Core  0 register dump:
10:23:01.545 -> MEPC    : 0x40031c8a  RA      : 0x4080c36e  SP      : 0x4087edf0  GP      : 0x4080dac0  
10:23:01.545 -> TP      : 0x4086b888  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
10:23:01.545 -> S0/FP   : 0x00000001  S1      : 0x00000011  A0      : 0x00000000  A1      : 0xffffffff  
10:23:01.545 -> A2      : 0x00000001  A3      : 0x7f7f7f7f  A4      : 0x00000000  A5      : 0x00000004  
10:23:01.545 -> A6      : 0x0000000c  A7      : 0x76757473  S2      : 0x000000b6  S3      : 0x4087ee20  
10:23:01.545 -> S4      : 0x4087ee38  S5      : 0x420580ae  S6      : 0x000c8e5e  S7      : 0x00000000  
10:23:01.545 -> S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
10:23:01.545 -> T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938 
10:23:01.545 -> MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000000  
10:23:01.545 -> MHARTID : 0x00000000  
10:23:01.545 -> 
10:23:01.545 -> Stack memory:
10:23:01.545 -> 4087edf0: 0x00000004 0x00000000 0x4081614c 0x00000000 0x00000000 0x00000000 0x4081614c 0x40800030
10:23:01.545 -> 4087ee10: 0x00000000 0x4080ea54 0x4080ea50 0x4080eda4 0x00000000 0x4080ea64 0x4087ee0c 0x4080ea68
10:23:01.545 -> 4087ee30: 0x00000000 0x4080e5ac 0x00000000 0x00000000 0x00000000 0x65737361 0x66207472 0x656c6961
10:23:01.545 -> 4087ee50: 0x08203a64 0x00000020 0x40815fa0 0x40802a9e 0x40815510 0x00000140 0x40815fa0 0x4081614c
10:23:01.545 -> 4087ee70: 0x42069c3c 0x00000000 0x40815fa0 0x4207cffc 0x00000000 0x00000000 0x4087cd70 0x40807a88
10:23:01.545 -> 4087ee90: 0x00000004 0x40815000 0x00000003 0xffffffff 0x4081fc54 0x40880000 0x40815510 0x40022574
10:23:01.545 -> 4087eeb0: 0x40815510 0x00000054 0x4087cd70 0x40806f9c 0x00000000 0x00000000 0x4087cd70 0x408074d2
10:23:01.545 -> 4087eed0: 0x000c8e5e 0x42080861 0x4087f040 0x000000ff 0x40880000 0x4081fc54 0x40815510 0x400225ca
10:23:01.545 -> 4087eef0: 0x40815000 0x4081fc5c 0x40815510 0x4080b0dc 0x40815000 0x00000000 0x4081fc5c 0x31d37469
10:23:01.545 -> 4087ef10: 0x00000001 0x00000000 0x00000000 0x00000000 0x000c8e5e 0x42080861 0x4087f040 0x4080fa30
10:23:01.545 -> 4087ef30: 0x42093000 0x40811a30 0x40810000 0x420580ae 0x42080000 0x42001864 0x40810000 0x42014880
10:23:01.545 -> 4087ef50: 0x00000000 0x00000062 0x00000001 0x00000000 0x42080000 0x42081000 0x4087f040 0x42008948
10:23:01.545 -> 4087ef70: 0x42004c96 0x00000336 0x42080861 0x0000005e 0x420811ec 0xdb800000 0x4080f2a0 0x4200186a
10:23:01.545 -> 4087ef90: 0x4081898c 0x00000000 0x00000001 0x000000fe 0x00000008 0x00000020 0x00000000 0x00000000
10:23:01.545 -> 4087efb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10:23:01.545 -> 4087efd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10:23:01.545 -> 4087eff0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10:23:01.545 -> 4087f010: 0x00000000 0x000c7487 0x42080000 0x4080f000 0x420801a6 0x4080f000 0x4080f2a0 0x42001aa2
10:23:01.545 -> 4087f030: 0x00000804 0x4080f000 0x4080f000 0x420002b0 0x00000000 0x00000000 0x0000000a 0x00000000
10:23:01.545 -> 4087f050: 0x420801c0 0x4080f000 0x4080f000 0x420002bc 0xffffffff 0x00000004 0x600c5000 0x00000001
10:23:01.545 -> 4087f070: 0x3fe00000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10:23:01.545 -> 4087f090: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10:23:01.545 -> 4087f0b0: 0x00000000 0x40815000 0x42003000 0x4200614a 0x00000000 0x00000000 0x00000000 0x4080962c
10:23:01.545 -> 4087f0d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10:23:01.545 -> 4087f0f0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160 0xabba1234
10:23:01.545 -> 4087f110: 0x00000154 0x4087ede0 0x00000157 0x40813058 0x40813058 0x4087f114 0x40813050 0x00000018
10:23:01.545 -> 4087f130: 0x00000000 0x00000000 0x4087f114 0x00000000 0x00000001 0x4087d104 0x706f6f6c 0x6b736154
10:23:01.545 -> 4087f150: 0x00000000 0x00000000 0x00000000 0x4087f100 0x00000001 0x00000000 0x00000000 0x00000000
10:23:01.545 -> 4087f170: 0x00000000 0x40815f38 0x40815fa0 0x40816008 0x00000000 0x00000000 0x00000001 0x00000000
10:23:01.545 -> 4087f190: 0x00000000 0x4087f278 0x4207bfe2 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10:23:01.545 -> 4087f1b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10:23:01.545 -> 4087f1d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
10:23:01.615 -> 
10:23:01.615 -> 
10:23:01.615 -> 
10:23:01.615 -> ELF file SHA256: 642b3a9b3e739c55
`

offbrandmark, 2024-11-08:
Ah I was using the wrong partition scheme. Duh