espressif/arduino-esp32 #10490: Maximum Data Transfer on HTTP using HTTPClient library

UrviZala, 2024-10-21:
### Board

ESP32-WROOM-32E

### Device Description

Custom board 

### Hardware Configuration

ESP32

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 10 Pro

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am using HTTPClient library by Arduino and library version is 0.5.0. I am sending data in string which is in JSON format and on sending data up to 1KB at a time which is giving me successfully 200 responses, but as length of data increases it is giving me -1 response and wifi also gets disconnect sometimes. In same code if I decrease size of data it is working perfectly. is it possible to post large amount of data(approx 50KB) on http at a time without chunk and if it is how and if it is not then what is maximum limit using same library?

### Sketch

```cpp
if (WiFi.status() == WL_CONNECTED) {
    HTTPClient http;
    http.begin("my_https_url");
    http.setTimeout(30000);
    http.addHeader("Content-Type", "application/json");
    int code = http.POST(data);
    http.end();
  }
```

### Debug Message

```plain
code = -1
```

### Other Steps to Reproduce

I have checked the free heap size while posting data; it might be an issue related to that. However, I need clarification on whether this is happening due to the ESP memory or if there are any other parameters that I may be missing.

Thank you....

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-10-21:
Arduino core 2.0.11 is out of support. Please try with actual core 3.0.6

enjoyneering, 2024-10-22:
The RX TX buffer size of the HTTPClient library for IDE v3.x can be found here - https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/src/HTTPClient.h#L59