espressif/arduino-esp32 #10494: [3.0.6] Unable to compile TinyUSB w/USB-MSC

brentru, 2024-10-21:
### Board

ESP32-S2, ESP32-S3, ESP32-Sx Variants

### Device Description

Adafruit Feather ESP32-S2 

### Hardware Configuration

No hardware is attached

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE, Arduino CLI

### Operating System

macOS, Ubuntu (GH:A)

### Flash frequency

n/a for this issue

### PSRAM enabled

yes

### Upload speed

115200

### Description

The [msc_external_flash.ino](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/MassStorage/msc_external_flash) sketch provided by the Adafruit TinyUSB Arduino Library compiles on 3.0.5 but not 3.0.6. There is a symbol collision between arduino-esp32 due to multiple definitions of the `usbd_edpt_iso_activate` and `usbd_edpt_iso_alloc` functions. 

@hathach  - you may want to look into this as well. But, I'm opening this issue on arduino-esp32 rather than Adafruit_TinyUSB_Arduino because it seems to be isolated to the arduino-esp32 BSP 3.0.6 release. 

### Sketch

```cpp
Sketch is `msc_external_flash.ino` provided by the Adafruit TinyUSB Arduino library: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/examples/MassStorage/msc_external_flash
```

### Debug Message

```plain
Compilation Log on BSP 3.0.6:

Linking everything together...
/Users/brentrubell/Library/Arduino15/packages/esp32/tools/esp-xs2/2302/bin/xtensa-esp32s2-elf-g++ @/Users/brentrubell/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s2/flags/ld_flags @/Users/brentrubell/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s2/flags/ld_scripts -Wl,--Map=/var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/msc_external_flash.ino.map -L/Users/brentrubell/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s2/lib -L/Users/brentrubell/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s2/ld -L/Users/brentrubell/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-632e0c2a/esp32s2/qio_qspi -Wl,--wrap=esp_panic_handler -Wl,--start-group /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/sketch/msc_external_flash.ino.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SPI/SPI.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FreeStack.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/MinimumSerial.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/ExFatLib/ExFatDbg.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/ExFatLib/ExFatFile.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/ExFatLib/ExFatFilePrint.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/ExFatLib/ExFatFileWrite.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/ExFatLib/ExFatFormatter.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/ExFatLib/ExFatName.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/ExFatLib/ExFatPartition.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/ExFatLib/ExFatVolume.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FatLib/FatDbg.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FatLib/FatFile.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FatLib/FatFileLFN.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FatLib/FatFilePrint.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FatLib/FatFileSFN.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FatLib/FatFormatter.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FatLib/FatName.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FatLib/FatPartition.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FatLib/FatVolume.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FsLib/FsFile.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FsLib/FsNew.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/FsLib/FsVolume.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SdCard/SdCardInfo.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SdCard/SdSpiCard.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SdCard/SdioTeensy.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SpiDriver/SdSpiArtemis.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SpiDriver/SdSpiChipSelect.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SpiDriver/SdSpiDue.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SpiDriver/SdSpiESP.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SpiDriver/SdSpiParticle.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SpiDriver/SdSpiSTM32.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SpiDriver/SdSpiSTM32Core.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/SpiDriver/SdSpiTeensy3.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/common/FmtNumber.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/common/FsCache.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/common/FsDateTime.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/common/FsName.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/common/FsStructs.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/common/FsUtf.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/common/PrintBasic.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/common/upcase.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/iostream/StdioStream.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/iostream/StreamBaseClass.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/iostream/istream.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/SdFat_-_Adafruit_Fork/iostream/ostream.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/Adafruit_SPIFlash/Adafruit_FlashCache.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/Adafruit_SPIFlash/Adafruit_SPIFlash.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/Adafruit_SPIFlash/Adafruit_SPIFlashBase.cpp.o /var/folders/ff/dmzflvf52tq9kzvt6g8jglxw0000gn/T/arduino_build_752479/libraries/Adafruit_SPI [... cut ...]
Using library Adafruit_SPIFlash at version 5.0.0 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_SPIFlash 
Using library Adafruit_TinyUSB_Library at version 3.3.1 in folder: /Users/brentrubell/Documents/Arduino/libraries/Adafruit_TinyUSB_Library 
exit status 1
Error compiling for board Adafruit Feather ESP32-S2.
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-10-22:
Looks like you are compiling the whole tinyusb again and that is provided in our core already. Please post in `Adafruit_TinyUSB_Library` repo. Our examples work

closed 2024-10-22 by brentru

luigijava, 2024-10-25:
how did you solve that I have the same problem?

brentru, 2024-10-25:
@luigijava  A new version of the Adafruit TinyUSB has been released, please use the latest version.

luigijava, 2024-10-25:
I resolve upgrade to 3.0.7