espressif/arduino-esp32 #10500: PSRAM stops working when importing `arduino-esp32` into ESP-IDF. Flag `CONFIG_SPIRAM` becomes undefined when `BOARD_HAS_PSRAM` is not set.

cjdell, 2024-10-22:
### Board

Xiao ESP32-S3 Sense

### Device Description

Xiao ESP32-S3 Sense connected to PC with USB-C

### Hardware Configuration

Just USB-C to PC

### Version

latest master (checkout manually)

### IDE Name

VSCode

### Operating System

NixOS

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

460800

### Description

Unfortunately my PSRAM stopped working as soon as I started using the `arduino-esp32` component. This code literally undefines the `CONFIG_SPIRAM` and `CONFIG_SPIRAM_SUPPORT` flag out of existence:

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-psram.h#L29

There is no way to override them back. I even tried defining `BOARD_HAS_PSRAM` as a build flag but it hasn't worked. I haven't found docs on how to set global builds flags nor is there any information on the `BOARD_HAS_PSRAM` flag and any explanation as to why it exists when we already have flags in `sdkconfig` for those.

Manually removing lines 24-31 in `esp32-hal-psram.h` will get PSRAM working again but this is an uncomfortable workaround. 

Thanks

### Sketch

```cpp
#include <Arduino.h>

void setup()
{
  delay(2000);

  // DOES NOT WORK WITHOUT "BOARD_HAS_PSRAM" WHICH IS NOT DEFINED !!!
#if CONFIG_SPIRAM
  ESP_LOGI("main", "CONFIG_SPIRAM DEFINED");
#else
  ESP_LOGI("main", "CONFIG_SPIRAM ** NOT ** DEFINED");
#endif

  ESP_LOGI("main", "PSRAM Size: %i", (int)ESP.getPsramSize());
}

void loop()
{
}
```

### Debug Message

```plain
No CONFIG_SPIRAM defined
PSRAM reports 0 bytes (incorrect)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

cjdell, 2024-10-22:
Better workaround for now, add the following line to `CMakeLists.txt` in the root of your project:

```cmake
add_compile_options(-DBOARD_HAS_PSRAM=1)
```

This may just be a documentation issue. It would be great to have the docs clarified when using `arduino-esp32` directly via the ESP-IDF and not via the Arduino IDE.

me-no-dev, 2024-10-22:
You should not define `BOARD_HAS_PSRAM` yourself. All you need to do is control it through menuconfig and it will work

me-no-dev, 2024-10-22:
Ahh you are correct... need to do some tests

me-no-dev, 2024-10-22:
@cjdell fix is coming [here](https://github.com/espressif/arduino-esp32/pull/10502/files). You will need to remove `add_compile_options(-DBOARD_HAS_PSRAM=1)` once it's applied

me-no-dev, 2024-10-22:
Unfortunately the proposed fix will break many libraries PlatformIO projects. I suspect that documenting the necessity to add `add_compile_options(-DBOARD_HAS_PSRAM)` to the project would be the only way to go.

me-no-dev, 2024-10-22:
BTW this code has been as-is for a very long time. You are the first to notice the issue

cjdell, 2024-10-22:
Thanks for the quick turnaround. I'm surprised using the ESP-IDF directly with the desire to use abundant Arduino libraries is not more common

Documentation seems the best way to go. Also on how to apply custom build flags in general as there is no info on that (that I can see).

me-no-dev, 2024-10-22:
I think I found a way that changes nothing in Arduino and does not delete the defines when using as component. Change is minimal and differentiates wether the prebuilt libs are used (Arduino IDE, PlatformIO) and if so it has the old behavior, else does not change anything. You can see it here if you want to try:  https://github.com/espressif/arduino-esp32/commit/1833eeb1b18d6aa3b27ff9dca71740ed0b22e6dc

kel30a, 2025-02-08:
@cjdell Did this work for you? It did stop CONFIG_SPIRAM and CONFIG_SPIRAM_SUPPORT flags from being undefined, but the esp32s3 was not detecting the spiram (spiramDetected in esp32-hal-psram.c would stay false)

kel30a, 2025-02-14:
Bump @me-no-dev any ideas?

kel30a, 2025-02-16:
Ahh thanks I should have tried that first... 

I was using arduino-esp32 v3.0.4 with ESP IDF v5.1.4

Problem is fixed in arduino-esp32 v3.1.2 with ESP IDF v5.3.2

closed 2025-03-12 by Jason2866