espressif/arduino-esp32 #10506: Download of version 3.0.6 fails

znmeb, 2024-10-22:
### Board

Any ESP32

### Device Description

Not device related

### Hardware Configuration

Not device related

### Version

v3.0.6

### IDE Name

Arduino 2

### Operating System

Fedora Linux

### Flash frequency

Not device related

### PSRAM enabled

no

### Upload speed

Not device related

### Description

Installing version 3.0.6 download fails - size mismatch

```
Failed to install platform: 'esp32:3.0.6'. 13 INTERNAL: Cannot install tool esp32:esp32-arduino-libs@idf-release_v5.1-632e0c2a: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 309895581 != 309891323
```

### Sketch

```cpp
Not sketch related
```

### Debug Message

```plain
Failed to install platform: 'esp32:3.0.6'. 13 INTERNAL: Cannot install tool esp32:esp32-arduino-libs@idf-release_v5.1-632e0c2a: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 309895581 != 309891323
```
```

### Other Steps to Reproduce

Just try to update from 3.0.5 to 3.0.6 and this happens.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

znmeb, 2024-10-22:
I can also reproduce this on Ubuntu 24.04 LTS with the Arduino CLI.

henrypalm, 2024-10-23:
I too am having this issue, win10

me-no-dev, 2024-10-23:
we have just released 3.0.7 that should solve the issue. Please try

SatriaUtamaPA, 2024-10-23:
3.0.7 isn't available on Arduino Board manager on win10

VojtechBartoska, 2024-10-23:
@SatriaUtamaPA what version of Arduino IDE? Do you have a proper URL link in Boards Manager?

henrypalm, 2024-10-23:
> we have just released 3.0.7 that should solve the issue. Please try

I am getting this error still. I believe it may be something I am doing.

Failed to install platform: 'esp32:3.0.7'. 13 INTERNAL: Cannot install tool esp32:esp32-arduino-libs@idf-release_v5.1-632e0c2a: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 309895581 != 309891323

SuGlider, 2024-10-23:
I have just installed 3.0.7 using Win11.

My `IDE Menu->File->Preferences->Additional Boards Manager URLs:` has:
```
https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json
https://espressif.github.io/arduino-esp32/package_esp32_index.json
```

me-no-dev, 2024-10-23:
@SuGlider can you help @henrypalm to delete the local libs file?

SuGlider, 2024-10-23:
@henrypalm - Please go to `C:\Users\<Your User Name>\AppData\Local\Arduino15\staging\packages` folder and delete the `esp32-arduino-libs-idf-release_v5.1-632e0c2a.zip` file.

It shall solve the issue.

SuGlider, 2024-10-23:
@henrypalm - Is this error related to Arduino IDE + Board Manager or to PlatformIO and platformio.ini settings?

henrypalm, 2024-10-23:
I have been using ardunio ide + board manager (version 2.3.4, but i think i updated it from 2.3.3 and while try downgrading to see if it solves)

henrypalm, 2024-10-23:
@SuGlider I have downgraded and get a much simpler error of

[23724] Failed to execute script 'esptool' due to unhandled exception!

SuGlider, 2024-10-23:
Thanks. I'll test the Arduino IDE 2.3.3 and 2.3.4 installation.

henrypalm, 2024-10-23:
in 1.8.19, this is the error i got

Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled, Disabled"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\ItMe\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\ItMe\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\ItMe\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none,ZigbeeMode=default -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\ItMe\AppData\Local\Temp\arduino_build_257904 -warnings=none -build-cache C:\Users\ItMe\AppData\Local\Temp\arduino_cache_743667 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.esptool_py-4.6.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.dfu-util.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.esp-xs3.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esp-xs3\2302 -prefs=runtime.tools.esp-xs3-2302.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esp-xs3\2302 -prefs=runtime.tools.esp-rv32.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.esp-rv32-2302.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.openocd-esp32.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20240821 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20240821.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20240821 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.riscv32-esp-elf-gdb-12.1_20231023.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.esp-xs2.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esp-xs2\2302 -prefs=runtime.tools.esp-xs2-2302.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esp-xs2\2302 -prefs=runtime.tools.esp-x32.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2302 -prefs=runtime.tools.esp-x32-2302.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2302 -prefs=runtime.tools.esp32-arduino-libs.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-632e0c2a -prefs=runtime.tools.esp32-arduino-libs-idf-release_v5.1-632e0c2a.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-632e0c2a -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.xtensa-esp-elf-gdb-12.1_20231023.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.mklittlefs.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -verbose C:\Users\ItMe\Documents\Arduino\3dES_blue_led_no_oled\3dES_blue_led_no_oled.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\ItMe\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\ItMe\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\ItMe\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none,ZigbeeMode=default -vid-pid=10C4_EA60 -ide-version=10819 -build-path C:\Users\ItMe\AppData\Local\Temp\arduino_build_257904 -warnings=none -build-cache C:\Users\ItMe\AppData\Local\Temp\arduino_cache_743667 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\ItMe\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\ItMe\AppData\ [... cut ...]
Uploading stub...

Running stub...

TypeError: argument of type 'NoneType' is not iterable

Stub running...

[3420] Failed to execute script 'esptool' due to unhandled exception!

Changing baud rate to 921600

Changed.

WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.

Configuring flash size...

the selected serial port Configuring flash size...

 does not exist or your board is not connected

henrypalm, 2024-10-23:
fyi, all pins have been disconnected for a while

henrypalm, 2024-10-23:
When i use 
```
python C:\Users\ItMe\AppData\Roaming\Python\Python312\Scripts\esptool.exe --port COM6 erase_flash
```

I end up with 
```
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Erasing flash (this may take a while)...

```

pschatzmann, 2024-10-24:
For be the installation of __any release__ is failing (even after deleting the staging/packages)
```
esp32:esp32-arduino-libs@idf-release_v5.1-632e0c2a
esp32:esp32@3.0.7
Installing esp32:esp32-arduino-libs@idf-release_v5.1-632e0c2a
Failed to install platform: 'esp32:esp32:3.0.7'.
Error: 13 INTERNAL: Cannot install tool esp32:esp32-arduino-libs@idf-release_v5.1-632e0c2a: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index: 309895581 != 30989132
```

VojtechBartoska, 2024-10-24:
Please @all update to version 3.0.7 in your Boards Manager, it should work as expected.

henrypalm, 2024-10-25:
> Please @ALL update to version 3.0.7 in your Boards Manager, it should work as expected.

I am still having issues, but it seems to be related to the EspTool install. I am about to try on a linux machine and will update if it works as normal. I believe my current issues are due to user error (me breaking esptool in my fiddling). What is the best place to support this work?

VojtechBartoska, 2024-11-04:
Hello, seems this issue can be closed.

3.0.7 was released and there is nothing else to be done.

closed 2024-11-04 by VojtechBartoska