espressif/arduino-esp32 #10518: Arduino-ESP32 version 3.0.x - ESP32-CAM OTA Compile failure

swordfish1945, 2024-10-24:
### Board

ESP32-CAM

### Device Description

ESP32 provides streaming video while executing commands from the web interface. Pan and Tilt servos are attached to the ESP32-CAM and controlled by commands from the web interface which are processed by the ESP32-CAM module.

### Hardware Configuration

Two servos are attached to GPIO pins 14 and 15. ESP32-CAM flash light is connected to GPIO pin 4.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 11 PRO

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

OTA speed?

### Description

Compiling and Uploading a sketch OTA. Expected sketch updates to be compiled without errors and uploaded to ESP32-CAM module.

### Sketch

```cpp
Arduino OTA update OTA.h unmodified.
```

### Debug Message

```plain
HttpsOTAUpdate.cpp:47:35: error: cannot convert 'const esp_http_client_config_t*' to 'const esp_https_ota_config_t*'
47 | esp_err_t ret = esp_https_ota((const esp_http_client_config_t *)param);
```

### Other Steps to Reproduce

Rolled back ESP32 to version 2.0.17 and compilation and OTA upload completes successfully with no errors.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-10-24:
Major changes in Arduino Core 3.0.x. The WebCam code you are using is probably not compatible with the new version.

swordfish1945, 2024-10-24:
Sorry, it's not the web cam code but the Update library code. The camera
works fine and has been for two years. I made some changes to my code and
the only way to get those changes onto the board is over-the-air. The
camera is mounted externally on my house and is a pain to bring it in to do
a USB update.
II rolled the ESP32 core back to version 2.0.17 and was able to do the
update.
I have opened up a bug report with espressif because the Update library is
under their management.

me-no-dev, 2024-10-24:
Can you please provide minimal example code with how you've setup your OTA? We do not need the camera code or web server, just the Update code that causes the problem

me-no-dev, 2024-10-24:
I think something went wrong with your installation, because the offending line is different on our end: https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/src/HttpsOTAUpdate.cpp#L52
Can you please try to install 3.0.7 and see how that goes then.

swordfish1945, 2024-10-24:
It is an OTA.h I inherited from another developer:

#ifdef ESP32
#include <WiFi.h>
#include <ESPmDNS.h>
#else
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#endif

#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#if defined(ESP32_RTOS) && defined(ESP32)
void ota_handle( void * parameter ) {
  for (;;) {
    ArduinoOTA.handle();
    delay(3500);
  }
}
#endif

void setupOTA(const char* nameprefix, const char* ssid, const char*
password) {
  // Configure the hostname
  uint16_t maxlen = strlen(nameprefix) + 7;
  char *fullhostname = new char[maxlen];
  uint8_t mac[6];
  WiFi.macAddress(mac);
  snprintf(fullhostname, maxlen, "%s-%02x%02x%02x", nameprefix, mac[3],
mac[4], mac[5]);
  ArduinoOTA.setHostname(fullhostname);
  delete[] fullhostname;

  // Configure and start the WiFi station
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  // Wait for connection
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 3232
  // ArduinoOTA.setPort(3232); // Use 8266 port if you are working in
Sloeber IDE, it is fixed there and not adjustable

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
//NOTE: make .detach() here for all functions called by Ticker.h library -
not to interrupt transfer process in any way.
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS
using SPIFFS.end()
    Serial.println("Start updating " + type);
  });

  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });

  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });

  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("\nAuth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("\nBegin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("\nConnect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("\nReceive Failed");
    else if (error == OTA_END_ERROR) Serial.println("\nEnd Failed");
  });

  ArduinoOTA.begin();

  Serial.println("OTA Initialized");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

#if defined(ESP32_RTOS) && defined(ESP32)
  xTaskCreate(
    ota_handle,          /* Task function. */
    "OTA_HANDLE",        /* String with name of task. */
    10000,            /* Stack size in bytes. */
    NULL,             /* Parameter passed as input of the task */
    1,                /* Priority of the task. */
    NULL);            /* Task handle. */
#endif
}

swordfish1945, 2024-10-24:
Here is code from HttpsOTAUpdate.cpp. It is different:

`void https_ota_task(void *param)
{
    if(ota_status) {
        xEventGroupSetBits(ota_status, OTA_UPDATING_BIT);
        xEventGroupClearBits(ota_status, OTA_IDLE_BIT);
    }
    esp_err_t ret = esp_https_ota((const esp_http_client_config_t *)param);
    //esp_err_t ret = esp_https_ota(const esp_https_ota_config_t *ota_config);
    if(ret == ESP_OK) {
        if(ota_status) {
            xEventGroupClearBits(ota_status, OTA_UPDATING_BIT);
            xEventGroupSetBits(ota_status, OTA_SUCCESS_BIT);
        }
    } else {
        if(ota_status) {  
            xEventGroupClearBits(ota_status, OTA_UPDATING_BIT);
            xEventGroupSetBits(ota_status, OTA_FAIL_BIT);
        }
    }
    vTaskDelete(NULL);
}
`

swordfish1945, 2024-10-24:
Sorry, having problems adding the code. Trying to use the code markers but not working.

swordfish1945, 2024-10-24:
`void https_ota_task(void *param)
{
    if(ota_status) {
        xEventGroupSetBits(ota_status, OTA_UPDATING_BIT);
        xEventGroupClearBits(ota_status, OTA_IDLE_BIT);
    }
    esp_err_t ret = esp_https_ota((const esp_http_client_config_t *)param);
    //esp_err_t ret = esp_https_ota(const esp_https_ota_config_t *ota_config);
    if(ret == ESP_OK) {
        if(ota_status) {
            xEventGroupClearBits(ota_status, OTA_UPDATING_BIT);
            xEventGroupSetBits(ota_status, OTA_SUCCESS_BIT);
        }
    } else {
        if(ota_status) {  
            xEventGroupClearBits(ota_status, OTA_UPDATING_BIT);
            xEventGroupSetBits(ota_status, OTA_FAIL_BIT);
        }
    }
    vTaskDelete(NULL);
}

swordfish1945, 2024-10-24:
OK, there is a difference for sure.

swordfish1945, 2024-10-24:
Here is library version info:
name=Update
version=2.0.0
author=Hristo Gochkov
maintainer=Hristo Gochkov ***@***.***>
sentence=ESP32 Sketch Update Library
paragraph=
category=Other
url=
architectures=esp32

closed 2024-10-24 by swordfish1945

swordfish1945, 2024-10-24:
I see the Github version is 3.0.5.
For some reason the Arduino IDE 2.3.3 library manager does not show the
library ' Update'.

swordfish1945, 2024-10-24:
Re-installed ESP32 3.0.7. Found two versions of Upate library. Deleted the
oldest version, now all is well.
Thanks for your help and sorry for screwing up messages.
Considered closed.