espressif/arduino-esp32 #10519: Crash when reconnecting to wifi

Laxilef, 2024-10-24:
### Board

ESP32-D0WD-V3

### Device Description

nodemcu32, wemos d1 mini32

### Hardware Configuration

-

### Version

v3.0.6

### IDE Name

platformio

### Operating System

win 11

### Flash frequency

40 mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Crash occurs when the connection is lost and an attempt is made to reconnect to Wi-Fi. However, the problem is difficult to reproduce, as it is floating and does not happen with all routers.
During reconnection only STA works, without AP. Therefore it is strange that errors are related to DHCP server.

### Sketch

```cpp
-
```

### Debug Message

```plain
**Crash 1, d1_mini32**
Software reset due to exception/panic
Backtrace:

.\xtensa-esp32-elf-addr2line.exe -pfiaC -e ".\firmware_d1_mini32_1.4.6-beta4.elf" 0x4015B2E0 0x4015B2DD 0x4012FA31 0x40132E1D 0x40137CA7 0x40127EB1
 0x40098F7E
0x4015b2e0: parse_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:923
 (inlined by) handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:1122
 (inlined by) handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:1038
0x4015b2dd: parse_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:925
 (inlined by) handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:1122
 (inlined by) handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:1038
0x4012fa31: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c:404
0x40132e1d: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:746
0x40137ca7: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186
0x40127eb1: tcpip_thread_handle_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:174
 (inlined by) tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:148
0x40098f7e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

[... 66 lines cut ...]
 (inlined by) nd6_get_next_hop_addr_or_queue at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv6/nd6.c:2296
0x40127acd: lwip_select at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c:1973 (discriminator 1)
0x40098f7e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

**Crash 6, d1_mini32**
Reset due to other watchdogs
Backtrace:
```
.\xtensa-esp32-elf-addr2line.exe -pfiaC -e ".\firmware_d1_mini32_1.4.6-beta4.elf" 0x4012A1F3 0x4012A1F0 0x4012E7A5 0x4013ACE8 0x4013BA2B 0x40127E7A 0x40098F7E
0x4012a1f3: pbuf_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c:230
0x4012a1f0: pbuf_alloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c:230
0x4012e7a5: tcp_pbuf_prealloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_out.c:263
 (inlined by) tcp_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_out.c:610
0x4013ace8: lwip_netconn_do_writemore at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_msg.c:1794
0x4013ba2b: lwip_netconn_do_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_msg.c:1921 (discriminator 2)
0x40127e7a: tcpip_thread_handle_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:202
 (inlined by) tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:148
0x40098f7e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Laxilef, 2024-10-24:
I found similar issues, but they are old and were stated to have been fixed in the latest (at that time) version:
https://github.com/espressif/arduino-esp32/issues/5453
https://github.com/espressif/arduino-esp32/issues/4961
https://github.com/espressif/arduino-esp32/issues/3599

I realize that without a sketch it is difficult to understand what the problem is, but this is a large firmware and the problem does not appear in all cases. I can upload ELF files if that helps.

Any information would be helpful, thanks!

me-no-dev, 2024-10-25:
To me it looks like there is something specific to your router. It sends some sort of packet that breaks the network stack. This would be very interesting to figure out as you have more than one chip to fail. We do not see this behavior though, so I doubt we could even replicate. What is your network setup there? Router make and model?

Laxilef, 2024-10-25:
We use various routers: Mikrotik hAP ac2, Mikrotik hAP ac Lite, Xiaomi MiWiFi 3G (R3G).
Today I tried to build firmware with [core 3.1.0 based on IDF 5.3.0](https://github.com/espressif/arduino-esp32/releases/tag/3.1.0-RC1), and got a crash with "Reset due to other watchdogs" when disconnecting from the AP:

```
.\xtensa-esp32-elf-addr2line.exe -pfiaC -e ".\firmware_d1_mini32_1.4.6-core310.elf" 0xFFFFFFFD 0x4015B32D 0x4012A6C8 0x401263CA 0x40127C08 0x40127CD2 0x400F0E5E 0x3FFC4684 0x400F11D6 0x401C9237 0x401C919D 0x400F03EA 0x400EF5C3 0x401C884A 0x401C85C5 0x40097082
0xfffffffd: ?? ??:0
0x4015b32d: esp_pbuf_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_netif/lwip/netif/esp_pbuf_ref.c:36
0x4012a6c8: pbuf_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c:768
 (inlined by) pbuf_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/pbuf.c:725
0x401263ca: lwip_recv_tcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c:1010
0x40127c08: lwip_recvfrom at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c:1218
0x40127cd2: lwip_recv at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/sockets.c:1283
0x400f0e5e: NetworkClientRxBuffer::fillBuffer() at C:/Users/Yurii/.platformio/packages/framework-arduinoespressif32-libs/esp32/include/lwip/include/lwip/sockets.h:38
 (inlined by) NetworkClientRxBuffer::fillBuffer() at C:/Users/Yurii/.platformio/packages/framework-arduinoespressif32/libraries/Network/src/NetworkClient.cpp:79
0x3ffc4684: ?? ??:0
0x400f11d6: NetworkClient::read(unsigned char*, unsigned int) at C:/Users/Yurii/.platformio/packages/framework-arduinoespressif32/libraries/Network/src/NetworkClient.cpp:475 (discriminator 1)
0x401c9237: NetworkClient::read() at C:/Users/Yurii/.platformio/packages/framework-arduinoespressif32/libraries/Network/src/NetworkClient.cpp:362
0x401c919d: MqttClient::clientRead() at D:\Projects\otgateway/.pio/libdeps/d1_mini32/ArduinoMqttClient/src/MqttClient.cpp:1145
0x400f03ea: MqttClient::poll() at D:\Projects\otgateway/.pio/libdeps/d1_mini32/ArduinoMqttClient/src/MqttClient.cpp:422
0x400ef5c3: MqttTask::loop() at D:\Projects\otgateway/src/MqttTask.h:205
0x401c884a: Task::loopWrapper() at D:\Projects\otgateway/.pio/libdeps/d1_mini32/ESP32Scheduler/src/Task.h:83
0x401c85c5: Task::xLoopWrapper(void*) at D:\Projects\otgateway/.pio/libdeps/d1_mini32/ESP32Scheduler/src/Task.h:48 (discriminator 1)
0x40097082: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```
I can assume that [ArduinoMqttClient](https://github.com/arduino-libraries/ArduinoMqttClient) tried to read data when the connection to the AP was lost, but I can't figure out why this causes a crash. Is there any additional testing I can do?

Laxilef, 2024-10-28:
Hi @me-no-dev 
I'm back with news. I tried 3.1.0 RC2 and it seems that this error is not there. At least there have been no crashes so far.

me-no-dev, 2024-10-28:
Awesome!

VojtechBartoska, 2024-11-04:
Hello, closing this as solved, if needed, feel free to reopen the ticket.

closed 2024-11-04 by VojtechBartoska

Laxilef, 2025-01-26:
@VojtechBartoska Unfortunately the problem has returned again :(

Board: nodemcu 32 (ESP32-D0WD-V3), core 3.1.1, sdk v5.3.2-282-gcfea4f7c98-dirty
Stack trace:
```
0x40162e28: parse_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:991
 (inlined by) handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:1190
 (inlined by) handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:1106
0x40162e25: parse_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:993
 (inlined by) handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:1190
 (inlined by) handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c:1106
0x40136a45: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c:404
0x40139eb3: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:746
0x4013efca: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186
0x4012ea61: tcpip_thread_handle_msg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:174
 (inlined by) tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:148
0x40096eea: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

What tests can I run to localize the problem? What could be breaking the network stack?