espressif/arduino-esp32 #10526: Crash with error message: "Required to lock TCPIP core functionality"

mathieucarbou, 2024-10-25:
### Board

N/A

### Device Description

N/A

### Hardware Configuration

N/A

### Version

latest development Release Candidate (RC-X)

### IDE Name

Pioarduino

### Operating System

N/A

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

N/A

### Description

I have several apps that I keep updated to lats Arduino RC.

[3.1.0-RC1](https://github.com/espressif/arduino-esp32/tree/3.1.0-RC1) work fine.

```
[env]
build_type = release
framework = arduino
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10-rc1/platform-espressif32.zip
```

Since RC2 (3.1.0-RC2), I am observing several crashes with always the same error.

```
[env]
build_type = release
framework = arduino
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.10-rc2/platform-espressif32.zip
```

```
[  3555][D][ESPCONNECT][1][arduino_events] [NETWORK_CONNECTING] WiFiEvent: ARDUINO_EVENT_WIFI_STA_GOT_IP
[  3558][D][ESPCONNECT][1][arduino_events] State: NETWORK_CONNECTING => NETWORK_CONNECTED
[  3568][D][JSY-UDP][1][arduino_events] NetworkState: NETWORK_CONNECTING => NETWORK_CONNECTED
[  3579][I][JSY-UDP][1][arduino_events] Connected with IP address 192.168.125.130
[  3579][I][JSY-UDP][1][core] Enable Network Services...
[  3589][I][JSY-UDP][1][core] Enable Web Server...

assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)

Backtrace: 0x400832e5:0x3ffb23e0 0x4008c429:0x3ffb2400 0x400926d6:0x3ffb2420 0x400fb4cb:0x3ffb2550 0x400fb645:0x3ffb2570 0x40173517:0x3ffb2590 0x400dd44d:0x3ffb25e0 0x400d41b3:0x3ffb2600 0x400d4245:0x3ffb2640 0x400e01b5:0x3ffb2660 0x400e0440:0x3ffb2690 0x400e0481:0x3ffb26c0 0x400e04a5:0x3ffb26e0 0x4008cefe:0x3ffb2700
  #0  0x400832e5 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:463
[... 65 lines cut ...]
      (inlined by) std::enable_if<is_invocable_r_v<void, networkConfigTask::{lambda(void*)#1}&, void*>, void>::type std::__invoke_r<void, networkConfigTask::{lambda(void*)#1}&, void*>(networkConfigTask::{lambda(void*)#1}&, void*&&) at /Users/mat/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/invoke.h:111
      (inlined by) std::_Function_handler<void (void*), networkConfigTask::{lambda(void*)#1}>::_M_invoke(std::_Any_data const&, void*&&) at /Users/mat/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/std_function.h:290
  #8  0x4010582a in std::function<void (void*)>::operator()(void*) const at /Users/mat/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/std_function.h:591
      (inlined by) Mycila::Task::_run(long long const&) at .pio/libdeps/pro-esp32/MycilaTaskManager/src/MycilaTaskManager.cpp:417
  #9  0x40105aec in Mycila::Task::tryRun() at .pio/libdeps/pro-esp32/MycilaTaskManager/src/MycilaTaskManager.cpp:334
  #10 0x40105b2d in Mycila::TaskManager::loop() at .pio/libdeps/pro-esp32/MycilaTaskManager/src/MycilaTaskManager.cpp:106
  #11 0x40105b51 in Mycila::TaskManager::_asyncTaskManager(void*) at .pio/libdeps/pro-esp32/MycilaTaskManager/src/MycilaTaskManager.cpp:196
  #12 0x4008fad2 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mathieucarbou, 2024-10-26:
I suspected this is caused by: `CONFIG_LWIP_CHECK_THREAD_SAFETY` being activated

Also: https://github.com/espressif/arduino-esp32/pull/10415

As I understand, `CONFIG_LWIP_CHECK_THREAD_SAFETY` will fail if we do not properly lock ?

That is not a bad check per see, but right now, activating it could break a lof of downstream projects, but also Arduino components right ? Example of the stack trace above, which lies in the sntp fature.

mathieucarbou, 2024-10-26:
Another ref: https://github.com/espressif/esp-matter/issues/853#issuecomment-1971065330

me-no-dev, 2024-10-26:
It is Matter that enabled it..
```
~/Desktop/ESP32/ESP-IDF-5/esp32-arduino-lib-builder/managed_components/espressif__esp_matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:
  220              bool "Enable LwIP Thread safety options"
  221              default y
  222:             select LWIP_TCPIP_CORE_LOCKING
  223              select LWIP_CHECK_THREAD_SAFETY
  224              help
  225                  CHIP SDK performs LwIP core locking before calling an LwIP API.
  226:                 To make the calls thread safe we have to enable LWIP_TCPIP_CORE_LOCKING.
  227                  Here, we are also enabling LWIP_CHECK_THREAD_SAFETY which will assert when
  228                  LwIP code gets called from any other context or without holding the LwIP lock.
```

AsyncUDP in this repo is already made to work with the locks. The fastest way is to have a look and see if you can solve it yourself. Not everything in Async is executed in the LwIP thread, so some locking will now be required. We can not get rid of Matter :)

mathieucarbou, 2024-10-26:
But for the last stack trace, only sntp is involved... 
I am not using AsyncTCP in the second app, but AsyncTcpSock (bsd sockets).

For the first stack trace, do you mean that I should port in AsyncTCP these locks like they are done in AsyncUDP ?

That's impressive the amount of libs, web servers and mqtt clients will break when they will adopt 3.1.0.

I wish AsyncTCP were included in Arduino Core !

me-no-dev, 2024-10-26:
> I wish AsyncTCP were included in Arduino Core !

I do plan to do this. Seems like it will be rather sooner than later 😄

mathieucarbou, 2024-10-26:
I've placed some locks in the AsyncTCP lib, but still have some issues with sntp:

```
[  5494][E][AsyncTCP.cpp:1492] begin(): [AsyncTCP] LOCK_TCPIP_CORE @ .pio/libdeps/pro-esp32/AsyncTCP/src/AsyncTCP.cpp:1492
[  5515][E][AsyncTCP.cpp:1494] begin(): [AsyncTCP] UNLOCK_TCPIP_CORE @ .pio/libdeps/pro-esp32/AsyncTCP/src/AsyncTCP.cpp:1494
6m[  5386][D][YASOLR][1][y-core] No electricity detected by ZCD module
[  5534][E][AsyncTCP.cpp:1526] begin(): [AsyncTCP] LOCK_TCPIP_CORE @ .pio/libdeps/pro-esp32/AsyncTCP/src/AsyncTCP.cpp:1526
[  5546][E][AsyncTCP.cpp:1529] begin(): [AsyncTCP] UNLOCK_TCPIP_CORE @ .pio/libdeps/pro-esp32/AsyncTCP/src/AsyncTCP.cpp:1529
[  5428][I][YASOLR][1][y-pio] Enable NTP

assert failed: sntp_setoperatingmode /IDF/components/lwip/lwip/src/apps/sntp/sntp.c:728 (Required to lock TCPIP core functionality!)

Backtrace: 0x400835fd:0x3ffd7760 0x4008f09d:0x3ffd7780 0x4009565e:0x3ffd77a0 0x4012acd0:0x3ffd78d0 0x40116556:0x3ffd78f0 0x4010276f:0x3ffd7960 0x400ef37e:0x3ffd79c0 0x400ef445:0x3ffd7a00 0x4010586a:0x3ffd7a20 0x40105b2c:0x3ffd7a50 0x40105b6d:0x3ffd7a80 0x40105b91:0x3ffd7aa0 0x4008fba6:0x3ffd7ac0
  #0  0x400835fd in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:463
  #1  0x4008f09d in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:92
  #2  0x4009565e in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:80
  #3  0x4012acd0 in sntp_setoperatingmode at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/apps/sntp/sntp.c:729 (discriminator 1)
  #4  0x40116556 in configTime at /Users/mat/.platformio/packages/framework-arduinoespressif32@src-e589a26bbeefbfd0fcf50e49e574e2c6/cores/esp32/esp32-hal-time.c:53
  #5  0x4010276f in Mycila::NTPClass::sync(char const*, unsigned char) at .pio/libdeps/pro-esp32/MycilaNTP/src/MycilaNTP.cpp:66 (discriminator 1)
```

mathieucarbou, 2024-10-26:
Ok so I've fixed AsyncTCP in https://github.com/mathieucarbou/AsyncTCP/pull/28:

Tested :
- I can spawn 16 concurrent client connection without issue
- I can spawn 16 concurrent requests (tested with ESPASyncWS)

[image]

The only issue now remains sntp, and I have no control over that.

Is it a bug in Arduino Core ? 

`configTime at /Users/mat/.platformio/packages/framework-arduinoespressif32@src-e589a26bbeefbfd0fcf50e49e574e2c6/cores/esp32/esp32-hal-time.c:53`

```
assert failed: sntp_setoperatingmode /IDF/components/lwip/lwip/src/apps/sntp/sntp.c:728 (Required to lock TCPIP core functionality!)

Backtrace: 0x400835fd:0x3ffd7760 0x4008f09d:0x3ffd7780 0x4009565e:0x3ffd77a0 0x4012acd0:0x3ffd78d0 0x40116556:0x3ffd78f0 0x4010276f:0x3ffd7960 0x400ef37e:0x3ffd79c0 0x400ef445:0x3ffd7a00 0x4010586a:0x3ffd7a20 0x40105b2c:0x3ffd7a50 0x40105b6d:0x3ffd7a80 0x40105b91:0x3ffd7aa0 0x4008fba6:0x3ffd7ac0
  #0  0x400835fd in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:463
  #1  0x4008f09d in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:92
  #2  0x4009565e in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:80
  #3  0x4012acd0 in sntp_setoperatingmode at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/apps/sntp/sntp.c:729 (discriminator 1)
  #4  0x40116556 in configTime at /Users/mat/.platformio/packages/framework-arduinoespressif32@src-e589a26bbeefbfd0fcf50e49e574e2c6/cores/esp32/esp32-hal-time.c:53
  #5  0x4010276f in Mycila::NTPClass::sync(char const*, unsigned char) at .pio/libdeps/pro-esp32/MycilaNTP/src/MycilaNTP.cpp:66 (discriminator 1)
```

NicoWallmeier, 2024-10-26:
From my point of view the sntp bug is a Arduino Core bug.

As a workaround I modified my sources in the following way:
```
#ifdef CONFIG_LWIP_TCPIP_CORE_LOCKING
        if (!sys_thread_tcpip(LWIP_CORE_LOCK_QUERY_HOLDER))
            LOCK_TCPIP_CORE();
#endif
        configTzTime(m_timeZone.c_str(), m_ntpServer.c_str());
#ifdef CONFIG_LWIP_TCPIP_CORE_LOCKING
        if (sys_thread_tcpip(LWIP_CORE_LOCK_QUERY_HOLDER))
            UNLOCK_TCPIP_CORE();
#endif
```
My sketch is running fine now with Arduino Core 3.1.0-rc2, the mentioned workaround and the fix in AsyncTCP...

mathieucarbou, 2024-10-26:
I have applied a similar patch and I confirm it works too.

me-no-dev, 2024-10-26:
can you please post a patch to the AsyncTCP repo and the ntp one to this repo? I will merge them asap

mathieucarbou, 2024-10-26:
For NTP, yes, will do later.

For AsyncTCP, it is done: https://github.com/mathieucarbou/AsyncTCP.

I don't see the point in sending PRs to the original repo, except if there is a clear motivation to "resurrect" it and keep it alive and maintained, with a few collaborators added in the project, who can also deploy to registries (Arduino and Pio).

FYI the [diff with original repo](https://github.com/me-no-dev/AsyncTCP/compare/master...mathieucarbou:AsyncTCP:main)

There has been many fixes / improvements from ESPHome, community and me. At any point these can be merge back to the original repo.

microfoundry, 2024-10-26:
I caught this one while chasing a system crash for a different issue. This is with Mathieu's updated AsyncTCP. The scenario is a simple EspAsyncWebServer (Mathieu's too) webserver/websockets with page data deliver via websockets. And in this case, an iPhone opened webpage (via Chrome) the page is delivered stats via websocket, but is not a page designed to be continually refreshed.  The ws.cleanupClients() is being called and the following occured when the iPhone finally hits the auto-lock timeout. No additional investigation has been performed.

assert failed: tcp_arg /IDF/components/lwip/lwip/src/core/tcp.c:2031 (Required to lock TCPIP core functionality!)

0x40376932: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 463
0x4037f0d9: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c line 92
0x4037f0d9: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c line 92
0x40385ccd: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 80
0x40385ccd: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 80
0x42061555: tcp_arg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 2031
0x42061555: tcp_arg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 2031
0x4203094d: AsyncClient::_error(signed char) at c:\Users\TerryPhillips\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp line 954
0x42030d33: _async_service_task(void*) at c:\Users\TerryPhillips\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp line 1414
0x4037fe52: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c

mathieucarbou, 2024-10-26:
@microfoundry : good catch! Indeed I forgot this one: 
https://github.com/mathieucarbou/AsyncTCP/blob/main/src/AsyncTCP.cpp#L954
I will fix tomorrow morning and issue a new release.

mathieucarbou, 2024-10-26:
@me-no-dev @NicoWallmeier : fyi PR opened for the sntp fix: https://github.com/mathieucarbou/AsyncTCP/blob/main/src/AsyncTCP.cpp#L954

mathieucarbou, 2024-10-27:
@microfoundry : I have released AsyncTCP [v3.2.12](https://github.com/mathieucarbou/AsyncTCP/releases/tag/v3.2.12). Please let me know if you encounter other issues. It is still possible that some other code path need locking but I cannot find them right now.

Nezaemmy, 2024-10-27:
on my code I got this error

assert failed: sntp_stop /IDF/components/lwip/lwip/src/apps/sntp/sntp.c:696 (Required to lock TCPIP core functionality!)

I used these libraries

#include "time.h"
#include "esp_sntp.h"

mathieucarbou, 2024-10-27:
> on my code I got this error
> 
> assert failed: sntp_stop /IDF/components/lwip/lwip/src/apps/sntp/sntp.c:696 (Required to lock TCPIP core functionality!)
> 
> I used these libraries
> 
> #include "time.h" #include "esp_sntp.h"

If you are not using the Arduino API you need to lock yourself the calls to sntp_* like I did: https://github.com/espressif/arduino-esp32/pull/10529/files

microfoundry, 2024-10-27:
> @microfoundry : I have released AsyncTCP [v3.2.12](https://github.com/mathieucarbou/AsyncTCP/releases/tag/v3.2.12). Please let me know if you encounter other issues. It is still possible that some other code path need locking but I cannot find them right now.

Thank you @mathieucarbou - that resolved the issue

me-no-dev, 2024-10-27:
@mathieucarbou it's up to you wether you will send a PR to the original repo. It's up to me wether will be "resurrected", but your ways do leave me wandering many things...

mathieucarbou, 2024-10-27:
> but your ways do leave me wandering many things...

If something is bothering you, you can tell me ;-) 
If something is bothering me, I will tell you (and I did, like above).
Transparency is key, this is OSS, and we all work for the main benefits of others. 
So if you see something wrong to improve, please explain :-)

I just said I cannot afford to spend my little spare time sending a PR to fix all the projects I am not using, cannot use or cannot maintain, which is totally understandable. I would be glad to merge all the changes of my fork back to the original AsyncTCP repo though, but for me to spend time here, I don't want it to be for nothing and I would do it if there's a plan to add collaborators to the project, CI, deployment and also time allocated for maintenance, review, pr merge, etc. Things that I am able to do in my fork for now.

me-no-dev, 2024-10-28:
> I would be glad to merge all the changes of my fork back to the original AsyncTCP repo

That would be great.

> if there's a plan to add collaborators to the project, CI, deployment and also time allocated for maintenance, review, pr merge, etc.

Plan is to add it directly to the core and be fully supported here. I have not decided on the WebServer yet, but the TCP should be here, just like UDP. I do have some fun plans for it's future also.

sivar2311, 2024-10-28:
Now I'm getting off-topic, but I can't resist:

>  I have not decided on the WebServer yet, but the TCP should be here, just like UDP. I do have some fun plans for it's future also.

Please also consider AsyncWebServer for this! 
It would be great to have an updated and maintained version of this masterpiece!

mathieucarbou, 2024-10-28:
> That would be great.
>
>
> Plan is to add it directly to the core and be fully supported here. I have not decided on the WebServer yet, but the TCP should be here, just like UDP. I do have some fun [...]

Ok! I will put that on my list and will do it then. Fist need to finish other things but I will for sure have time to do it before the end of the year.

closed 2024-10-29 by me-no-dev

microfoundry, 2024-11-12:
Going to kick this one again with:

```
assert failed: tcp_close_ext /IDF/components/lwip/lwip/src/core/tcp.c:492 (Required to lock TCPIP core functionality!)

0x4037f0d9: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c line 92
0x40385ccd: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 80
0x42065aa6: tcp_close_ext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 492
0x42065ad5: tcp_close at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 486
0x42033b40: AsyncServer::end() at c:\Users\TerryPhillips\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp line 1535
0x4203b4c1: AsyncWebServer::end() at c:\Users\TerryPhillips\Documents\Arduino\libraries\ESP_Async_WebServer\src\WebServer.cpp line 124
```

mathieucarbou, 2024-11-12:
> Going to kick this one again with:
>
> ```
> assert failed: tcp_close_ext /IDF/components/lwip/lwip/src/core/tcp.c:492 (Required to lock TCPIP core functionality!)
>
> 0x4037f0d9: esp_system_abort at [...]

I've mistakenly copy pasted and set the unlock at the wrong place 😓 Fixing...

mathieucarbou, 2024-11-25:
> Sorry for reviving this, but @mathieucarbou can you release a new version of ESPAsyncWebServer which uses v3.2.14 of AsyncTCP? Now we need to manually add the reference.

done.

usernametaken, 2024-12-26:
I am not sure it this is related a related issue.
I get the following error when I try to enable NAPT:

**ip_napt_enable(WiFi.softAPIP(), 1);**

assert failed: sys_timeout /IDF/components/lwip/lwip/src/core/timeouts.c:308 (Required to lock TCPIP core functionality!)

mathieucarbou, 2024-12-26:
> ip_napt_enable(WiFi.softAPIP(), 1);

you can surround this line with the code from thsi comment above: https://github.com/espressif/arduino-esp32/issues/10526#issuecomment-2439483380

This is the same kind of error, but not related to the same root cause and stack trace.

garykuipers, 2025-01-13:
assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)

Is an error I get with this code (painlessmesh.h). 

Strange thing was working yesterday, today all my working code has this error. I think stuff got updated on my Arduino IDE 2.3.4 and this may have ingested some bad code?

Anyway, here is my program:
```
#include "painlessMesh.h"
#include <WiFi.h>

#define MESH_PREFIX "yourMeshName"
#define MESH_PASSWORD "yourMeshPassword"
#define MESH_PORT 5555

#define ROOT_NODE_ID 2748336925  // The known ID of the root node

Scheduler userScheduler;
painlessMesh mesh;

void sendMessage(); // Prototype for sendMessage function

// Use the correct constructor for Task
Task taskSendMessage(TASK_SECOND * 1, TASK_FOREVER, &sendMessage);

void sendMessage() {
  String msg = "Hola from node ";
  msg += mesh.getNodeId();
  bool success = mesh.sendSingle(ROOT_NODE_ID, msg);
  Serial.printf("Message sent to root node %u: %s - %s\n", ROOT_NODE_ID, msg.c_str(), success ? "Success" : "Failed");

  taskSendMessage.setInterval(random(TASK_SECOND * 1, TASK_SECOND * 5));
}

void receivedCallback(uint32_t from, String &msg) {
  Serial.printf("Received from %u: %s\n", from, msg.c_str());
}

void setup() {
  Serial.begin(115200);

  mesh.setDebugMsgTypes(ERROR | STARTUP | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE);
  mesh.init(MESH_PREFIX, MESH_PASSWORD, &userScheduler, MESH_PORT);
  mesh.onReceive(&receivedCallback);

  userScheduler.addTask(taskSendMessage);
  taskSendMessage.enable();
}

void loop() {
  mesh.update();
}
```

The error occurs in the 
mesh.init(MESH_PREFIX, MESH_PASSWORD, &userScheduler, MESH_PORT);
line

mathieucarbou, 2025-01-13:
@garykuipers : please post the decoded stack...

garykuipers, 2025-01-14:
PLease excuse my ignorance. UNder help from grock I spnt aboutan hour trying to figure out how to do this. All of my attempts failed. I am going to try to install the ESP-IDF and use that, but before that I am going to see if I can go back a version of two on the Arduino IDE 2.3.4 I am using. I can't be sure but if seemed like there was an undate and then things stopped working.

I will get nbback to this as soon as I can.

(.venv) gary@gary-17Z90R-H-ADC8U1:~/PycharmProjects/Shrimptek$ python /home/gary/esp/esp-idf/components/espcoredump/espcoredump.py info_corefile -t /home/gary/PycharmProjects/Shrimptek/shrimp_sensors/shrimp6b/build/esp32.esp32.esp32/shrimp6b.ino.elf --core-format b '0x40082471:0x3ffb1f90 0x4008c25d:0x3ffb1fb0 0x40092512:0x3ffb1fd0 0x400f6cab:0x3ffb2100 0x400f6e25:0x3ffb2120 0x400dd0dc:0x3ffb2140 0x400d86de:0x3ffb2190 0x400d8811:0x3ffb21f0 0x400e4143:0x3ffb2270 0x4008cfa6:0x3ffb2290'
usage: espcoredump.py info_corefile [-h] [--debug DEBUG] [--gdb GDB] [--extra-gdbinit-file EXTRA_GDBINIT_FILE] [--core CORE] [--core-format {auto,b64,elf,raw}] [--off OFF] [--parttable-off PARTTABLE_OFF] [--save-core SAVE_CORE]
                                    [--rom-elf ROM_ELF] [--print-mem]
                                    prog
espcoredump.py info_corefile: error: argument --core-format/-t: invalid choice: '/home/gary/PycharmProjects/Shrimptek/shrimp_sensors/shrimp6b/build/esp32.esp32.esp32/shrimp6b.ino.elf' (choose from 'auto', 'b64', 'elf', 'raw')
(.venv) gary@gary-17Z90R-H-ADC8U1:~/PycharmProjects/Shrimptek$ ^C
(.venv) gary@gary-17Z90R-H-ADC8U1:~/PycharmProjects/Shrimptek$ python /home/gary/esp/esp-idf/components/espcoredump/espcoredump.py info_corefile -t b /home/gary/PycharmProjects/Shrimptek/shrimp_sensors/shrimp6b/build/esp32.esp32.esp32/shrimp6b.ino.elf '0x40082471:0x3ffb1f90 0x4008c25d:0x3ffb1fb0 0x40092512:0x3ffb1fd0 0x400f6cab:0x3ffb2100 0x400f6e25:0x3ffb2120 0x400dd0dc:0x3ffb2140 0x400d86de:0x3ffb2190 0x400d8811:0x3ffb21f0 0x400e4143:0x3ffb2270 0x4008cfa6:0x3ffb2290'
usage: espcoredump.py info_corefile [-h] [--debug DEBUG] [--gdb GDB] [--extra-gdbinit-file EXTRA_GDBINIT_FILE] [--core CORE] [--core-format {auto,b64,elf,raw}] [--off OFF] [--parttable-off PARTTABLE_OFF] [--save-core SAVE_CORE]
                                    [--rom-elf ROM_ELF] [--print-mem]
                                    prog
espcoredump.py info_corefile: error: argument --core-format/-t: invalid choice: 'b' (choose from 'auto', 'b64', 'elf', 'raw')
(.venv) gary@gary-17Z90R-H-ADC8U1:~/PycharmProjects/Shrimptek$ python /home/gary/esp/esp-idf/components/espcoredump/espcoredump.py info_corefile -t b64 /home/gary/PycharmProjects/Shrimptek/shrimp_sensors/shrimp6b/build/esp32.esp32.esp32/shrimp6b.ino.elf '0x40082471:0x3ffb1f90 0x4008c25d:0x3ffb1fb0 0x40092512:0x3ffb1fd0 0x400f6cab:0x3ffb2100 0x400f6e25:0x3ffb2120 0x400dd0dc:0x3ffb2140 0x400d86de:0x3ffb2190 0x400d8811:0x3ffb21f0 0x400e4143:0x3ffb2270 0x4008cfa6:0x3ffb2290'
usage: espcoredump.py [-h] [--chip {auto,esp32,esp32s2,esp32s3,esp32c3,esp32c2,esp32c6,esp32h2,esp32p4,esp32c5,esp32c61}] [--chip-rev CHIP_REV] [--port PORT] [--baud BAUD] [--gdb-timeout-sec GDB_TIMEOUT_SEC] [--version]
                      {dbg_corefile,info_corefile} ...
espcoredump.py: error: unrecognized arguments: 0x40082471:0x3ffb1f90 0x4008c25d:0x3ffb1fb0 0x40092512:0x3ffb1fd0 0x400f6cab:0x3ffb2100 0x400f6e25:0x3ffb2120 0x400dd0dc:0x3ffb2140 0x400d86de:0x3ffb2190 0x400d8811:0x3ffb21f0 0x400e4143:0x3ffb2270 0x4008cfa6:0x3ffb2290
(.venv) gary@gary-17Z90R-H-ADC8U1:~/PycharmProjects/Shrimptek$ python /home/gary/esp/esp-idf/components/espcoredump/espcoredump.py info_corefile -t b64 /home/gary/PycharmProjects/Shrimptek/shrimp_sensors/shrimp6b/build/esp32.esp32.esp32/shrimp6b.ino.elf
WARNING: /home/gary/PycharmProjects/Shrimptek/shrimp_sensors/shrimp6b/build/esp32.esp32.esp32/project_description.json does not exist. Please build the app with "idf.py build"
Please set up ESP-IDF to complete the action. Read more: https://github.com/espressif/esp-coredump/blob/master/README.md#installation
(.venv) gary@gary-17Z90R-H-ADC8U1:~/PycharmProjects/Shrimptek$

garykuipers, 2025-01-14:
Appears to have been resolved by changing from Arduino IDE 2.3.4 to platformio. Programs now compile and run as they did before I ran into this problem:

Suggestions fo people using Arduino IDEs: It will waste less of your time if you just use platformio. Arduino IDEs look easier but in the long ruun they will eat up days of your time pursuing errors that are not yours.

Arduino IDE was born of Windows ... so it is inherently flawed at its core.

me-no-dev, 2025-01-14:
@garykuipers this has nothing to do with the IDE. It only has to work with the versions of the ESP32 core. If you stay on PlatformIO, you are stuck on old core version (2.0.17) and on unsupported build tool (PlatformIO).

me-no-dev, 2025-01-14:
@garykuipers also you mention the IDE version a few times and that has nothing to do wit the ESP32 core version, which in your case is unknown to us.

garykuipers, 2025-01-14:
Thanks for answering:

ALl I know is: switched to platformio and the problems disappeared.
The Arduino is junky, I am glad you have nothing to do with it.
As for the core, I suspect that the discussion here found its way into the
core? And got added to my environment by an update a couple of days ago.
HOpefully whoever needs to fix this can fix it. In the meanwhile, I will
stick with platformio and occasionally install and compile the arduino ide
to see if it works if that is what contains a newer core.

I appreciate your time

jnivard, 2025-01-31:
I develop an distributed teperature control system using painlessmesh. After an upgrade to the new arduino IDE version 2.3.4 the application is not working becouse it runs into truble starting painlessmesh. I recieve the assert failed : tcp_alloc /IDF etc message. I have upgraded all the dependensies but that is not helping. All the examples in the example library give the same problem. I did try to go back to a running version of arduino but that is giving now the same problem arduino 1.8.13. So I am stuck and can not go backward or forward

me-no-dev, 2025-01-31:
@jnivard you need to downgrade the ESP32 Arduino core version to 3.0.7, not the IDE itself

jnivard, 2025-02-01:
@me-no-dev Thanks I look into that

jnivard, 2025-02-01:
@me-no-dev I recieved just now an upgrade from painlessmesh and the problem is solved

commit 71396de8 (2024-12-13): fix(sntp): Lock / Unlock LWIP if CONFIG_LWIP_TCPIP_CORE_LOCKING is set
- Fixes: #10526
- Completes old PR #10529