espressif/arduino-esp32 #10527: GPIO6 set high inside Bootloader

microbotsio, 2024-10-26:
### Board

ESP32C3 Dev Board

### Device Description

Tested n the ESP32-C3-MINI-1-N4

### Hardware Configuration

GPIO6

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Gpio pin 6 is set high inside the bootloader - eg. when re-flashing the device or initially resting the device

Other gpio pins like 1,2,3,5,7 are all keep low during this state. Is there a way to keep gpio 6 set it low? 

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
Hardware issue
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-10-26:
GPIO 6 is a special GPIO and has restrictions/special behaviour. Short answer no.

microbotsio, 2024-10-30:
Is there any specifications about the behavior of this pin during boot mode?

Jason2866, 2024-10-30:
Yes, see [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-c6_datasheet_en.pdf)

VojtechBartoska, 2024-11-04:
Closing this as answered.

closed 2024-11-04 by VojtechBartoska

microbotsio, 2025-02-24:
> Solution for GPIO6 is to use stronger external pull down or change EFUSE
> "- If eFuse bit EFUSE_DIS_PAD_JTAG = 1, the pin MTCK is left floating
> after reset, i.e. IE = 1. If eFuse bit EFUSE_DIS_PAD_JTAG = 0, the pin MTCK
> is connected to internal pull-up resistor, i.e. IE = 1, WPU = 1."

Thanks for the information :)