espressif/arduino-esp32 #10534: Analogwrite to RGB LED on Nano ESP32 does not turn off completely

ednieuw, 2024-10-28:
### Board

Arduino Nano ESP32

### Device Description

Arduino Nano ESP32 

### Hardware Configuration

No

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE 1.0 and 2.0

### Operating System

Windows 11

### Flash frequency

NA

### PSRAM enabled

yes

### Upload speed

NB Arduino settings in board

### Description

The RGB LED keeps on faintly when it should be off when writing 512 to the analog port of the LED_RED, LED_GREEN or LED_BLUE

When writing 513 the LED colour turns off.
So there is an one bit discrepantie 

### Sketch

```cpp
int Toggle=0;
void setup() {}
void loop() 
{
 UpdateStatusLEDs(Toggle);
 Toggle = 1 - Toggle;
 delay (1000);
}

//--------------------------------------------                                                //
// COMMON Update routine for the status LEDs
//-------------------------------------------- 
void UpdateStatusLEDs(int Toggle)
{
 if(Toggle)   
   {
    SetStatusLED(Toggle * random(100), Toggle * random(100), Toggle * random(100)); 
    analogWrite(48,  Toggle * 50);     
   }
   else
   {
    SetStatusLED(0,0,0); 
    analogWrite(48, 0);                                                               // on Nano ESP32 Turn the LED off    
   }
}
//--------------------------------------------
// COMMON Control the RGB LED on the Nano ESP32
// Analog range 0 - 512. 0 is LED On max intensity
// 512 is LED off. Therefore the value is subtracted from 512 
//--------------------------------------------
void SetStatusLED(int Red, int Green, int Blue)
{
 analogWrite(LED_RED,   512 - Red);                                                            // !Red (not Red) because 1 or HIGH is LED off
 analogWrite(LED_GREEN, 513 - Green);
 analogWrite(LED_BLUE,  513 - Blue);
}
```

### Debug Message

```plain
none
```

### Other Steps to Reproduce

With the Arduino Nano ESP32 core 2.0.18 the RGB turns off completely.
When using core 3.0.5 up to the latest core 3.1.0 RC2 this occurs. 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ednieuw, 2024-10-28:
What also not works correctly is the pin numbering when using the RGB LED and using LED_RED,LED_GREEN,LED_BLUE
It should be set to By GPIO number (legacy) and not  By Arduino pin (default).

P-R-O-C-H-Y, 2024-10-29:
@pillo79