espressif/arduino-esp32 #10541: ESP-MQTT Intermittent Connection Failures

vicatcu, 2024-10-29:
### Board

ESP32 Dev Module

### Device Description

Custom PCBA

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Ubuntu 22.04

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

When connecting tot my MQTT Server (self-hosted, Mosquitto), I intermittently experience the following outcome in console output.

E (15049) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read(): EOF
E (15050) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read() error: errno=119
[MQTT EVENT] reported from esp-tls (0x8008)
[MQTT EVENT] Last errno string (Success)
E (15065) mqtt_client: esp_mqtt_connect: mqtt_message_receive() returned -2
E (15068) mqtt_client: MQTT connect failed

### Sketch

```cpp
My code effectively looks like this:

  static esp_mqtt_client_config_t mqtt_cfg;
  mqtt_cfg.broker.address.hostname = settings->server;
  mqtt_cfg.broker.address.transport = MQTT_TRANSPORT_OVER_SSL;
  mqtt_cfg.broker.address.port = 8883;
  mqtt_cfg.credentials.username = settings->username;
  mqtt_cfg.credentials.client_id = settings->identity;
  mqtt_cfg.credentials.authentication.password = settings->password;
  mqtt_cfg.broker.verification.certificate = root_ca;
  esp_mqtt_client_handle_t _client = esp_mqtt_client_init(&mqtt_cfg);
  esp_mqtt_client_register_event(_client, (esp_mqtt_event_id_t) ESP_EVENT_ANY_ID, mqtt_event_handler, NULL);
  const esp_err_t ret = esp_mqtt_client_start(_client);
  boolean connected = false;
  if (ESP_OK == ret) {
      connected = waitForMqttConnected();
      // polls for global _mqttIsConnected global variable set in mqtt event handler to become true
      // or times out after 15 seconds
  }

```
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Try running that code and see what happens over several iterations of it

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-10-29:
I suggest you ask [here](https://github.com/espressif/esp-mqtt) instead :)

VojtechBartoska, 2024-11-04:
Hello, closing this as it is tracked in MQTT repository.

Thanks for creating a ticket there @vicatcu.

closed 2024-11-04 by VojtechBartoska