espressif/arduino-esp32 #10547: SPIFFS Mounting Causing Error

alogusAdam, 2024-10-31:
### Board

ESP32 Dev Module

### Device Description

DevKitC (4MB)

### Hardware Configuration

No Connections

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Copied from my old issue #10499 that I marked as closed: Whenever I attempt to mount SPIFFS with `SPIFFS.begin(true);`, the following message prints over and over again in the serial monitor:

`E (1030) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1030) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1062) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1063) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1064) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1066) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1102) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1103) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1103) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1134) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1135) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1136) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1138) task_wdt: esp_task_wdt_reset(763): task not found`
`E (1175) task_wdt: esp_task_wdt_reset(763): task not found`

and the SPIFFS mount fails. I am running into this issue with all sorts of firmware that uses SPIFFS and used to work without any problems. To simplify things, I am running the SPIFFS_Test.ino example and running into the same issue. I have tried this on multiple boards, updated Arduino IDE to the latest version, and updated the ESP32 board package to the latest version, none of which resolve the issue. When `SPIFFS.begin(false);` is run, the error does not print, but SPIFFS mount fails.

Why I'm reopening this issue:
I closed the last issue with "Not exactly sure what the issue was here, but it ran fine when compiled and uploaded from another computer. I deleted the folder at "C:/Users//AppData/Roaming/arduino-ide", and this seems to have resolved the issue."

Since then, I have run into the issue again--seemingly out of nowhere. I tried uploading from another computer again--no issue. So, I thought something was wrong with my Arduino IDE. I uninstalled, deleted the ".arduinoIDE" folder under C:/Users/Owner along with local and roaming Arduino IDE data in C:/Users/Owner/AppData, restarted my computer, and reinstalled a fresh copy of Arduino IDE. Uploading to the same board that was previously experiencing issues (but most recently uploaded from another computer where it ran fine), I encountered no issue. I now am realizing this is likely because SPIFFS was already mounted from the code uploaded from another computer. After erasing flash and reuploading, I ran into the same issue.

So, I tried uninstalling Arduino IDE, reinstalling, etc., and nothing worked. I went to the computer that was able to upload successfully, and after uploading from that computer, I am still getting this task_wdt error! Nothing changed. Was this board "infected" somehow? I tried with a board that had never been used before--same issue. I tried an ESP32-S3--same issue.

I downgraded to board package release 2.0.17, and SPIFFS seems to mount no problem. I've looked through the migration guide and can't find anything that I should have changed. And I'd at least expect the example code to work fine here. Happy to include any other information I can.

### Sketch

```cpp
#include "FS.h"
[... 196 lines cut ...]
E (1297) task_wdt: esp_task_wdt_reset(763): task not found
E (1329) task_wdt: esp_task_wdt_reset(763): task not found
```

### Other Steps to Reproduce

The following settings were used in Arduino IDE:
Board: "ESP32 Dev Module"
CPU Frequency: "240MHz (WiFi/BT)"
Core Debug Level: "None" Tried setting this to verbose with no additional info gained
Erase All Flash Before Sketch Upload: "Disabled" Tried setting to true and got the same results
Events Run On: "Core 1"
Flash Frequency: "80MHz"
Flash Mode: "QIO"
Flash Size: "4MB (32Mb)"
JTAG Adapter: "Disabled"
Arduino Runs On: "Core 1"
Partition Scheme: "Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS)"
PSRAM: "Disabled"
Upload Speed: "921600"
Zigbee Mode: "Disabled"

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

alogusAdam, 2024-11-05:
Not the most elegant solution, but I copied and pasted the SPIFFS folder from 2.0.17 into C:\Users\<OWNER>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\libraries, and this seems to have resolved the issue. Hopefully there will be a fix for this, but it seems something from the upgrade to 2.0.17 to 3.0.0 is causing this issue--at least it is contained within the SPIFFS src code.

alogusAdam, 2024-11-05:
The issue returned and the temporary solution from my most recent comment did not resolve the issue this time.

me-no-dev, 2024-11-07:
it's just trying to feed the idle task WDT while formatting. But because we disable it on the second core, it prints error. We will look into suppressing it

dquadros, 2024-11-07:
> it's just trying to feed the idle task WDT while formatting. But because we disable it on the second core, it prints error. We will look into suppressing it

Not directly related to this issue, but when (which version) was the idle task WDT disabled on the second core? 

I've seem this "task not found" messages with another library. What should we do with code that is supposed to work with arduino-esp32 version 3 and previous versions? Is there a way to check if the idle task WDT is enabled at the current core?

me-no-dev, 2024-11-07:
All dual-core chips have it disabled for the second core. Has been always like that, but the logs were not printed before. Maybe they were added in a more recent IDF versions

admire123hk, 2025-01-31:
i experienced the same problem, with my ESP32-S3 N16R8 , but after waited 3-4 minutes formatting the 
SPIFFS partition , it becomes working normal again.

KorochanOne, 2025-02-04:
I got the same problem with Serial1 master.
using Serial1 with core0, the same flud of error happend.

  xTaskCreatePinnedToCore(SubTask, "SubTask", 24576, NULL, 10, &gTaskHandle, 0);
in SubTask, if Serial communication takes a longer than first error message come. 
xTaskCreatePinnedToCore() could not exit.

change to 3.1.1 to 2.0.17, it works without no message.
I hope this info helps us.
please fix this known problem, otherwise I can't move to 3.11.

Killbyfire, 2025-02-22:
@admire123hk That actually solved it, spent a whole day scratching my head wondering how i could fix it. From my whole heart, thank you so much. <3

Killbyfire, 2025-02-22:
Did some more testing turns out i had erase flash before uploading new code, @admire123hk your solution helped me, thank you so much <3 I just waited 4 minutes and it works everytime now ty

Jason2866, 2025-03-12:
Please test with Arduino Core 3.2.0-rc2  A fix has been applied

closed 2025-03-27 by Parsaabasi