espressif/arduino-esp32 #10550: HttpClient errorToSTring does not return values for 401 / HTTP_CODE_UNAUTHORIZED

CybDis, 2024-11-01:
### Board

Arduino-esp32

### Device Description

x

### Hardware Configuration

x

### Version

v3.0.6

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

x

### PSRAM enabled

yes

### Upload speed

x

### Description

The HttpClient class does not return string for HTTP_CODE_UNAUTHORIZED - 401

### Sketch

```cpp
Serial.println(http.errorToString(httpCode)); // httpCode = 401
```

### Debug Message

```plain
x
```

### Other Steps to Reproduce

x

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-11-01:
It is meant for internal errors, not for HTTP ones. Like why connection to the server failed

Parsaabasi, 2025-03-12:
Hi @CybDis 

Do you still need assistance with this?

Jason2866, 2025-03-12:
Closing since outdated and no reply. Feel free to ask to reopen if the issue still exists with actual Arduino core

closed 2025-03-12 by Jason2866