espressif/arduino-esp32 #10557: always crashes (interrupt watchdog timeout) in a spinlock when doing "heavy" Ethernet traffic on a wt32-eth01

folkertvanheusden, 2024-11-02:
### Board

WT-ETH01

### Device Description

This is a regular esp32 with an ethernet-port bolted on. 

### Hardware Configuration

On pin 12 - 15 (miso, mosi, sclk and cs) there's an SD-Card reader.

### Version

v3.0.6

### IDE Name

PlatformIO

### Operating System

Ubuntu Linux 24.10

### Flash frequency

default

### PSRAM enabled

no

### Upload speed

115200

### Description

The problem occurs with an iSCSI target running on the ESP32. During setup (DHCP, iSCSI logon) all is good.
Yet when regular i/o occurs (e.g. reads and writes to the device), an interrupt watchdog happens.

### Sketch

```cpp
Sofar I haven't been able to make a smaller sketch showing the behaviour.
The same 'sketch' runs fine on teensy and on linux (yes).
```

### Debug Message

```plain
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4008c651  PS      : 0x00060535  A0      : 0x80168ce2  A1      : 0x3ffb2aa0  
A2      : 0x3ffbddec  A3      : 0xffffffff  A4      : 0x0000cdcd  A5      : 0xb33fffff  
A6      : 0xb48bfd02  A7      : 0x00060523  A8      : 0x8008c651  A9      : 0x3ffb2a80  
A10     : 0x00000000  A11     : 0xb33f5454  A12     : 0x0000abab  A13     : 0x00003126  
A14     : 0x3ffdac88  A15     : 0x00000004  SAR     : 0x00000015  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40088f34  LEND    : 0x40088f4a  LCOUNT  : 0xffffffff  

Backtrace: 0x4008c64e:0x3ffb2aa0 0x40168cdf:0x3ffb2ad0 0x4008c771:0x3ffb2af0 0x401679c2:0x3ffb2b10 0x40109396:0x3ffb2b50 0x40109596:0x3ffb2b70 0x40109af8
:0x3ffb2b90 0x4010af42:0x3ffb2bb0 0x400fab05:0x3ffb2bd0
```

That backtrace resolves to:

```
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/spinlock.h:112
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:575
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2414
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:906
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/freertos/sys_arch.c:77
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/freertos/sys_arch.c:483
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_lib.c:1149
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/api_msg.c:1899
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c:202
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

folkertvanheusden, 2024-11-02:
Another backtrace gave:

```
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/cpu.c:423
/Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/spinlock.h:112
...
```

Jason2866, 2025-03-12:
Does the issue still exists with actual Arduino core? If yes please provide a minimal example sketch to reproduce.

Parsaabasi, 2025-03-26:
Hello, 

Since there were no replied for the question above, I close this issue for now. In case it persists, please reopen or create another issue. 

Thanks.

closed 2025-03-26 by Parsaabasi