espressif/arduino-esp32 #10568: 3.1.0-RC2 and ACAN2517 library

EarlVadim, 2024-11-05:
### Board

ESP32-S3

### Device Description

plain module

### Hardware Configuration

#define MCP2517_CS   10 
#define MCP2517_MOSI 11
#define MCP2517_SCK  12
#define MCP2517_MISO 13

### Version

latest development Release Candidate (RC-X)

### IDE Name

A-IDE 2.3.3

### Operating System

Win11

### Flash frequency

QIO 80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I have a project using an external CAN controller MCP2518. 
On the current version 3.0.7 everything compiles and works, but on the RC2 version the same project compiles, but when trying to initialize the CAN, the controller reboots abnormally.

### Sketch

```cpp
void CAN1_INIT() {  
   SPI.begin (MCP2517_SCK, MCP2517_MISO, MCP2517_MOSI) ;
   Serial.println("SPI started");
   ACAN2517Settings settings2517 (ACAN2517Settings::OSC_40MHz, 500 * 1000) ; 
   settings2517.mRequestedMode = ACAN2517Settings::ListenOnly ; 
   const uint32_t errorCode2517 = can.begin (settings2517, NULL) ;

Here ERROR without any messages and reboot
```

### Debug Message

```plain
I don't have any logs. I deleted RC2 and reinstalling is quite complicated. I just want to let you know that you have seriously messed something up in the new version.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-11-05:
> I just want to let you know that you have seriously messed something up in the new version.

We do not provide Arduino abstraction for TWAI, so the issue must be in ESP-IDF v5.3

EarlVadim, 2024-11-05:
> > I just want to let you know that you have seriously messed something up in the new version.
> 
> We do not provide Arduino abstraction for TWAI, so the issue must be in ESP-IDF v5.3

But, this NOT TWAI... It external CAN-controller via SPI.

me-no-dev, 2024-11-05:
Sure. My bad. We still need some logs. SPI is so far working OK on our end

Jason2866, 2025-03-12:
Closing, since no feedback. Feel free to ask to reopen when the needed logs are provided

closed 2025-03-12 by Jason2866