espressif/arduino-esp32 #10569: cores/esp32/esp32-hal-uart.c error:  'uart_config_t' has no member named 'flags'

TroyMitchell911, 2024-11-05:
### Board

self board based on ESP32C3

### Device Description

myself board

### Hardware Configuration

Nope

### Version

latest master (checkout manually)

### IDE Name

Clion

### Operating System

Ubuntu22.04

### Flash frequency

24M

### PSRAM enabled

no

### Upload speed

115200

### Description

I run the following command to add the component:
```bash
❯ idf.py add-dependency "espressif/arduino-esp32^3.0.7"
Executing action: add-dependencyif/arduino-esp32^3.0.7"
Created "/home/troy/mcu/esp/arduino-prj/sshd/main/idf_component.yml"
NOTICE: Skipping optional dependency: espressif/rmaker_common
NOTICE: Skipping optional dependency: espressif/qrcode
NOTICE: Skipping optional dependency: espressif/esp_rainmaker
NOTICE: Skipping optional dependency: espressif/esp_insights
NOTICE: Skipping optional dependency: espressif/esp-zigbee-lib
NOTICE: Skipping optional dependency: espressif/esp-zboss-lib
NOTICE: Skipping optional dependency: espressif/esp-sr
NOTICE: Skipping optional dependency: espressif/esp-dsp
Successfully added dependency "espressif/arduino-esp32^3.0.7" to component "main"
Done
```
and when I try to build it, I get error:
```bash
/home/troy/mcu/esp/arduino-prj/sshd/managed_components/espressif__arduino-esp32/cores/esp32/esp32-hal-uart.c: In function 'uartBegin':
/home/troy/mcu/esp/arduino-prj/sshd/managed_components/espressif__arduino-esp32/cores/esp32/esp32-hal-uart.c:501:14: error: 'uart_config_t' has no member named 'flags'
  501 |   uart_config.flags.backup_before_sleep = false;  // new flag from IDF v5.3
```
I know deleting this line can fix the error, but isn't the one I added the latest version? I checked this version I didn't find this line of code in the source code.
Can anyone provide some clues? I'd appreciate it.

btw, I try to create a project from arduino example, but same:
```bash
idf.py create-project-from-example "espressif/arduino-esp32^3.0.7:hello_world"
```

### Sketch

```cpp
#include "Arduino.h"

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("Hello world!");
  delay(1000);
}
```

### Debug Message

```plain
Nope
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TroyMitchell911, 2024-11-05:
esp idf v5.3

SuGlider, 2024-11-05:
Please use Arduino Core 3.1.0 in order to use IDF 5.3. It shall be cloned from `release/v3.1.x` branch.
https://github.com/espressif/arduino-esp32/tree/release/v3.1.x

The Master branch id Arduino Core 3.0.x based on IDF 5.1.

TroyMitchell911, 2024-11-05:
> Please use Arduino Core 3.1.0 in order to use IDF 5.3. It shall be cloned from `release/v3.1.x` branch. https://github.com/espressif/arduino-esp32/tree/release/v3.1.x
> 
> The Master branch is the Arduino Core 3.0.x based on IDF 5.1.
> 
> This information about which Arduino Core version uses which IDF version can be viewed in the [release notes](https://github.com/espressif/arduino-esp32/releases)

are u sure? 
I download [this](https://github.com/espressif/arduino-esp32/releases/tag/3.1.0-RC2) and I find same code:
```c
uart_config.flags.backup_before_sleep = false;  // new flag from IDF v5.3
```
and `release/v3.1.x` too!

SuGlider, 2024-11-05:
> how to make sure idf version?

`idf.py --version` will tell the exact IDF version that is being used.
In https://github.com/espressif/esp-idf/releases/tag/v5.3 there is some instructions about how to clone IDF 5.3, if necessary.

TroyMitchell911, 2024-11-06:
> > how to make sure idf version?
> 
> `idf.py --version` will tell the exact IDF version that is being used. In https://github.com/espressif/esp-idf/releases/tag/v5.3 there is some instructions about how to clone IDF 5.3, if necessary.

I run this command and find Im in dirty branch.
so I download this release branch and check `/components/esp_driver_uart/include/driver/uart.h`
also, don't has `flags` member.
https://github.com/espressif/esp-idf/blob/e0991facf5ecb362af6aac1fae972139eb38d2e4/components/esp_driver_uart/include/driver/uart.h#L36

TroyMitchell911, 2024-11-06:
release/v5.4 exists...

https://github.com/espressif/esp-idf/blob/906f36bdb9bcc572fff5ac26a305e2d166895f7c/components/esp_driver_uart/include/driver/uart.h#L36

SuGlider, 2024-11-06:
Arduino Core was built using such flag. The Arduino Core is correct, for sure.

SuGlider, 2024-11-06:
>I think this comment is wrong..It's from v5.3.1 instead of v5.3

IDF 5.3 includes all minor versions. Anyway this comment is for internal use.

TroyMitchell911, 2024-11-06:
> > I think this comment is wrong..It's from v5.3.1 instead of v5.3
> 
> IDF 5.3 includes all minor versions. Anyway this comment is for internal use.

get it. tnx for ur response.

TroyMitchell911, 2024-11-06:
Hi, SuGlider.
I went a trouble from hello_world example.
code here: 
```c
//file: main.cpp
#include "Arduino.h"

extern "C" void app_main()
{
    initArduino();

    // Arduino-like setup()
    Serial.begin(115200);
    while(!Serial){
        ; // wait for serial port to connect
    }
    // Arduino-like loop()
    while(true){
        Serial.println("loop");
        printf("loop from print\n");
        delay(1000);
    }

    // WARNING: if program reaches end of function app_main() the MCU will restart.
}
```

and I can't get any output by `Serial.println`, but `printf` is ok.

Arduino core: 3.0.10-rc2
esp-idf: v5.3.1
chip: esp32c3
uart: cdc

do u have any clue?

me-no-dev, 2024-11-06:
you are calling CPP object from C function
```cpp
//file: main.cpp
#include "Arduino.h"

void setup(){
    Serial.begin(115200);
    while(!Serial){
        ; // wait for serial port to connect
    }
}

void loop(){
    Serial.println("loop");
    printf("loop from print\n");
    delay(1000);
}

extern "C" void app_main()
{
    initArduino();

    // Arduino-like setup()
    setup();
    // Arduino-like loop()
    while(true){
        loop();
    }

    // WARNING: if program reaches end of function app_main() the MCU will restart.
}
```

SuGlider, 2024-11-06:
> Hi, SuGlider. I went a trouble from hello_world example. code here:
>
> ```c
> //file: main.cpp
> #include "Arduino.h"
>
> extern "C" void app_main()
> {
>     initArduino();
>
>     // Arduino-like setup() [...]

Besides @me-no-dev suggestion, I would also suggest to use a CPP file with a regular sketch.
In that case, please execute `idf.py menuconfig` and change the Arduino setting to use `CONFIG_AUTOSTART_ARDUINO=y` to allow the Arduino component to call `setup()` and `loop()` as a regular sketch.
You can also start using this project: https://github.com/espressif/arduino-esp32/tree/master/idf_component_examples/hello_world

SuGlider, 2024-11-06:
@TroyMitchell911 - Espressif created a technical and community event called Espressif DevCon.
There many videos in the Espressif Youtube Channel at https://www.youtube.com/@espressifsystems

There is a video about Arduino as IDF Component.
If you consider it a valid and a viable resource, the video is at  https://www.youtube.com/watch?v=8w3MmsZBa9o

SuGlider, 2024-11-06:
More Arduino as IDF Component examples can be found in https://github.com/SuGlider/devcon2024

TroyMitchell911, 2024-11-07:
I change my env:
idf: v5.1.4
arduino core: 3.0.7

and I create a example from `Arduino core`:
```bash
idf.py create-project-from-example "espressif/arduino-esp32^3.0.7:hello_world"
```
then build and flash, I can't get any `hello world`. It returns.
below is log:
```
❯ idf.py monitor -p /dev/ttyACM0
Executing action: monitor
Running idf_monitor in directory /home/troy/mcu/esp/arduino-prj/hello_world
Executing "/home/troy/.espressif/python_env/idf5.1_py3.10_env/bin/python /home/troy/repo/esp-idf-v5.1.4/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --revision 3 --decode-panic backtrace /home/troy/mcu/esp/arduino-prj/hello_world/build/main.elf -m '/home/troy/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/home/troy/repo/esp-idf-v5.1.4/tools/idf.py'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyACM0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038300c
--- 0x4038300c: rv_utils_wait_for_intr at /home/troy/repo/esp-idf-v5.1.4/components/riscv/include/riscv/rv_utils.h:33
 (inlined by) esp_cpu_wait_for_intr at /home/troy/repo/esp-idf-v5.1.4/components/esp_hw_support/cpu.c:119

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1704
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710
I (24) boot: ESP-IDF v5.1.4 2nd stage bootloader
I (24) boot: compile time Nov  7 2024 10:17:19
I (25) boot: chip revision: v0.4
I (27) boot.esp32c3: SPI Speed      : 80MHz
I (32) boot.esp32c3: SPI Mode       : DIO
I (37) boot.esp32c3: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0c348h ( 49992) map
I (101) esp_image: segment 1: paddr=0001c370 vaddr=3fc8c400 size=01288h (  4744) load
I (102) esp_image: segment 2: paddr=0001d600 vaddr=40380000 size=02a18h ( 10776) load
I (112) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=23070h (143472) map
I (141) esp_image: segment 4: paddr=00043098 vaddr=40382a18 size=098c4h ( 39108) load
I (153) boot: Loaded app from partition at offset 0x10000
I (153) boot: Disabling RNG early entropy source...
I (164) cpu_start: Unicore app
I (165) cpu_start: Pro cpu up.
I (173) cpu_start: Pro cpu start user code
I (173) cpu_start: cpu freq: 160000000 Hz
I (173) cpu_start: Application information:
I (176) cpu_start: Project name:     main
I (181) cpu_start: App version:      1
I (185) cpu_start: Compile time:     Nov  7 2024 10:17:12
I (192) cpu_start: ELF file SHA256:  d4b10e23616fd420...
I (198) cpu_start: ESP-IDF:          v5.1.4
I (202) cpu_start: Min chip rev:     v0.3
I (207) cpu_start: Max chip rev:     v1.99
I (212) cpu_start: Chip rev:         v0.4
I (217) heap_init: Initializing. RAM available for dynamic allocation:
I (224) heap_init: At 3FC8E860 len 000317A0 (197 KiB): DRAM
I (230) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (237) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (245) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (252) spi_flash: detected chip: generic
I (256) spi_flash: flash io: dio
W (260) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (273) sleep: Configure to isolate all GPIO pins in sleep state
I (279) sleep: Enable automatic switching of GPIO sleep configuration
I (287) app_start: Starting scheduler on CPU0
I (292) main_task: Started on CPU0
I (296) main_task: Calling app_main()
I (310) main_task: Returned from app_main()
I (310) uart: queue free spaces: 20

```

TroyMitchell911, 2024-11-07:
> @TroyMitchell911 - Espressif created a technical and community event called Espressif DevCon. There are many videos in the Espressif Youtube Channel https://www.youtube.com/@espressifsystems
> 
> There is a video about Arduino as IDF Component. If you consider it a valid and a viable resource, the video is on https://www.youtube.com/watch?v=8w3MmsZBa9o

I add these to CMakeLists.txt and CDC-UART runs! thanks for ur link.
```
list(APPEND compile_definitions "ARDUINO_USB_CDC_ON_BOOT=1")
list(APPEND compile_definitions "ARDUINO_USB_MODE=1")
```

closed 2024-11-07 by TroyMitchell911