espressif/arduino-esp32 #10573: Dual-mode BLEServer misidentifies client connection as server connection, performs unnecessary transmission and warning

ArdenKolodner, 2024-11-05:
### Board

ESP32C3

### Device Description

Plain module attached by USB-C

### Hardware Configuration

The issue occurs when no other hardware is present.

### Version

latest master (checkout manually)

### IDE Name

VSCode

### Operating System

macOS 14.6.1

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I have an ESP32-C3 in dual-mode BLE. I'll call it the central device. It has a client connection to another device (also an ESP32-C3), which I'll call the peripheral device. It is also a server (`BLEServer` via `BLEDevice::createServer()`) which has a client connected to it. That client is my laptop.

When sending a BLE notification with a characteristic value longer than 20 bytes, I got the following warning:
`[W][BLECharacteristic.cpp:527] notify(): - Truncating to 20 bytes (maximum notify size)`
However, the server had set its MTU to 500 via `BLEDevice::setMTU(500)`. I tried using multiple BLE client applications (LightBlue, Python via Bleak, and react-native-ble-plx), confirming that the warning occurs with all of them. Additionally, all can support an MTU of 500, so there shouldn't be any truncation. Indeed, all of the data was being transmitted successfully, with no truncation, despite the warning.

During debugging, I found that the warning is not being generated when the server sends the notification to my laptop. It's actually trying to send 2 notifications: one to the peripheral device, and one to my laptop. The laptop one works fine, but the peripheral device's negotiated MTU is 23, which is too short. I didn't bother to have the peripheral select an MTU because it's not supposed to be receiving notifications from the server – it's a server as well, not a client! The server should not be sending notifications to it in the first place.

I think the problem is that the connection to the peripheral device, which is a client connection, is getting misidentified as a server connection. When the central device connects to the peripheral, `BLEServer::addPeerDevice` gets called with `_client` set to false, so the central device mistakenly thinks this connection is from an incoming client, instead of what's really going on, which is that the central device is the client. Then the connection gets added to `m_connectedServersMap `. However, this would happen even if `_client` were true – the argument is totally ignored. But then later, when `BLECharacteristic::notify()` calls `getService()->getServer()->getPeerDevices(false)`, the `getPeerDevices()` method ignores the `_client` argument again and returns the whole list of peer devices (`m_connectedServersMap`) which now includes the client connection.

Thus `notify()` thinks there are two clients to notify instead of just one, and it tries to notify both. The notification to the peripheral device does not throw an error, for some reason - `esp_ble_gatts_send_indicate` returns ESP_OK for both notifications. But the peripheral device shouldn't be getting it anyway.

The root of the problem seems to be that `BLEServer::handleGATTServerEvent()` is getting called on an `ESP_GATTS_CONNECT_EVT` event, when the peripheral device connects. This prompts an `addPeerDevice(..., false)` call that should not occur. As I don't have a ton of experience with BLE outside of using this library, I don't know why this is happening.

### Sketch

```cpp
Start a BLE server with minimal MTU.

Start a dual-mode BLE device. Have it connect to the server, then start advertising itself. It needs a characteristic that's readable and notifiable.

Connect a BLE client (I recommend LightBlue) to the dual-mode device. Have it subscribe to the characteristic.

Have the server set the characteristic value to something several hundred bytes long, and send a notification.
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
[... 215 lines cut ...]
[ 50870][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[ 50876][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
W (53094) BT_HCI: hcif disc complete: hdl 0x2, rsn 0x13
W (53095) BT_APPL: gattc_conn_cb: if=3 st=0 id=259 rsn=0x13
[ 52237][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_DISCONNECT_EVT
Central is disconnected from laptop. Note: the peripheral is still connected at this point.
[ 52251][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[ 52256][D][BLEAdvertising.cpp:199] start(): - advertising service: 94289ed5-65a5-463c-9730-71e4513964ae
[ 52266][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_DISCONNECT_EVT
[ 52277][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_DISCONNECT_EVT
W (53148) BT_BTM: BTM_BleWriteScanRsp, Partial data write into ADV
[ 52297][D][BLEClient.cpp:493] handleGAPEvent(): BLEClient ... handling GAP event!
[ 52302][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[ 52307][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 0]
[ 52315][D][BLEClient.cpp:493] handleGAPEvent(): BLEClient ... handling GAP event!
[ 52322][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[ 52328][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 1]
[ 52335][D][BLEClient.cpp:493] handleGAPEvent(): BLEClient ... handling GAP event!
[ 52342][D][BLEDevice.cpp:579] getAdvertising(): get advertising
[ 52348][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 6]
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ArdenKolodner, 2024-11-06:
Another note on this that I just discovered: the `log_w("- Truncating to %d bytes (maximum notify size)", _mtu - 3);` line turns out to take a total of 9ms to run(!!), which was single-handedly throttling my device's BLE throughput rates – our goal is 10ms per packet, but we have our own processing to do between packets, of course. This could be solved by having the peripheral device increase its MTU, but then there's more unnecessary data being transmitted, also throttling throughput. So a benefit of fixing this issue will be an immediate and automatic increase in BLE transmission times for packets above 23 bytes!

SuGlider, 2024-11-09:
@ArdenKolodner -
From the log, it sounds like it is an Arduino as Component project.

In order to verify the issue, we need a minimum Arduino Sketch that can be used to reproduce the issue and latter to verify a potential fix.
Please provide such sketch, but have in mind that  we don't fix IDF Code, only the ESP32 Arduino Core or its official libraries.

ArdenKolodner, 2024-11-10:
@SuGlider Here is a sketch. I cobbled it together from the Arduino example sketches for BLE client and server, so it should be as close to a textbook example as possible. The steps to use it: set up a BLE server that has a service and characteristic given by clientServiceUUID and clientCharUUID (or change their values), then on another device (I used an ESP32S3 dev board), turn the debug log level to Warning or above and upload this sketch. You should see that the client connection triggers the server connection callback, and when notify() is called, it will cause the truncation warning to occur because the server is trying to incorrectly send notification data to the client. You can also connect a client with high MTU, such as a laptop to see that that will receive the data correctly, without causing another warning. Please let me know if anything is unclear.

[dual_mode_double_notify_bug.ino.zip](https://github.com/user-attachments/files/17689880/dual_mode_double_notify_bug.ino.zip)

SamObde, 2025-04-04:
Im having this same issue and would love to help fix it... or test : )

SamObde, 2025-04-04:
Did you ever figure out how to solve the server disconnects when a client disconnects?

ArdenKolodner, 2025-04-07:
Unfortunately, no. We've just been trying to make the peripherals more reliable, and living with this issue. Would love an update or confirmation that my sketch reproduces the issue for you @SuGlider

SamObde, 2025-04-07:
I just switched my entire project over to NimBLE and am no longer having the problem.  I also gained a lot of memory by doing so.  I did have to ditch my classic bt scanning on this project but I can work around that for now.

ArdenKolodner, 2025-04-07:
Thank you! Unfortunately NimBLE had [this issue](https://www.esp32.com/viewtopic.php?t=31715) which I could not solve and which made it unusable for my use case, where I need several peripherals connected at once streaming data quickly. But it's been a while since I tested, maybe they've fixed it. And either way, thank you for letting me know about the roadmap, which I was unaware of! Glad you managed to find a workaround with it :)

SuGlider, 2025-04-07:
@ArdenKolodner @SamObde 
Have you guys tried this NimBLE Arduino Library: https://github.com/h2zero/NimBLE-Arduino
We are currently refactoring the Bluedroid BLE Arduino Library to use NimBLE instead.
This new library will take some time to be fully ready (maybe by end of june).

In the meanwhile, I'd suggest trying the h2zero NimBLE-Arduino library. 
I hope it will solve the issue.

We won't work/fi the current Arduino BLE Library (based on Bluedroid) in favor of the new one that shall be launched by May/June.

SuGlider, 2025-04-07:
> [@SuGlider](https://github.com/SuGlider) Here is a sketch. I cobbled it together from the Arduino example sketches for BLE client and server, so it should be as close to a textbook example as possible. The steps to use it: set up a BLE server that has a service and characteristic given by clientServiceUUID and clientCharUUID (or change their values), then on another device (I used an ESP32S3 dev board), turn the debug log level to Warning or above and upload this sketch. You should see that the client connection triggers the server connection callback, and when notify() is called, it will cause the truncation warning to occur because the server is trying to incorrectly send notification data to the client. You can also connect a client with high MTU, such as a laptop to see that that will receive the data correctly, without causing another warning. Please let me know if anything is unclear.
> 
> (Side note: I've noticed a possibly related bug where disconnecting the peripheral device will also cause the server to forcibly disconnect all clients connected to it, because it thinks the server's connections are actually client connections with 0 servers using them. This may be related, but is definitely a bug in the Bluedroid stack implementation that ESP32 uses, not Arduino. I can provide more info on it if you discover that it's relevant.)
> 
> [dual_mode_double_notify_bug.ino.zip](https://github.com/user-attachments/files/17689880/dual_mode_double_notify_bug.ino.zip)

I'll give it a try and check if there is something simple that would fix it.