espressif/arduino-esp32 #10574: Esp32 Psram api broken within the past few months.

XanCraft21, 2024-11-06:
### Board

ESP32 Dev Module and ESP32-S3 Dev Module

### Device Description

ESP32-S3 DevKitC board with 8M flash and 8M psram, bought from amazon on the official Espressif seller. ESP32 (old wrover version) DevKitC VE from Aliexpress from a different seller with same amount of memory.

### Hardware Configuration

Oled display using the default SPI interface and a few random pins for control signals.

### Version

v3.0.0

### IDE Name

Arduino IDE (legacy 1.8.19)

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

Whatever the highest one is starting with the number 9 (i forgot off the top of my head).

### Description

I was working with a display using a canvas or display buffer using the internal ram, and that works fine. My library of choice was Arduino_GFX (aka GFX_Library_For_Arduino), and it was fine. Earlier this year i enabled the use of psram, and it was fine. I could put on a very large display and have no problem (except for slow refresh rate but that’s from such a large display on a serial interface).

Recently (including after updating the software) i tested it and it suddenly started refusing to work correctly. The psram is detected but it refuses to allocate the memory. I don’t use these boards that much so it’s definitely not a hardware problem.

I was going to try to use a complex 3D graphics library that indefinitely overflows the main ram so it needs to be modified to use the psram, no luck. I went back to an extremely basic test sketch i made with Arduino_GFX (had to downgrade it because versions above 1.4.7 look like they may have stopped supporting the psram memory), and it kept rebooting for no reason. I know good and well it worked just fine earlier this year, and i never modified anything to make it not work. Looking at the print logs it appears to be falsely rebooting from a watchdog timer, if I understand the error code correctly.

How it should work is with the 3D library i should see a plain red triangle in the center of the screen, and with the basic test sketch the display should be slowly cycling between basic colors.

this bug appears to be happening between Arduino ESP32 releases between version 3.0.0 and up. I may be the only one with this issue since i don’t see any other posts talking about this issue.

I must also add that my computer gets hacked a lot, i have a virus that slows down downloads from the boards manager, a virus that selectively corrupts code when i try to download certain versions (in this case 3.0.2, it went super slow at first so i did a hard reset after properly stopping it and deleting the package logs, now it refuses to install entirely), and a virus that corrupts code as it’s being uploaded to the dev board when certain features are used (in this case psram), and what’s weird is that my antivirus cannot detect them. Please don’t take this as the final answer and refuse to help me, i need to make sure other people can replicate this issues so i know if i’m crazy or not. I plan to test this on a mac computer in a few weeks, so i hope it’s just my windows 10 pc that’s the problem, but until then please help me with this.

Thank you for any help you can give me, i need my psram to work properly for some projects i have in mind.

### Sketch

```cpp
I will add this later, but for now include the Arduino_GFX library, select a display, create a canvas (it should automatically convert to psram), and start using it and sending drawing commands to it. Your display should flash infinitely without updating the screen, and you will see the error message in the Serial monitor.
```

### Debug Message

```plain
I will add this later, but i think it’s just a watchdog timer reset.
```

### Other Steps to Reproduce

Not much info can be put here. I already tried 2 ESP32 boards, and none of them work. I haven’t tried an ESP32-S2 yet but it will most likely have the same issue as the others.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-11-06:
If your board uses less than 8MB of PSRAM, please try v3.0.7. Some things about how we initialize PSRAM has changed. For 8MB (OPI) it is initialized in the bootloader by ESP-IDF, so we do not expect there to be issues. Still using the latest version is best.

XanCraft21, 2024-11-07:
I have already tried that version and a couple other versions and i still have the same problem. I will do further testing later, but it still concerns me that this is happening.

TD-er, 2024-11-09:
Just some ideas, no idea if those apply here, but maybe they can help you find similarities in your code (or the lib)

When accessing PSRAM, you cannot access flash at the same time. At least not without some penalty in performance.
So maybe you have code reading from flash while also accessing PSRAM? For example large functions which cannot all be pre-cached in RAM. Or flash-strings processing while also accessing PSRAM?

Do you have some callback functions which access PSRAM, which may not be tagged by IRAM? Those could be not in RAM and thus needed to be read from flash to execute.

Not 100% sure if correct, but I think memory-access to PSRAM should be 4-byte aligned.

There is some setting which set some threshold on where memory allocation should be done (PSRAM/internal RAM).
For example all allocations larger than 1k (just a number, not idea what the actual threshold is)
Maybe this has recently changed and for your specific use case some allocation may now be done in PSRAM and then accessed in such a way which is allowed for internal RAM but not for PSRAM?

Are floats being used in callback functions?

Maybe also check if these issues also occur when running the PSRAM at a lower frequency.

XanCraft21, 2024-11-15:
Sorry for the late reply.

I am not accessing the flash alongside the psram, i am not using any callback functions, i am not using any byte alignment, just 16-bit integers, i will need to look more into the threshold stuff, as far as i can tell i cannot change the psram frequency since i’m using Arduino.

I am honestly not sure what’s happening, but i need some professionals to help me with this. Maybe if possible could someone test the setup i was using? I will open a github gist when possible.

XanCraft21, 2024-11-30:
Here’s a quick update, i think either my computer got hacker or in the latest esp32 version the psram api is broken. I can only manually enter values from scratch, i can’t have a library do it. It’s been broken since version 3.0.0. Is it possible to revert the changes in the next version? I don’t want to have to downgrade the software to get it working again. Thank you for your help.

XanCraft21, 2024-12-01:
I have discovered that the problem is definitely on my end, and downgrading the software package below version 3.0.0 didn’t fix it. I am going to temporarily close this issue until i can fix it. Thank you for the help you gave me so far.

closed 2024-12-01 by XanCraft21