espressif/arduino-esp32 #10580: WiFi.localIP() reports 0.0.0.0 after succesfull connect on Arduino Core 3.0.7

CelliesProjects, 2024-11-06:
### Board

esp32-s3

### Device Description

Nothing, just a plain old esp32-s3

### Hardware Configuration

Nothing of consequence

### Version

v3.0.6

### IDE Name

PlatformIO

### Operating System

Mint Linux

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

WiFi.localIP() should report the IP address. Instead it displays 0.0.0.0 after a succesfull connection.

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>

const char *SIDD = "xxx";
const char *PSK = "xxx";

void setup()
{
    Serial.begin(115200);
    while (!Serial)
        delay(10);

        Serial.println("connecting..");

    WiFi.begin(SIDD, PSK);

    while (!WiFi.isConnected())
        delay(10);

    Serial.println(WiFi.localIP());
}

void loop()
{
    delay(100);
}
```

### Debug Message

```plain
connecting..
0.0.0.0
```

### Other Steps to Reproduce

I did not try any other hw.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2024-11-09:
I do agree it may sound confusing, but I am not sure this really is a bug.
Afterall these are two quite different concepts, connected and gotIP.
It can take a while to get an IP from the DHCP server, especially for IPv6 it may take several seconds.

Not sure what "Arduino" claims how 'connected' should be interpreted.

CelliesProjects, 2024-11-09:
@TD-er Maybe not all, but most example sketches  for esp32 use this construct to connect to WiFi.
```c++
    while (!WiFi.isConnected())
        delay(10);
```
I don't claim it is a bug, but it surprised me. And my assumptions about code behaviour might be wrong.

TD-er, 2024-11-09:
Yep, that's why I also wonder what "the Arduino" expected interpretation should be.

SuGlider, 2024-11-09:
@me-no-dev - FYI.

Issue report confirmed. This behaviour started with Arduino Core 3.0.0.
Such behaviour is not noticed with Arduino Core 2.0.17, which marks a change break.

Arduino WiFi Library (upstream) has no `isConnect()` API.
It relies on `status()` to know when it is connected to WiFi.
https://docs.arduino.cc/libraries/wifi/

Possible Status are:
``` cpp
typedef enum {
	WL_NO_SHIELD = 255,
        WL_IDLE_STATUS = 0,
        WL_NO_SSID_AVAIL,
        WL_SCAN_COMPLETED,
        WL_CONNECTED,
        WL_CONNECT_FAILED,
        WL_CONNECTION_LOST,
        WL_DISCONNECTED
} wl_status_t;
```
The WiFi Arduino API may consider implicitly that as soon as status is `WL_CONNECT`, there is a valid IP address associated.

The current behaviour may be acceptable or not. We will discuss it internally.
I'll report here the results.

SuGlider, 2024-11-11:
Decision: We will change `WiFi.isConnected()` to only return when WiFi is connected and ESP32 has got an IP address to make it work as before. I'll add a PR to fix it.

SuGlider, 2024-11-11:
@CelliesProjects - Thank you for the report and quick fix. We will apply it!

PR #10595 (2024-11-11): fix(wifi): fixes WiFi.isconnected() to return true when it is connected and it has an IP Addr
## Description of Change
After 3.0.x (including 3.1.x) the function `WiFiSTAClass::isConnected()` returns when it is connected to the WiFi AP but not necessarily when it has an IP address.
This is a breaking change because 2.0.x only returns true when both, it is connected and it has an IP.

This PR fixes it.

## Tests scenarios
``` cpp
#include <Arduino.h>
#include <WiFi.h>

const char *SSID = "xxx";
const char *PSK = "xxx";

void setup()
{
    Serial.begin(115200);
    while (!Serial)
        delay( [... cut]
closed 2024-11-11 by me-no-dev