espressif/arduino-esp32 #10587: USB-JTAG pin19 and pin20 to be used as GPIO

muzahid13, 2024-11-09:
### Board

esp32-s3

### Device Description

i am using esp32-s3 based LVGL 7 inch display. I need to use the USB-JTAG pin19 and pin20 as gpio. i already prepared the PCB. i just made high and low pin20 at regular interval, then i noticed that it impacted on other code, i am using visual studio code platformio and framework: arduino.how can i disbale usb-jtag in platformio and framework arduino. my code is completed, i stuck on pin19 and pin20 to be gpio. thanks

### Hardware Configuration

GPIO pin19 and pin20 are connected to max485 D-enable and R-enable pin. so those pin shoudl be used as GPIO pin

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

pin19 and pin20 must be used as GPIO. when i enable PIN19 and 20 as GPIO, the code impact on other portion like the mcu is hanged

### Sketch

```cpp
void loop()
{
  lv_timer_handler(); /* let the GUI do its work */
  delay(5);

  pinMode(20,OUTPUT);
  digitalWrite(20,LOW);
 if (Serial1.available()>0)
  {
  digitalWrite(20,HIGH);
  Serial1.write(Serial1.read());
  delay(2);
  digitalWrite(20,LOW);
  }

}
```

### Debug Message

```plain
no error found on debug
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-11-09:
Idk if it is possible at all to get this restricted GPIOs glitch free.
There are some sdkconfig options to disable some USB JTAG features. Maybe it needs burning some efuses to solve.
In every case you have to compile your project with Arduino as an component of IDF to have the possibility to change the sdkconfig settings.

In short not possible with Arduino "only"

SuGlider, 2024-11-09:
There should be no problem using pins 20 and 19 from S3.
The configuration shall not enable USB. 
Check if `ARDUINO_USB_CDC_ON_BOOT=0` to disable this USB.

muzahid13, 2024-11-10:
> configuration

I placed 
#define ARDUINO_USB_CDC_ON_BOOT 0
but unfortunately the my lvgl code didn’t work.  Actually usb pin19 and 20 works as gpio but impact on others code. I need to disable usb jtag then use as gpio

SuGlider, 2024-11-10:
> > configuration
> 
> I placed #define ARDUINO_USB_CDC_ON_BOOT 0 but unfortunately the my lvgl code didn’t work. Actually usb pin19 and 20 works as gpio but impact on others code. I need to disable usb jtag then use as gpio

It doesn't work like that (it can't be defined in the sketch).
I have limited knowledge about PlatformIO, but it shall be done like here:
https://community.platformio.org/t/enabling-usb-cdc-on-boot-on-esp32-c3-devkit/33346/2

Maybe @Jason2866 can help it.

muzahid13, 2024-11-10:
> Idk if it is possible at all to get this restricted GPIOs glitch free. There are some sdkconfig options to disable some USB JTAG features. Maybe it needs burning some efuses to solve. In every case you have to compile your project with Arduino as an component of IDF to have the possibility to change the involved sdkconfig settings.
> 
> In short not possible with Arduino "only"

i am using platformio on vs code, below is my platformio.ini data
[env:4d_systems_esp32s3_gen4_r8n16]
platform = espressif32
board = 4d_systems_esp32s3_gen4_r8n16
framework = arduino
monitor_speed = 115200
lib_deps = lvgl/lvgl@^8.3.11
please suggest where i need to change or add any contents. thanks

Jason2866, 2024-11-10:
This:
```
[env:4d_systems_esp32s3_gen4_r8n16]
platform = espressif32
board = 4d_systems_esp32s3_gen4_r8n16
framework = arduino
monitor_speed = 115200
lib_deps = lvgl/lvgl@^8.3.11
```
uses any Arduino version Platformio founds. You don't know if this is the latest. I highly doubt.
Use that
```
[env:4d_systems_esp32s3_gen4_r8n16]
platform = https://github.com/pioarduino/platform-espressif32.git
board = 4d_systems_esp32s3_gen4_r8n16
framework = arduino
monitor_speed = 115200
lib_deps = lvgl/lvgl@^8.3.11
```
The board you have specified use this settings
```
      "-DARDUINO_ESP32_S3R8N16",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_MODE=1",
      "-DARDUINO_USB_CDC_ON_BOOT=1",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1
```
for sure wrong regarding USB. To help what exactly S3 variant does you have.
Flash size and type and PSRAM type (QIO or OPI)?
With this Infos it is possible to help

muzahid13, 2024-11-11:
> platform = https://github.com/pioarduino/platform-espressif32.git

[image]
i shared the esp32s3 picture here. i am using this mcu. please check. thanks

Jason2866, 2024-11-11:
16MB QIO Flash and 8MB OPI PSRAM 
Use this https://github.com/sivar2311/ESP32-S3-PlatformIO-Flash-and-PSRAM-configurations?tab=readme-ov-file#esp32-s3-wroom-11u-n16r8

SuGlider, 2024-11-11:
@muzahid13 
Anyway, this issue has a resolution. pin 19 and 20 work correctly as GPIO.
The rest of the code that is not working is not related to an Arduino Core issue, right?

Therefore, should we close this issue?

muzahid13, 2024-11-12:
> @muzahid13 Anyway, this issue has a resolution. pin 19 and 20 work correctly as GPIO. The rest of the code that is not working is not related to an Arduino Core issue, right?
> 
> Therefore, should we close this issue?

but if i dont use pin19 and 20 as gpio, my other code works good. only impact appeared on other code when i use pin19 and 20 as GPIO.

everslick, 2024-11-26:
Small advice: Look up how to properly format github issues and comments... e.g. the code above is unreadable. Enclose it like this:

\`\`\`
`this is`
`my`
`program`
\`\`\`

muzahid13, 2024-11-27:
> but if i dont use pin19 and 20 as gpio, my other code works good. only impact appeared on other code when i use pin19 and 20 as GPIO. below is my complete code:
> 
> ```
> /*******************************************************************************
>  * LVGL Widgets
>  * This is a widgets demo for LVGL - Light and Versatile Graphics Library
>  * import from: https://github.com/lvgl/lv_demos.git
>  *
>  * Dependent libraries:
>  * LVGL: https://github.com/lvgl/lvgl.git
> 
>  * Touch libraries:
>  * FT6X36: https://github.com/strange-v/FT6X36.git
>  * GT911: https://github.com/TAMCTec/gt911-arduino.git
>  * XPT2046: https://github.com/PaulStoffregen/XPT2046_Touchscreen.git
>  *
>  * LVGL Configuration file:
>  * Copy your_arduino_path/libraries/lvgl/lv_conf_template.h
>  * to your_arduino_path/libraries/lv_conf.h
>  * Then find and set:
>  * #define LV_COLOR_DEPTH     16
>  * #define LV_TICK_CUSTOM     1
>  *
>  * For SPI display set color swap can be faster, parallel screen don't set!
>  * #define LV_COLOR_16_SWAP   1
>  *
>  * Optional: Show CPU usage and FPS count
>  * #define LV_USE_PERF_MONITOR 1
>  ******************************************************************************/
> 
> #include <Arduino.h>
> #include <lvgl.h>
> #include "uart0.h"
> #include "ui.h"
> 
> 
> #define TX1_PIN 17
> #define RX1_PIN 18
> 
> #define TX_PIN 43
> #define RX_PIN 44
> 
> #include <Arduino_GFX_Library.h>
> #define TFT_BL 2
> // #define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin
> 
> /* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */
> #if defined(DISPLAY_DEV_KIT)
> Arduino_GFX *gfx = create_default_Arduino_GFX();
> #else /* !defined(DISPLAY_DEV_KIT) */
> 
> /* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
> // Arduino_DataBus *bus = create_default_Arduino_DataBus();
> 
> /* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
> // Arduino_GFX *gfx = new Arduino_ILI9341(bus, DF_GFX_RST, 0 /* rotation */, false /* IPS */);
> 
> Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
>     GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
>     41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
>     14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
>     9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
>     15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */
> );
> // option 1:
> // 7寸 50PIN 800*480
> Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
>     bus,
>     //  800 /* width */, 0 /* hsync_polarity */, 8/* hsync_front_porch */, 2 /* hsync_pulse_width */, 43/* hsync_back_porch */,
>     //  480 /* height */, 0 /* vsync_polarity */, 8 /* vsync_front_porch */, 2/* vsync_pulse_width */, 12 /* vsync_back_porch */,
[... 406 lines cut ...]
uint8_t DontHvLmt[6] = {0x02, 0x29, 0x03, 0xE9, 0x39,0xF8};
uint8_t blockCD[6]   = {0x02, 0x29, 0x03, 0xE9, 0x3A,0xFB};
uint8_t carPlate[6]  = {0x02, 0x29, 0x03, 0xE9, 0x44,0x85};
uint8_t EnterPin[6]  = {0x02, 0x29, 0x03, 0xE9, 0x45,0x84};

uint8_t CDbelongTo[32]= {0x02, 0x29, 0x03, 0xE9, 0x37,'A','d','h','a','m','A','d','h','a','m','A','d','h','a','m','A',0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xF6};
//02 29 03 E9 37 41 64 68 61 6D 41 64 68 61 6D 41 64 68 61 6D 41 64 68 61 6D 41 64 68 61 6D 78 CF
uint8_t disAuthLmt[10]= {0x02, 0x29, 0x03, 0xE9, 0x36,0x11, 0x22, 0x33, 0x44,0xB3};
                         //02 29 03 E9 36 11 22 33 44 B3
uint8_t sendAck[6]        = {0x02, 0x29, 0x03, 0xE9, 0xAF,0x6E};
uint8_t recvAck[6]        = {0x02, 0x29, 0x03, 0xE9, 0xBF,0x7E};
 

void loop()
{
  lv_timer_handler(); /* let the GUI do its work */
  delay(5);

pinMode(20,OUTPUT);
digitalWrite(20,LOW);
delay(1000);
digitalWrite(20,HIGH);
delay(1000);
}
for the above code my lvgl code is stuck when pin20 is active as GPIO but if i remove pin20 gpio code, my lvgl code is working good. could you please help to solve this?

muzahid13, 2024-11-27:
> Small advice: Look up how to properly format github issues and comments... e.g. the code above is unreadable. Enclose it like this:
> 
> ``` `this is` `my` `program` ```
i updated my below code. could anyone help me to solve the issue
but if i dont use pin19 and 20 as gpio, my other code works good. only impact appeared on other code when i use pin19 and 20 as GPIO. 
below is my complete code:
```
/*******************************************************************************
 * LVGL Widgets
 * This is a widgets demo for LVGL - Light and Versatile Graphics Library
 * import from: https://github.com/lvgl/lv_demos.git
 *
 * Dependent libraries:
 * LVGL: https://github.com/lvgl/lvgl.git

 * Touch libraries:
 * FT6X36: https://github.com/strange-v/FT6X36.git
 * GT911: https://github.com/TAMCTec/gt911-arduino.git
 * XPT2046: https://github.com/PaulStoffregen/XPT2046_Touchscreen.git
 *
 * LVGL Configuration file:
 * Copy your_arduino_path/libraries/lvgl/lv_conf_template.h
 * to your_arduino_path/libraries/lv_conf.h
 * Then find and set:
 * #define LV_COLOR_DEPTH     16
 * #define LV_TICK_CUSTOM     1
 *
 * For SPI display set color swap can be faster, parallel screen don't set!
 * #define LV_COLOR_16_SWAP   1
 *
 * Optional: Show CPU usage and FPS count
 * #define LV_USE_PERF_MONITOR 1
 ******************************************************************************/

#include <Arduino.h>
#include <lvgl.h>
#include "uart0.h"
#include "ui.h"

#define TX1_PIN 17
#define RX1_PIN 18

#define TX_PIN 43
#define RX_PIN 44

#include <Arduino_GFX_Library.h>
#define TFT_BL 2
// #define GFX_BL DF_GFX_BL // default backlight pin, you may replace DF_GFX_BL to actual backlight pin

/* More dev device declaration: https://github.com/moononournation/Arduino_GFX/wiki/Dev-Device-Declaration */
#if defined(DISPLAY_DEV_KIT)
Arduino_GFX *gfx = create_default_Arduino_GFX();
#else /* !defined(DISPLAY_DEV_KIT) */

/* More data bus class: https://github.com/moononournation/Arduino_GFX/wiki/Data-Bus-Class */
// Arduino_DataBus *bus = create_default_Arduino_DataBus();

/* More display class: https://github.com/moononournation/Arduino_GFX/wiki/Display-Class */
// Arduino_GFX *gfx = new Arduino_ILI9341(bus, DF_GFX_RST, 0 /* rotation */, false /* IPS */);

Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
    GFX_NOT_DEFINED /* CS */, GFX_NOT_DEFINED /* SCK */, GFX_NOT_DEFINED /* SDA */,
    41 /* DE */, 40 /* VSYNC */, 39 /* HSYNC */, 42 /* PCLK */,
    14 /* R0 */, 21 /* R1 */, 47 /* R2 */, 48 /* R3 */, 45 /* R4 */,
    9 /* G0 */, 46 /* G1 */, 3 /* G2 */, 8 /* G3 */, 16 /* G4 */, 1 /* G5 */,
    15 /* B0 */, 7 /* B1 */, 6 /* B2 */, 5 /* B3 */, 4 /* B4 */
);
// option 1:
// 7寸 50PIN 800*480
Arduino_RPi_DPI_RGBPanel *gfx = new Arduino_RPi_DPI_RGBPanel(
[... 158 lines cut ...]
uint8_t blockCD[6]   = {0x02, 0x29, 0x03, 0xE9, 0x3A,0xFB};
uint8_t carPlate[6]  = {0x02, 0x29, 0x03, 0xE9, 0x44,0x85};
uint8_t EnterPin[6]  = {0x02, 0x29, 0x03, 0xE9, 0x45,0x84};

uint8_t CDbelongTo[32]= {0x02, 0x29, 0x03, 0xE9, 0x37,'A','d','h','a','m','A','d','h','a','m','A','d','h','a','m','A',0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0x0,0xF6};
//02 29 03 E9 37 41 64 68 61 6D 41 64 68 61 6D 41 64 68 61 6D 41 64 68 61 6D 41 64 68 61 6D 78 CF
uint8_t disAuthLmt[10]= {0x02, 0x29, 0x03, 0xE9, 0x36,0x11, 0x22, 0x33, 0x44,0xB3};
                         //02 29 03 E9 36 11 22 33 44 B3
uint8_t sendAck[6]        = {0x02, 0x29, 0x03, 0xE9, 0xAF,0x6E};
uint8_t recvAck[6]        = {0x02, 0x29, 0x03, 0xE9, 0xBF,0x7E};
 

void loop()
{
  lv_timer_handler(); /* let the GUI do its work */
  delay(5);

pinMode(20,OUTPUT);
digitalWrite(20,LOW);
delay(1000);
digitalWrite(20,HIGH);
delay(1000);
}
```
for the above code my lvgl code is stuck when pin20 is active as GPIO but if i remove pin20 gpio code, my lvgl code is working good. could you please help to solve this?

SuGlider, 2025-01-13:
@Rob58329 - please try this code in the begining of `setup()`:

``` cpp
void setup()
{
  pinMode(19, OUTPUT);
  pinMode(20, OUTPUT);
  uart0_init();

```

Rob58329, 2025-01-13:
@SuGlider;  I think you maybe wanted to @ reference someone else, as above issue is for ESP32-S3, and mine was re. ESP32-C3?

SuGlider, 2025-01-13:
> @SuGlider;  I think you maybe wanted to @ reference someone else, as above issue is for ESP32-S3, and mine was re. ESP32-C3?

Thanks for the warning.
It was for @muzahid13 instead.